robotology/robotology-superbuild #176: Test the superbuild on iCubGenova01 and iCubGenova02

traversaro, 2019-02-25:


traversaro, 2019-03-13:
Test on iCubGenova01 are blocked until it will be updated to Ubuntu 18.04 .

traversaro, 2019-06-25:
Some discussion about the deployment of the superbuild on iCubGenova01 was ongoing in https://github.com/robotology/robotology-superbuild/pull/212 . I renamed the issue as here we can discuss the deployment on iCubGenova01, for the one on iCubGenova02 we can open new issues (or not, if it is not necessary).

traversaro, 2019-06-25:
Issues that appeared when testing on iCubGenova01: :
* https://github.com/robotology/robotology-superbuild/pull/214
* https://github.com/robotology/robotology-superbuild/pull/216

traversaro, 2019-06-25:
Nicola (@xEnVrE) mentioned that in that setup they may need to install OpenCV from sources. In that case, it definitely make sense to just clone ÒpenCV in `$ROBOT_CODE` as usual, but it would probably make sense to agree on where to install non-superbuild projects.

xEnVrE, 2019-06-25:
For now, we decided to clone `OpenCV` and the `OpenCV contrib` repository in `$ROBOT_CODE`.

traversaro, 2019-06-26:
Problems still pending: 
* robots-configuration should be moved in the core repo ( https://github.com/robotology/robotology-superbuild/issues/217 )
* `ROBOTOLOGY_USES_LUA` should be tested and documented ( https://github.com/robotology/robotology-superbuild/issues/30 ) 
* `ROBOTOLOGY_USES_PYTHON` should be documented  https://github.com/robotology/robotology-superbuild/issues/28

pattacini, 2019-06-26:
One aspect that could be relevant to track is that the general `make` could change cmake flag that the user may have customized for the single project.

If we flag that project as under `DEVEL_MODE` we should prevent it from being overwritten.

traversaro, 2019-06-26:
> One aspect that could be relevant to track is that the general make could change cmake flag that the user may have customized for the single project.
>
> If we flag that project as under DEVEL_MODE we should prevent it from being overwritten.

Good point (as we discussed f2f), perhaps @drdanz could have some insight on this. A related issue (even if there the focus was more on "provide the correct documentation on how to handle options not touched by the superbuild") is https://github.com/robotology/robotology-superbuild/issues/67 .

drdanz, 2019-07-01:
Since we updated to the latest [`ExternalProject`](https://cmake.org/cmake/help/latest/module/ExternalProject.html), we can now rely on `CMAKE_CACHE_DEFAULT_ARGS` instead of passing arguments as `CMAKE_ARGS` and `CMAKE_CACHE_ARGS`. In this way, arguments are never overwritten

traversaro, 2019-08-23:
> Since we updated to the latest [`ExternalProject`](https://cmake.org/cmake/help/latest/module/ExternalProject.html), we can now rely on `CMAKE_CACHE_DEFAULT_ARGS` instead of passing arguments as `CMAKE_ARGS` and `CMAKE_CACHE_ARGS`. In this way, arguments are never overwritten

The main problem (described in https://github.com/robotology/robotology-superbuild/issues/68) is that for some options, we want the option to be overwritten, for example when in a superbuild installation in which `ROBOTOLOGY_USES_LUA` is first set to `OFF`, and as soon as the user set it to `ON`, we want the relevant YARP cache option to be overwritten. We can track this aspect in https://github.com/robotology/robotology-superbuild/issues/68 . 
Regarding the testing on iCubGenova01, I think we can consider that successful, and we can now close this issue.

closed 2019-08-23 by traversaro