robotology/robotology-superbuild #177: brew install dartsim/dart/ipopt

sangwonH, 2019-02-25:
I ran 'brew install dartsim/dart/ipopt' on mac(sierra 10.12) and got the message below.
I think while processing 'make install' some error occurs.
Can anyone give me advice? Thank you.

Warning: dartsim/dart/ipopt dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
🍺  /usr/local/Cellar/ipopt/3.12.9: 91 files, 6.5MB, built in 1 minute 38 seconds

traversaro, 2019-02-26:
Hi @sangwonH, I think that is a normal warning, and should not affect the behavior of the ipopt library. Are you experiencing any issue running the software (i.e. "at runtime")?

sangwonH, 2019-02-26:
Noted. I appreciate your comment/advice.

closed 2019-02-26 by sangwonH