robotology/robotology-superbuild #178: Compilation fails on icub-head when icub-main is in devel branch

gabrielenava, 2019-03-06:
I tried to compile the robotology superbuild on `icub-head`. The compilation failed when trying to compile `icub-main` with the following error:

```
Scanning dependencies of target embObjMotionControl
[ 23%] Building CXX object src/libraries/icubmod/embObjMotionControl/CMakeFiles/embObjMotionControl.dir/yarp_plugin_icubmod_embObjMotionControl.cpp.o
[ 23%] Building CXX object src/libraries/icubmod/embObjMotionControl/CMakeFiles/embObjMotionControl.dir/embObjMotionControl.cpp.o
[ 23%] Building CXX object src/libraries/icubmod/embObjMotionControl/CMakeFiles/embObjMotionControl.dir/eomcParser.cpp.o
[ 24%] Linking CXX shared module ../../../../lib/iCub/embObjMotionControl.so
/usr/bin/ld: cannot find -lethResources
collect2: error: ld returned 1 exit status
src/libraries/icubmod/embObjMotionControl/CMakeFiles/embObjMotionControl.dir/build.make:181: recipe for target 'lib/iCub/embObjMotionControl.so' failed
make[5]: *** [lib/iCub/embObjMotionControl.so] Error 1
CMakeFiles/Makefile2:2018: recipe for target 'src/libraries/icubmod/embObjMotionControl/CMakeFiles/embObjMotionControl.dir/all' failed
make[4]: *** [src/libraries/icubmod/embObjMotionControl/CMakeFiles/embObjMotionControl.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/ICUB.dir/build.make:108: recipe for target 'robotology/ICUB/CMakeFiles/YCMStamp/ICUB-build' failed
make[2]: *** [robotology/ICUB/CMakeFiles/YCMStamp/ICUB-build] Error 2
CMakeFiles/Makefile2:3871: recipe for target 'CMakeFiles/ICUB.dir/all' failed
make[1]: *** [CMakeFiles/ICUB.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
```

I verified that `icub-firmware-shared` is also installed by the superbuild. However, despite I manually specified the path to the install folder of `icub-firmware-shared`, there is an error while configuring the `icub-main` that states that `icub-firmware-shared` folder is not found, and I guess this is the reason why the compilation fails later.

I managed to compile the superbuild after switching `icub-main` and `icub-firmware-shared` to `master` branch. I don't know if the problem is related to the superbuild itself, or maybe to a bug in `icub-main`, or something local inside `icub-head`.

gabrielenava, 2019-03-06:
cc @S-Dafarra

traversaro, 2019-03-06:
> I verified that `icub-firmware-shared` is also installed by the superbuild. However, despite I manually specified the path to the install folder of `icub-firmware-shared`, there is an error while [...]

What is the precise error? This sounds like a wrong version in the installed `icub-firmware-shared` or in the version of icub-firmware-shared requested by `icub-main` in the devel branch.

traversaro, 2019-03-06:
The version of `icub-firmware-shared` https://github.com/robotology/icub-firmware-shared/blob/devel/CMakeLists.txt#L11 seems to be compatible with the one requested  in https://github.com/robotology/icub-main/blob/devel/conf/iCubFindDependencies.cmake#L34 .

gabrielenava, 2019-03-06:
I replicated the error by turning the `icub-main` and the `icub-firmware-shared` on `devel` branch. Here is the message I've got when running `ccmake . ` in the `icub-main` build folder:

```
 ERROR embObj: cannot find icub_firmware_shared library, turn off device embObjDevices
   embObj library can now be found in the icub-firmware-shared package that
   can be downloaded from here:
   .
        https://github.com/robotology/icub-firmware-shared
   .

```

gabrielenava, 2019-03-07:
The issue has been fixed, it turns out that the local installation of `icub-firmware-shared` was not pointing the right firmware version. Removing the `icub-firmware-shared` source folder and reinstalling it fixed the issue.

Closing.

closed 2019-03-07 by gabrielenava