robotology/robotology-superbuild #190: libqt5multimediawidgets5

triccyx, 2019-03-25:
Duging robotology-superbuild on Ubuntu 16.04 the apt-get install package list 
should be updated adding:
_libqt5multimediawidgets5_

Without this package an error is raised:

```
The following packages have unmet dependencies:
 qtmultimedia5-dev : Depends: libqt5multimediawidgets5 (= 5.5.1-4ubuntu2) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

traversaro, 2019-03-25:
As `libqt5multimediawidgets5` is a dependency of the  `qtmultimedia5-dev` package, in a clean system  installing `qtmultimedia5-dev` should result in an automatic installation of  `libqt5multimediawidgets5`, so I guess there is something strange on your system.

traversaro, 2019-04-05:
Friendly ping @triccyx .

triccyx, 2019-05-17:
Ok for me

closed 2019-05-17 by triccyx