robotology/robotology-superbuild #191: [Make Error] recipe for target 'all' failed

VenusPasandi, 2019-04-01:
I tried to install the robotology-superbuild and I got the following error when I do `make`,

>[ 63%] Built target yarp_test_grabber
[ 63%] Linking CXX shared module ../../../lib/yarp/yarp_SDLJoypad.so
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libSDL.a(SDL.o): relocation R_X86_64_PC32 against symbol `SDL_InitSubSystem' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
src/devices/SDLJoypad/CMakeFiles/yarp_SDLJoypad.dir/build.make:125: recipe for target 'lib/yarp/yarp_SDLJoypad.so' failed
make[5]: *** [lib/yarp/yarp_SDLJoypad.so] Error 1
CMakeFiles/Makefile2:2417: recipe for target 'src/devices/SDLJoypad/CMakeFiles/yarp_SDLJoypad.dir/all' failed
make[4]: *** [src/devices/SDLJoypad/CMakeFiles/yarp_SDLJoypad.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/YARP.dir/build.make:108: recipe for target 'robotology/YARP/CMakeFiles/YCMStamp/YARP-build' failed
make[2]: *** [robotology/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/YARP.dir/all' failed
make[1]: *** [CMakeFiles/YARP.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

I don't know what is the reason but I got the following warning when I did `cmake` 

> CMake Warning (dev) at build/CMakeFiles/YCMEPHelper.cmake:171 (message):
>    YCM_BOOTSTRAP HASH mismatch
> 
>      for file: [/home/vpasandi/Documents/repositories/robotology-superbuild/cmake/IncludeUrl.cmake]
>        expected hash: [e31dffed0897729283ef4e945bbdeb1921b438b5]
>          actual hash: [ae6de83398115e5b8fd6c261972dea82d1aeb3fd]
> 
>    Perhaps it is outdated or you have local modification.  Please consider
>    upgrading it, or contributing your changes to YCM.
> 
>  Call Stack (most recent call first):
>    build/CMakeFiles/YCMEPHelper.cmake:1330 (_ycm_hash_check)
>    cmake/YCMBootstrap.cmake:114 (ycm_bootstrap)
>    CMakeLists.txt:104 (include)
>  This warning is for project developers.  Use -Wno-dev to suppress it.

triccyx, 2019-04-01:
I have the same issue, @Nicogene was able to workaround the problem with a trick.

traversaro, 2019-04-01:
Reproduced locally in my PC, the problem was not affecting Travis only because on Travis `libsdl1.2-dev` is not installed.

traversaro, 2019-04-01:
> I don't know what is the reason but I got the following warning when I did cmake

That is unrelated, it should be fixed by https://github.com/robotology/robotology-superbuild/pull/193 .

traversaro, 2019-04-01:
A workaround for this issue is this modification: https://github.com/robotology/robotology-superbuild/commit/b6ed1b3392a9cf94b2600c766f963283a52f1331 .

traversaro, 2019-04-01:
The problem  is that for some reason [this line](https://github.com/Kitware/CMake/blob/v3.10.3/Modules/FindSDL.cmake#L90) in the upstream `FindSDL.cmake` is finding `/usr/lib/x86_64-linux-gnu/libSDL.a` instead of `/usr/lib/x86_64-linux-gnu/libSDL.so`, and then passing a package static library (compiled without `-fPIC`) cause the error.  

I have no idea what is going on, but with YCM 0.8.0 the find_library founds the correct library `/usr/lib/x86_64-linux-gnu/libSDL.so`.

traversaro, 2019-04-01:
cc @drdanz

traversaro, 2019-04-01:
It may  be worth to try to create a [minimal working example](https://en.wikipedia.org/wiki/Minimal_working_example) and if we are  able to do so, move the bug in YCM.

traversaro, 2019-04-01:
I found the root cause. On YCM  0.8, [`CMAKE_FIND_LIBRARY_SUFFIXES`](https://cmake.org/cmake/help/v3.10/variable/CMAKE_FIND_LIBRARY_SUFFIXES.html) has the correct value `.so;.a`: the precedence of `.so` is what ensure that dynamic libraries are found before static ones. With YCM 0.10,  [`CMAKE_FIND_LIBRARY_SUFFIXES`](https://cmake.org/cmake/help/v3.10/variable/CMAKE_FIND_LIBRARY_SUFFIXES.html) has the wrong value `.a`, meaning that `find_library` only looks for static libraries. 
This is caused by the call to the FindGLEW script in https://github.com/robotology/yarp/blob/master/cmake/YarpFindDependencies.cmake#L420 : moving the `find_package(GLEW QUIET)` call **after** `find_package(SDL QUIET)` and  cleaning the cache solves the problem. The problem was introduced by https://gitlab.kitware.com/cmake/cmake/merge_requests/3094, that introduced some `    set(CMAKE_FIND_LIBRARY_SUFFIXES ".a" PARENT_SCOPE)` calls in the `FindGLEW.cmake` script, that corrupt the `CMAKE_FIND_LIBRARY_SUFFIXES` variable also outside the context of the `FindGLEW.cmake` script. This file has been imported in YCM with PR https://github.com/robotology/ycm/pull/226 .

cc @drdanz @claudiofantacci

claudiofantacci, 2019-04-02:
If this is the case, then I have to fix this upstream asap.

claudiofantacci, 2019-04-02:
Ok, you can replicate this issue by simply running this:

```
cmake_minimum_required(VERSION 3.5)

project(prova_glew)

find_package(YCM 0.10.0 REQUIRED)

message(STATUS "CMAKE_FIND_LIBRARY_SUFFIXES: ${CMAKE_FIND_LIBRARY_SUFFIXES}")

find_package(GLEW REQUIRED)

message(STATUS "CMAKE_FIND_LIBRARY_SUFFIXES: ${CMAKE_FIND_LIBRARY_SUFFIXES}")
```

I'll fix this upstream and post here the fix. It should be a matter of adding two lines in `FindGLEW.cmake`.

claudiofantacci, 2019-04-02:
To fix this issue, open `<YCM build/installation dir>/share/YCM/cmake-next/Modules/FindGLEW.cmake` and replace [these lines](https://gitlab.kitware.com/cmake/cmake/blob/master/Modules/FindGLEW.cmake#L120-143) with the following:

```cmake
set(USER_FIND_LIBRARY_SUFFIXES ${CMAKE_FIND_LIBRARY_SUFFIXES})

__glew_set_find_library_suffix(SHARED)

find_library(GLEW_SHARED_LIBRARY_RELEASE
             NAMES GLEW glew glew32
             PATH_SUFFIXES lib lib64 libx32 lib/Release/${_arch}
             PATHS ENV GLEW_ROOT)

find_library(GLEW_SHARED_LIBRARY_DEBUG
             NAMES GLEWd glewd glew32d
             PATH_SUFFIXES lib lib64
             PATHS ENV GLEW_ROOT)

__glew_set_find_library_suffix(STATIC)

find_library(GLEW_STATIC_LIBRARY_RELEASE
             NAMES GLEW glew glew32s
             PATH_SUFFIXES lib lib64 libx32 lib/Release/${_arch}
             PATHS ENV GLEW_ROOT)

find_library(GLEW_STATIC_LIBRARY_DEBUG
             NAMES GLEWds glewds glew32ds
             PATH_SUFFIXES lib lib64
             PATHS ENV GLEW_ROOT)

set(CMAKE_FIND_LIBRARY_SUFFIXES ${USER_FIND_LIBRARY_SUFFIXES})
```

Since `FindGLEW.cmake` is downloaded from upstream CMake, be sure to apply this patch every time the file is changed by the build/installation process.

I'm opening a PR upstream to get this fixed once and for all.

claudiofantacci, 2019-04-02:
Upstream PR: https://gitlab.kitware.com/cmake/cmake/merge_requests/3179

Sorry for the inconvenience this may caused and thanks for your help in detecting this issue 🥇

VenusPasandi, 2019-04-02:
> A workaround for this issue is this modification: b6ed1b3

@traversaro, using this trick, I did a clean installation (I mean without any error)!

claudiofantacci, 2019-04-04:
The fix was [merged upstream](https://gitlab.kitware.com/cmake/cmake/merge_requests/3179) 🎉 

We now need to update the YCM, cc @drdanz

drdanz, 2019-04-05:
Thanks @traversaro for debugging and @claudiofantacci for the fix! Imported it in YCM. I will import it in YARP as soon as I release YCM 0.10.1.

traversaro, 2019-04-05:
As it can be verified by https://github.com/robotology/robotology-superbuild/pull/198, the bootstrapped version has been fixed. I will wait for YCM 0.10.1 release to had the 0.10.1 YCM version as a requirement and close this issue.

drdanz, 2019-04-08:
YCM 0.10.0 just released

closed 2019-05-07 by traversaro