robotology/robotology-superbuild #192: Superbuid failed

triccyx, 2019-04-01:
Tring to compile using the superbuild I face this error.
(pull from master)

```
CMake Error at /home/ltricerri/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:183 (find_package):
  Found package configuration file:

    /home/ltricerri/robotology-superbuild/build/install/lib/cmake/YARP_manager/YARP_managerConfig.cmake

  but it set YARP_manager_FOUND to FALSE so package "YARP_manager" is
  considered to be NOT FOUND.  Reason given by package:

  The following imported targets are referenced, but are missing:
  YARP::YARP_math

Call Stack (most recent call first):
  CMakeLists.txt:15 (find_package)

-- Configuring incomplete, errors occurred!
```

traversaro, 2019-04-01:
Reproducible on Travis: https://travis-ci.org/robotology/robotology-superbuild/jobs/514246582 .

traversaro, 2019-04-01:
Workaround for the issue provided in https://github.com/robotology/icub-tests/pull/46 .

traversaro, 2019-04-02:
As the fix for this have been merged in https://github.com/robotology/icub-tests/pull/46 and the related YARP issue has been reported in https://github.com/robotology/yarp/issues/1989 , I think we can close this issue.

closed 2019-04-02 by traversaro