robotology/robotology-superbuild #195: Enable ccache in Travis CI for linux jobs

diegoferigo, 2019-04-03:
At a first glance, it doesn't seem that we exploit ccache in our linux jobs running in Travis CI. The folder is mounted:

https://github.com/robotology/robotology-superbuild/blob/288b636fdc3498403bcbdc6f6c9d32d300cc7759/.travis.yml#L38-L49

 but inside the container 1) [there is no ccache](https://github.com/robotology/robotology-superbuild/blob/master/.ci/install_debian.sh) and 2) is not enabled ([how to enable ccache](https://manpages.ubuntu.com/manpages/xenial/man1/ccache.1.html)). Addressing it might solve most of the timeouts we experience.

cc @traversaro

traversaro, 2019-04-03:
Indeed, adding it  would be good idea. However to be honest I am personally a bit afraid to invest time in the Travis infrastructure due to the current situation: https://devclass.com/2019/03/29/travis-ci-users-left-hanging-as-platform-lies-down/  .

diegoferigo, 2019-04-03:
I know, but it's a 2 lines modification. I already opened a PR that enables ccache in the meantime :laughing:

traversaro, 2019-04-03:
Ok perfect!

PR #196 (2019-04-04): Enable ccache in linux travis jobs
Closes #195
closed 2019-04-04 by traversaro