robotology/robotology-superbuild #202: Update MATLAB installation instructions

traversaro, 2019-05-06:
Recently `whole-body-controllers` added some installation steps (see https://github.com/robotology/whole-body-controllers#installation-and-usage), that are not mentioned nor referenced in the MATLAB instructions:
https://github.com/robotology/robotology-superbuild#matlab  .

We should update the section to either explain what to do or link the relevant documentation.

gabrielenava, 2019-07-12:
Entry point to WBC documentation added with https://github.com/robotology/robotology-superbuild/pull/229

In this specific case I think that the user that is interested in installing whole-body-controllers should read the associated README and documentation, that will be the only one mantained and updated. Therefore I prefer to provide an entry point to the WBC README rather than duplicate the info in the `robotology-superbuild` readme.

traversaro, 2019-07-25:
I do not know. While I guess that you need to read the docs for each repo you want to actually use, having **manual** installation steps for single subprojects seems to defeat the purpose of having a superbuild at all. I need to think a bit about a proper trade-off.

traversaro, 2019-08-22:
I did not have time to work on a better solution, so for the time being let's just merge your docs, and if in the future we want to improve this we can always do that.

traversaro, 2019-11-26:
@gabrielenava  Is this related to https://github.com/robotology/whole-body-controllers/issues/75 ?

gabrielenava, 2019-11-26:
Yes, there are some on going modifications in whole-body-controllers concerning the installation of the repo. I will handle all these issues.

gabrielenava, 2020-04-25:
I think the latest [WBC release](https://github.com/robotology/whole-body-controllers/releases/tag/v2.5) removed the necessity of extra installation steps for the repo: https://github.com/robotology/whole-body-controllers#installation-and-usage.

traversaro, 2020-04-25:
Great, so I guess we can close this issue? There is something to update in the README?

gabrielenava, 2020-04-25:
No I updated the robotology superbuild readme when releasing WBC 2.5. We can close the issue :-)

traversaro, 2020-04-25:
Great, thanks!

closed 2020-04-25 by traversaro