robotology/robotology-superbuild #213: Option to enable/disable testing

Nicogene, 2019-06-20:
Is there a way to compile the `robotology-superbuild` skipping the test part? (`robottestingframwork` and `icub-tests`)

If not, it would be nice adding it, since in some cases it is not useful and can give some compilation troubles (see robotology/yarp#2050).

traversaro, 2019-06-21:
I am a bit afraid about adding more cmake options, beside the profile ( https://github.com/robotology/robotology-superbuild#profile-cmake-options ) and dependencies one ( https://github.com/robotology/robotology-superbuild#dependencies-cmake-options ), as the project configuration is already quite complex as it is to maintain and understand for users. Something that I think could make sense is to add a "testing" profile that includes rtf, icub-tests and eventually blocktest, so that it would be possible to disable those projects via the usual profile systems. Do you think it would make sense?

Nicogene, 2019-06-21:
Fine for me :+1:

traversaro, 2019-10-25:
Ok for me. For anyone interested, please open a PR with this change.

closed 2019-11-25 by traversaro