robotology/robotology-superbuild #215: Unable to compile YARP_robottestingframework on Windows

Nicogene, 2019-06-21:
After renaming `RTF` in `robottestingframework` (see 06ae1329cfaa618c2af9457b8a4dee3a9cfb58ba, https://github.com/robotology/robot-testing-framework/issues/101, https://github.com/robotology/robot-testing-framework/pull/103) it is impossible to compile library containing the `robottestingframework` plugins a.k.a `YARP_robottestingframework` on windows, due to the reach the OS max path limit.
E.g.:
```
The expression "[MSBuild]::NormalizePath(C:\Users\ngenesio\robotology\robotology-superbuild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpplugin, robottestingframework_fixturemanager_yarpplugin.dir\Debug\, robottestingframework_fixturemanager_yarpplugin.vcxproj.CopyComplete)"
cannot be evaluated.
Path: C:\Users\ngenesio\robotology\robotology-superbuild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpplugin\robottestingframework_fixturemanager_yarpplugin.dir\Debug\robottestingframework_fixturemanager_yarpplugin.vcxproj.CopyComplete
exceeds the OS max path limit. The fully qualified file name must be less than 260 characters.  ALL_BUILD  C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets
```

Workaround(or fix):

https://www.howtogeek.com/266621/how-to-make-windows-10-accept-file-paths-over-260-characters/

Edit: it doesn't work for me

Nicogene, 2019-06-21:
Issue moved from YARP

traversaro, 2019-06-21:
The mitigation strategy used in https://github.com/microsoft/vcpkg/issues/426 was to warn the user if the base path was too short. Probably we could use the same strategy?

Nicogene, 2019-06-21:
> The mitigation strategy used in microsoft/vcpkg#426 was to warn the user if the base path was too short. Probably we could use the same strategy?

The same strategy in case the base path is too **long** ?

kouroshD, 2019-07-30:
I want to add the HDE profile to the superbuild, but unfortunately I have the same problem;, I cannot even build the core profile of the superbuild in windows.
CC @DanielePucci  @traversaro

traversaro, 2019-07-30:
> > The mitigation strategy used in microsoft/vcpkg#426 was to warn the user if the base path was too short. Probably we could use the same strategy?
>
> The same strategy in case the base path is too **long** ?

Sorry, that was a typo. The error was if the base path was too long. 

> I want to add the HDE profile to the superbuild, but unfortunately I have the same problem;, I cannot even build the core profile of the superbuild in windows.
CC @DanielePucci @traversaro

That is really unfortunate. What is the length of your base path (i.e. the length of the path of the directory in which you cloned the robotology-superbuild)? Can you try to clone the superbuild in `C:/robotology-superbuild` as a temporary workaround?

traversaro, 2019-07-30:
@kouroshD did you tried removing the 260-char limit with the strategy described in the article linked by @Nicogene ?

traversaro, 2019-07-30:
You can also try to check Visual Studio specific docs such as https://social.msdn.microsoft.com/Forums/office/en-US/e3517a4d-5e30-4fce-80cf-4b5e9b541a12/how-to-enable-long-file-paths-under-visual-studio-2017-enterprise-in-c-projects-running-on-windows?forum=visualstudiogeneral .

Nicogene, 2019-07-31:
> @kouroshD did you tried removing the 260-char limit with the strategy described in the article linked by @Nicogene ? 

I don't know if I remember well but this worked on VS2019 but **not** VS2017

kouroshD, 2019-07-31:
> Sorry, that was a typo. The error was if the base path was too long.
>
>
> That is really unfortunate. What is the length of your base path (i.e. the length of the path of the directory in which you [...]

This is the path in which I am cloning the superbuild ` C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild`. I prefer to not clone it in `C:\` since in this laptop there are other users.

kouroshD, 2019-07-31:
> @kouroshD did you tried removing the 260-char limit with the strategy described in the article linked by @Nicogene ?

I will try to follow the steps mentioned in Registry edittor in the article @Nicogene  mentioned in the first comment. Otherwise, I will try to add the manifest as it is mentioned in the article you have posted.

kouroshD, 2019-07-31:
I can confirm that, the solution of the article in the first comment did not work for me as well.

Nicogene, 2019-07-31:
Do you have VS2017?

kouroshD, 2019-07-31:
Yes.

traversaro, 2019-07-31:
> This is the path in which I am cloning the superbuild C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild. I prefer to not clone it in C:\ since in this laptop there are other users.

What is the precise error? This base prefix in your case is 52 char plus the size of the name of the build directory. The complete error should tell you how many char you need to save to get the compilation to work. Beside cloning in `C:\`, other strategy you can use is to clone directly in `Kouro`, and also to clone in a directory with a name shorter then `robotology-superbuild`, something like: 
~~~
git clone https://github.com/robotology/robotology-superbuild rob-sb
~~~
You can also use a shorter directory name for the build, i.e. `bld`.

traversaro, 2019-07-31:
For reference, this is the warning that I think it could make sense to have also for the superbuild: 
https://github.com/microsoft/vcpkg/blob/2019.06/ports/qt5-base/portfile.cmake#L3
In our case, we should check the size of `PROJECT_BINARY_DIR`.

kouroshD, 2019-07-31:
This is the complete Error:
```
Build FAILED.

"C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\ICUB.vcxproj" (default target) (3) ->
"C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\YARP.vcxproj" (default target) (4) ->
(CustomBuild target) ->
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets(359,25): error MSB4184: The expression "[MSBuild]::NormalizePath(C:\Users\Kouro\Desktop\icub_ws\robotology-superbu
ild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpmanager, robottestingframework_fixturemanager_yarpmanager.dir\Release\, robottestingframework_fixturemanager_yarpmanager.vcxproj.CopyComplete)" cannot be e
valuated. Path: C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpmanager\robottestingframework_fixturemanager_yarpmanager.dir\Release\robottestingframework
_fixturemanager_yarpmanager.vcxproj.CopyComplete exceeds the OS max path limit. The fully qualified file name must be less than 260 characters. [C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\robotology\YARP\src\robottesting
framework-plugins\fixture-managers\yarpmanager\robottestingframework_fixturemanager_yarpmanager.vcxproj] [C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\YARP.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets(359,25): error MSB4184: The expression "[MSBuild]::NormalizePath(C:\Users\Kouro\Desktop\icub_ws\robotology-superbu
ild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpplugin, robottestingframework_fixturemanager_yarpplugin.dir\Release\, robottestingframework_fixturemanager_yarpplugin.vcxproj.CopyComplete)" cannot be eval
uated. Path: C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpplugin\robottestingframework_fixturemanager_yarpplugin.dir\Release\robottestingframework_fixt
uremanager_yarpplugin.vcxproj.CopyComplete exceeds the OS max path limit. The fully qualified file name must be less than 260 characters. [C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\robotology\YARP\src\robottestingframew
ork-plugins\fixture-managers\yarpplugin\robottestingframework_fixturemanager_yarpplugin.vcxproj] [C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\YARP.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets(359,25): error MSB4184: The expression "[MSBuild]::NormalizePath(C:\Users\Kouro\Desktop\icub_ws\robotology-superbu
ild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpserver, robottestingframework_fixturemanager_yarpserver.dir\Release\, robottestingframework_fixturemanager_yarpserver.vcxproj.CopyComplete)" cannot be eval
uated. Path: C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpserver\robottestingframework_fixturemanager_yarpserver.dir\Release\robottestingframework_fixt
uremanager_yarpserver.vcxproj.CopyComplete exceeds the OS max path limit. The fully qualified file name must be less than 260 characters. [C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\robotology\YARP\src\robottestingframew
ork-plugins\fixture-managers\yarpserver\robottestingframework_fixturemanager_yarpserver.vcxproj] [C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\YARP.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets(359,25): error MSB4184: The expression "[MSBuild]::NormalizePath(C:\Users\Kouro\Desktop\icub_ws\robotology-superbu
ild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpmanager, robottestingframework_fixturemanager_yarpmanager.dir\Release\, robottestingframework_fixturemanager_yarpmanager.vcxproj.CopyComplete)" cannot be e
valuated. Path: C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpmanager\robottestingframework_fixturemanager_yarpmanager.dir\Release\robottestingframework
_fixturemanager_yarpmanager.vcxproj.CopyComplete exceeds the OS max path limit. The fully qualified file name must be less than 260 characters. [C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\robotology\YARP\src\robottesting
framework-plugins\fixture-managers\yarpmanager\robottestingframework_fixturemanager_yarpmanager.vcxproj] [C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\YARP.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets(359,25): error MSB4184: The expression "[MSBuild]::NormalizePath(C:\Users\Kouro\Desktop\icub_ws\robotology-superbu
ild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpplugin, robottestingframework_fixturemanager_yarpplugin.dir\Release\, robottestingframework_fixturemanager_yarpplugin.vcxproj.CopyComplete)" cannot be eval
[... cut ...]
uated. Path: C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpserver\robottestingframework_fixturemanager_yarpserver.dir\Release\robottestingframework_fixt
uremanager_yarpserver.vcxproj.CopyComplete exceeds the OS max path limit. The fully qualified file name must be less than 260 characters. [C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\robotology\YARP\src\robottestingframew
ork-plugins\fixture-managers\yarpserver\robottestingframework_fixturemanager_yarpserver.vcxproj] [C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\YARP.vcxproj]

    0 Warning(s)
    6 Error(s)
```

traversaro, 2019-07-31:
The actual lenghts seems to be 266, while the max is 260. Probably you can just try to rename the robotology-superbuild folder to rob-sb and check if that is sufficient.

kouroshD, 2019-07-31:
Thank you @traversaro . It worked, however I was wondering to know which path did you evaluated that is 266 charcaters. because I checked this path `C:\Users\Kouro\Desktop\icub_ws\robotology-superbuild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpmanager\robottestingframework_fixturemanager_yarpmanager.dir\Release\robottestingframework_fixturemanager_yarpmanager.vcxproj.CopyComplete`  and it has 268 characters!?

traversaro, 2019-08-01:
@kouroshD not sure how I computed 266, probably I used a different file.

traversaro, 2020-04-26:
This has been mitigated by setting the robot testing component by default to OFF in https://github.com/robotology/robotology-superbuild/pull/326 , but to close it we should at least document the workarounds for this problem in https://github.com/robotology/robotology-superbuild#robot-testing , volunteers for doing this are warmly welcome.

closed 2020-10-12 by traversaro

traversaro, 2024-11-29:
Apparently @antonellopaolino and @LudovicaDanovaro experience a similar problem even without enabling `ROBOTOLOGY_ENABLE_ROBOT_TESTING`, so it is possible that there was a regression on this.