robotology/robotology-superbuild #221: Fail to compile walking-controllers in Dynamics profile when ipopt is OFF

lrapetti, 2019-06-27:
By enabling the dynamics profile, the  `walking-controllers` are installed. However, if `IDynTree` is compiled without `USE_IPOPT` (this flag is automatically enabled/disabled if IPOPT is found in the systems), the compilation of `walking-controllers` raises the following errors because it requires some `IDynTree` classes that are compiled only with IPOPT:
```
/home/icub/Software/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/include/WalkingDCMModelPredictiveController.hpp:21:10: fatal error: iDynTree/ConvexHullHelpers.h: No such file or directory
 #include <iDynTree/ConvexHullHelpers.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
modules/Walking_module/CMakeFiles/WalkingModule.dir/build.make:216: recipe for target 'modules/Walking_module/CMakeFiles/WalkingModule.dir/src/WalkingModule.cpp.o' failed
make[5]: *** [modules/Walking_module/CMakeFiles/WalkingModule.dir/src/WalkingModule.cpp.o] Error 1
In file included from /home/icub/Software/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/include/WalkingModule.hpp:28:0,
                 from /home/icub/Software/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/src/main.cpp:14:
/home/icub/Software/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/include/WalkingDCMModelPredictiveController.hpp:21:10: fatal error: iDynTree/ConvexHullHelpers.h: No such file or directory
 #include <iDynTree/ConvexHullHelpers.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
modules/Walking_module/CMakeFiles/WalkingModule.dir/build.make:72: recipe for target 'modules/Walking_module/CMakeFiles/WalkingModule.dir/src/main.cpp.o' failed
make[5]: *** [modules/Walking_module/CMakeFiles/WalkingModule.dir/src/main.cpp.o] Error 1
[ 95%] Completed 'codyco-modules'
In file included from /home/icub/Software/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/src/WalkingInverseKinematics.cpp:29:0:
/home/icub/Software/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/include/WalkingInverseKinematics.hpp:15:10: fatal error: iDynTree/InverseKinematics.h: No such file or directory
 #include <iDynTree/InverseKinematics.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
modules/Walking_module/CMakeFiles/WalkingModule.dir/build.make:240: recipe for target 'modules/Walking_module/CMakeFiles/WalkingModule.dir/src/WalkingInverseKinematics.cpp.o' failed
make[5]: *** [modules/Walking_module/CMakeFiles/WalkingModule.dir/src/WalkingInverseKinematics.cpp.o] Error 1
[ 97%] Built target codyco-modules
CMakeFiles/Makefile2:199: recipe for target 'modules/Walking_module/CMakeFiles/WalkingModule.dir/all' failed
make[4]: *** [modules/Walking_module/CMakeFiles/WalkingModule.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/walking-controllers.dir/build.make:112: recipe for target 'robotology/walking-controllers/CMakeFiles/YCMStamp/walking-controllers-build' failed
make[2]: *** [robotology/walking-controllers/CMakeFiles/YCMStamp/walking-controllers-build] Error 2
CMakeFiles/Makefile2:1331: recipe for target 'CMakeFiles/walking-controllers.dir/all' failed
make[1]: *** [CMakeFiles/walking-controllers.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

A possible solution would be to install `walking-controllers` only with USE_IPOPT ON, or raise an error/warning when it is OFF.

lrapetti, 2019-06-27:
CC @kouroshD

traversaro, 2019-06-27:
> However, if IDynTree is compiled without USE_IPOPT (this flag is automatically enabled/disabled if IPOPT is found in the systems)

I guess given that `ipopt` is a documented dependency of the superbuild, we should just enforce that option in the `BuildiDynTree.cmake` script. If we enforce externally it to ON and ipopt is not found, iDynTree CMake's configuration will fail with a clear message.

closed 2019-07-01 by traversaro