robotology/robotology-superbuild #223: Cannot configure robots-configuration if YARP_ROBOT_NAME is set for Gazebo robots

diegoferigo, 2019-06-27:
If I have for instance an environment variable `YARP_ROBOT_NAME=iCubGazeboV2_5` on a clean system, the superbuild fails with the following error:

```
[ 55%] Performing configure step for 'robots-configuration'
Not searching for unused variables given on the command line.
loading initial cache file /iit/sources/robotology-superbuild/build/robotology/robots-configuration/CMakeFiles/YCMTmp/robots-configuration-cache-Release.cmake
-- Found YARP: /iit/local/lib/cmake/YARP (found version "3.2.0")
-- Found YARP: /iit/local/lib/cmake/YARP (found version "3.2.0")
CMake Deprecation Warning at /iit/local/lib/cmake/YARP/YARPConfig.cmake:295 (message):
  The YARP_MODULE_PATH variable is deprecated.  CMake find package modules
  are now in YCM.
Call Stack (most recent call first):
  CMakeLists.txt:9999 (_yarp_module_path_is_deprecated)
  CMakeLists.txt:11 (list)

-- Libraries are placed in /iit/sources/robotology-superbuild/build/robotology/robots-configuration/lib
-- Executables are placed in /iit/sources/robotology-superbuild/build/robotology/robots-configuration/bin
CMake Warning at /iit/local/share/ICUBcontrib/cmake/ICUBcontribHelpers.cmake:16 (message):
  Installation prefix is different from the ICUBcontrib one, which is
  /iit/local.
Call Stack (most recent call first):
  CMakeLists.txt:22 (icubcontrib_set_default_prefix)

CMake Warning at CMakeLists.txt:75 (message):
  ROBOT_NAME is set to iCubGazeboV2_5 but the
  /iit/sources/robotology-superbuild/robotology/robots-configuration/iCubGazeboV2_5
  directory does not exist
```

The reason is that gazebo robots do not have any configuration in the robots-configuration repository, and the following logic tries to wrongly bypass the problem:

https://github.com/robotology/robotology-superbuild/blob/518c51f28281ab9bad82e9147adce8f8822c29f0/cmake/Buildrobots-configuration.cmake#L11-L17

cc @traversaro 

Asking around also @lrapetti and @kouroshD have been affected in the past days.

diegoferigo, 2019-06-27:
Temporarily solved by configuring a fresh build with:

```
YARP_ROBOT_NAME= cmake [other options]
```

GiulioRomualdi, 2019-07-16:
otherwise, we can pass the option `INSTALL_ALL_ROBOTS=ON` from the superbuild

GiulioRomualdi, 2019-07-16:
cc @nunoguedelha @kouroshD @fjandrad

CarlottaSartore, 2019-12-16:
Any news about this issue? @diegoferigo @GiulioRomualdi ? 
Thanks !

fjandrad, 2020-01-15:
> otherwise, we can pass the option INSTALL_ALL_ROBOTS=ON from the superbuild

I like that idea. Is it an existing option? I did not see it.

traversaro, 2020-01-15:
> 
> 
> > otherwise, we can pass the option INSTALL_ALL_ROBOTS=ON from the superbuild
> 
> I like that idea. Is it an existing option? I did not see it.

Yes, we already pass it when `YARP_ROBOT_NAME` is not defined, see https://github.com/robotology/robotology-superbuild/blob/master/cmake/Buildrobots-configuration.cmake#L15 . I preferred to try to avoid to install all the robots conf files on systems like the old pc104, but it probably make sense to just remove the `if("$ENV{YARP_ROBOT_NAME}" STREQUAL "")` to solve this issue.

closed 2020-02-16 by traversaro