robotology/robotology-superbuild #225: Add an option to use the DYNAMICS profile with Matlab but without yarp-matlab-bindings

gabrielenava, 2019-07-03:
Indeed, I don't think all the `ROBOTOLOGY_ENABLE_DYNAMICS` + `ROBOTOLOGY_USES_MATLAB` users also need to download and compile `yarp-matlab-bindings`. Given that the repo is naturally broken almost every time a new version of Yarp is released - See e.g.:

https://github.com/robotology-playground/yarp-matlab-bindings/issues/38
https://github.com/robotology-playground/yarp-matlab-bindings/issues/23
https://github.com/robotology-playground/yarp-matlab-bindings/issues/15
https://github.com/robotology-playground/yarp-matlab-bindings/issues/14
https://github.com/robotology-playground/yarp-matlab-bindings/issues/12
https://github.com/robotology-playground/yarp-matlab-bindings/issues/9
https://github.com/robotology-playground/yarp-matlab-bindings/issues/17

I think it makes sense to have an option for disabling it in the superbuild when not needed. Maybe the option exists already, if so I was not able to find it.

traversaro, 2019-08-29:
I would strongly prefer to avoid switches to enable compilation per-project, as the resulting amount of possible combinations would become enormous and difficult to test/maintain. I think most of the problem you mention and link actually result from the lack of the automatic testing of the Octave option, that made `yarp-matlab-bindings` compilation failures hidden. This has been addressed with https://github.com/robotology/robotology-superbuild/pull/234 that adds a nightly job that check compilation with Octave, so now as soon as yarp-matlab-bindings is broken we will discover it without the need for someone to manually compile it.

traversaro, 2019-08-29:
Automatically updating the `yarp-matlab-bindings` as suggested in https://github.com/robotology-playground/yarp-matlab-bindings/issues/13 is an even better solution, but it definitely takes more work.

traversaro, 2019-08-29:
See https://github.com/robotology/codyco-superbuild/issues/65 for a related historical issues on per-project options.

traversaro, 2019-09-06:
After the merge of https://github.com/robotology/robotology-superbuild/pull/234, the successful compilation of `yarp-matlab-bindings` is tested every night and at every PR, preventing the usual problem "Gabri will notice that is not compiling when he updates the superbuild". Do you think this is sufficient for closing this issue @gabrielenava ?

gabrielenava, 2019-09-06:
I still have to perform my monthly "complete removal of the superbuild from my laptop and re-installation from scratch", but I think we can close it and if there will be problems I will open a new issue in the future.

closed 2019-09-06 by gabrielenava