robotology/robotology-superbuild #226: WBToolbox not building in robotology-superbuild on Windows

RAR1989, 2019-07-08:
I tried to build the robotology-superbuild on Windows starting from a clean build, however it always failed. I got the error below:
```
 Build files have been written to: C:/Users/rromeo/devel/robotology-superbuild/build/robotology/WBToolbox
8>Performing build step for 'WBToolbox'
8>Microsoft (R) Build Engine version 16.1.76+g14b0a930a7 for .NET Framework
8>Copyright (C) Microsoft Corporation. All rights reserved.
8>
8>  ClockRpc.vcxproj -> C:\Users\rromeo\devel\robotology-superbuild\build\robotology\WBToolbox\lib\Release\ClockRpc.lib
8>  WBToolboxBase.vcxproj -> C:\Users\rromeo\devel\robotology-superbuild\build\robotology\WBToolbox\lib\Release\WBToolboxBase.lib
8>  Auto build dll exports
8>     Creating library C:/Users/rromeo/devel/robotology-superbuild/build/robotology/WBToolbox/lib/Release/WBToolbox.lib and object C:/Users/rromeo/devel/robotology-superbuild/build/robotology/WBToolbox/lib/Release/WBToolbox.exp
8>SimulatorSynchronizer.obj : error LNK2019: unresolved external symbol "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const blockfactory::core::BlockOptionPrioritizeOrder" (?BlockOptionPrioritizeOrder@core@blockfactory@@3V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@B) referenced in function "public: virtual class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > __cdecl wbt::block::SimulatorSynchronizer::additionalBlockOptions(void)" (?additionalBlockOptions@SimulatorSynchronizer@block@wbt@@UEAA?AV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@XZ) [C:\Users\rromeo\devel\robotology-superbuild\build\robotology\WBToolbox\toolbox\library\WBToolbox.vcxproj]
8>C:\Users\rromeo\devel\robotology-superbuild\build\robotology\WBToolbox\bin\Release\WBToolbox.dll : fatal error LNK1120: 1 unresolved externals [C:\Users\rromeo\devel\robotology-superbuild\build\robotology\WBToolbox\toolbox\library\WBToolbox.vcxproj]
8>Done building project "WBToolbox.vcxproj" -- FAILED.
```

cc @traversaro @diegoferigo @nunoguedelha @isorrentino

diegoferigo, 2019-07-08:
It is a problem about global variables in Windows using VS compiler. We already encountered it in the past I think with @bemilio https://github.com/robotology/blockfactory/issues/44. The fix however still didn't go upstream.

isorrentino, 2019-07-09:
@diegoferigo, @RAR1989 and I tried the solutions suggested in the issue you linked but they did not work. 

cc @drdanz

diegoferigo, 2019-07-09:
If @bemilio has still access to the PC where we fixed it, maybe you can check what we did. I don't remember exactly the patch I applied.

bemilio, 2019-07-09:
I remember we finally managed to complete the installation. The only thing not working was the WBToolbox-MATLAB integration.

I am not sure what we should check on the monster. Maybe the global variable? I remember that in solving this issue I didn't understand much of what was going on and that in the end you and silvio managed to solve it. But I have no idea what the steps were.

drdanz, 2019-07-10:
We managed to build the superbuild with @RAR1989, and we found a few issues in several projects.

1) `blockfactory` is built with the [`CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS`](https://cmake.org/cmake/help/latest/variable/CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS.html) that as explained in the [`WINDOWS_EXPORT_ALL_SYMBOLS`](https://cmake.org/cmake/help/latest/prop_tgt/WINDOWS_EXPORT_ALL_SYMBOLS.html) property documentation requires to use `__declspec(dllimport)` in order to import global symbols from the dll (in this case the symbol is `blockfactory::core::BlockOptionPrioritizeOrder`). We fixed this by changing in `blockfactory/sources/Core/include/BlockFactory/Core/BlockInformation.h` (this is obviously not a good fix because it won't work on linux)

```diff
+extern const std::string BlockOptionPrioritizeOrder;
-extern __declspec(dllimport) const std::string BlockOptionPrioritizeOrder;
```

2) `qpOases` and `blockfactory` have a broken `FindMatlab.cmake` file, applying the fix proposed in the [upstream issue](https://gitlab.kitware.com/cmake/cmake/issues/19382) on all the files worked.

3) the `yarp-matlab-bindings` has 2 files `Vocab.m` and `VOCAB.m` in the same directory. As a consequence the `update` step fails. In order to fix this we disabled the `update` step by enabling the `DEVEL_MODE` for the project. I cannot guarantee that this will not cause other issues.

diegoferigo, 2019-07-10:
Thanks @drdanz for taking care of that. I'll take care of blockfactory as soon as I'll be back to IIT and have access to a Windows machine.

traversaro, 2019-11-14:
> the yarp-matlab-bindings has 2 files Vocab.m and VOCAB.m in the same directory. As a consequence the update step fails. In order to fix this we disabled the update step by enabling the DEVEL_MODE for the project. I cannot guarantee that this will not cause other issues.

This was fixed in https://github.com/robotology/yarp-matlab-bindings/pull/43 .

traversaro, 2020-02-18:
> 1. `blockfactory` is built with the [`CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS`](https://cmake.org/cmake/help/latest/variable/CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS.html) that as explained in the [...]

This was fixed by https://github.com/robotology/blockfactory/pull/54 .

closed 2020-02-18 by traversaro