robotology/robotology-superbuild #227: Add Release versions for superbuild

kouroshD, 2019-07-11:
When there are new updates on the core repositories of the superbuild, e.g. YARP, RobotTestingFramework, ICUB, etc , normally there are unseen bugs to fix (build/run time problems). Also, other repositories are based on the mentioned ones, and therefore they might be affected.
I am suggesting here to add releases for superbuild, so that when there are updates like recent updates on YARP, RTF, etc. , we can go to a release of the superbuild, and we can install the packages with Tags (maybe) different from `HEAD`. 
For example, now HDE works with the YARP version `v3.1.1` not with the new version, but then this version is not compatible with new versiona of RTF, ... .
This suggestion may increase the time for fixing the bugs of new released repositories, but on the other side, it does not stop others from developments.

diegoferigo, 2019-07-12:
The support of defining releases has been introduced in https://github.com/robotology/robotology-superbuild/pull/100. Users can create a new tag file (potentially in a new branch of this project if it has to be shared with other users) waiting all upstream post-release fixes.

The current status unfortunately is a corner case since we changed few things in the superbuild structure (RTF as you wrote).

I can think of a per-profile tags (e.g. andy), but handling release constraints of multiple profiles that might be conflicting is not easy.

kouroshD, 2019-07-15:
Making a new branch will work as you mention, but defining releases in a new project may not always work, e.g., our current situation and superbuild should be in previous commits. Well, I am not sure if having multiple branches is a good response to this problem.

The profiles defined in superbuild (especially `core` and `dynamics`) they should be compatible to each other, and they should provide functionality the user is seeking.

diegoferigo, 2019-07-15:
Having multiple branches is not the right solution, but it might address the corner case we encountered this case (that is the impossibility to roll back to old releases by just changing the project TAG). Generally, it should not be required.

> I am suggesting here to add releases for superbuild, so that when there are updates like recent updates on YARP, RTF, etc. , we can go to a release of the superbuild, and we can install the packages with Tags (maybe) different from HEAD.

What you wrote in the first post (use superbuild releases) would not solve it because for most projects we do not check-out specific commits. This is clear in the [devel case](https://github.com/robotology/robotology-superbuild/blob/master/cmake/ProjectsTagsDevel.cmake). We point to branches, not commits.

Using releases would mean checking out the repository to a specific point in history. Referring to the RTF change we introduced, rolling back would still checkout the `master` branch of both yarp and rtf.

What you mean is having something comparable to vcpkg where as far as I know rolling back to a specific point in history also installs old versions of the packages that it contains. We can achieve it quite simply also in our superbuild by fixing commits instead of branches in `ProjectTags*.cmake`, introducing the trade off between having this new feature and maintenance effort (updating the tags of all our projects every time we need to bump). Let's wait @traversaro return to discuss it further.

diegoferigo, 2019-07-15:
Actually what it would be nice to have and it can be an intermediate solution, is having a new `ProjectTagsRelease.cmake` that checks out _released_ versions of our packages. In this way reverting to an old commit and using this tag file would work.

kouroshD, 2019-07-15:
> Actually what it would be nice to have and it can be an intermediate solution, is having a new `ProjectTagsRelease.cmake` that checks out _released_ versions of our packages. In this way reverting to an old commit and using this tag file would work.

Actually, what I meant was for superbuild we can go to a release version, and then we have a specific `ProjectTags*.cmake` file, which we can install stable release versions of other packages. I like your idea to have `ProjectTagsRelease.cmake`. I think we have converged to the same point if I understood correctly what you are saying.
In any case, we will discuss it later with @traversaro .

traversaro, 2019-10-19:
Does the options added in https://github.com/robotology/robotology-superbuild/pull/275 help w.r.t. to what is discussed in this issue? The basic reasoning is that the  specific combination of releases of the packages of the superbuild will always be demo specific, so it makes more sense to just version it somewhere in which the people that manage the demo have access.

diegoferigo, 2019-10-20:
Definitely, thanks @traversaro for taking care of it. Before, the usage of `ProjectTags` was not exposed to the users and not documented. The pathological case described in this issue would not be solved anyway (and I don't think it can be solved easily), though generally #275 provides a sound solution. Luckily changes like the ones that affected RTF are rather uncommon.

traversaro, 2020-02-20:
> The pathological case described in this issue would not be solved anyway (and I don't think it can be solved easily), though generally #275 provides a sound solution. Luckily changes like the ones that affected RTF are rather uncommon.

Yes, a lot of metadata is still used in the superbuild (dependencies, CMake options, etc, etc). The only proper way of dealing with this for having reproducible build is to to use a given project tags file, together with a given tag of the superbuild.

traversaro, 2020-02-26:
Fixed by https://github.com/robotology/robotology-superbuild/pull/350 and https://github.com/robotology/robotology-superbuild/releases/tag/v2020.02 .

closed 2020-02-26 by traversaro