robotology/robotology-superbuild #228: Installation fails with Mojave MAC os

claudia-lat, 2019-07-11:
I tried to build the latest version od robotology-superbuild on the new Mojave MAC os. It fails with the following error:

```bash
-- Found YARP: /Users/claudialatella/src/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.2.0")
-- YARP is version: 3.2.0
CMake Error at CMakeLists.txt:23 (message):
  YARP was found, but no libYARP_math was detected, please recompile yarp

```

cc @traversaro @kouroshD @lrapetti

kouroshD, 2019-07-12:
I tried tonight in my machine (macOS-Mojave) to build superbuild core and dynamic profiles. I could build them without any error.

diegoferigo, 2019-07-13:
@claudia-lat Did you use a clean `build` folder? The default flag to enable the math library changed with YARP 3.2, and I suspect you're experiencing a problem related to the cmake cache.

kouroshD, 2019-07-15:
As far as I know the `build` folder was clean.

claudia-lat, 2019-07-16:
Yes, the build folder was clean.

lrapetti, 2019-07-16:
I think it can be relevant the fact that the computer was updated from a fairly version of the OS (`Mountain Lion`) where all the dependencies were installed with an older version of `homebrew` and where not up-to-date. So, `homebrew` together with the dependencies were updated after moving to `Mojave`, but the process was not really clean I think.

diegoferigo, 2019-07-16:
I have always struggled with such big updates (2012 -> 2019?). In these cases from my experience (if possible) wiping everything can prevent many future problems of leftovers in the OS.

traversaro, 2019-07-28:
That specific error should go away by manually enabling the option `YARP_COMPILE_libYARP_math` in the YARP build.

claudia-lat, 2019-07-29:
I've just installed from scratch (and in a brand new Mojave MAC laptop) the robotology-superbuild. I've compiled it with `ROBOTOLOGY_USES_MATLAB` OFF. It worked.  The error I had in the previous installation didn't appear anymore.

traversaro, 2019-07-29:
So I guess we can  close the issue?

claudia-lat, 2019-07-30:
> So I guess we can close the issue?

Yes. Closing it!

closed 2019-07-30 by claudia-lat