robotology/robotology-superbuild #232: The iCub simulator will not start..

smwasif, 2019-08-12:
Hi dear,
              After robotology-superbuild installation i started yarpserver which is working fine, however when I tried to run iCub_SIM I got following error.

iCub_SIM
[DEBUG]
ODE configuration: ODE ODE_EXT_trimesh ODE_EXT_opcode ODE_OPC_new_collider ODE_EXT_threading ODE_double_precision

[DEBUG]SimConfig::configure
||| clearing context
||| adding context [simConfig]
||| configuring
||| default config file specified as simulator.ini
||| checking [/home/wasif/simulator.ini] (pwd)
||| checking [/home/wasif/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/wasif/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/wasif/opt/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/plugins/robots/default] (robot YARP_DATA_DIRS)
||| checking [config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/wasif/opt/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/plugins/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/wasif/.config/yarp/contexts/simConfig] (context YARP_CONFIG_HOME)
||| checking [/home/wasif/.local/share/yarp/contexts/simConfig] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/home/wasif/opt/yarp/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/plugins/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/home/wasif/.config/yarp/simulator.ini] (YARP_CONFIG_HOME)
||| checking [/home/wasif/.local/share/yarp/simulator.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/simulator.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/simulator.ini] (YARP_CONFIG_DIRS)
||| checking [simulator.ini] (YARP_DATA_DIRS)
||| checking [/home/wasif/opt/yarp/simulator.ini] (YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/simulator.ini] (YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/plugins/simulator.ini] (YARP_DATA_DIRS)
||| did not find simulator.ini
yarp: cannot read from simulator.ini
[INFO]default module name:  /icubSim 
[INFO]default verbosity level:  0 
||| finding file []
||| checking [/home/wasif/] (pwd)
||| found /home/wasif/
||| finding file [parts]
||| checking [/home/wasif/parts] (pwd)
||| checking [/home/wasif/.config/yarp/parts] (YARP_CONFIG_HOME)
||| checking [/home/wasif/.local/share/yarp/parts] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/parts] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/parts] (YARP_CONFIG_DIRS)
||| checking [parts] (YARP_DATA_DIRS)
||| checking [/home/wasif/opt/yarp/parts] (YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/parts] (YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/plugins/parts] (YARP_DATA_DIRS)
||| did not find parts
yarp: cannot read from 
||| finding file [parts]
||| did not find parts
||| finding file [general]
||| checking [/home/wasif/general] (pwd)
||| checking [/home/wasif/.config/yarp/general] (YARP_CONFIG_HOME)
||| checking [/home/wasif/.local/share/yarp/general] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/general] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/general] (YARP_CONFIG_DIRS)
||| checking [general] (YARP_DATA_DIRS)
||| checking [/home/wasif/opt/yarp/general] (YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/general] (YARP_DATA_DIRS)
||| checking [/home/wasif/opt/icub/share/iCub/plugins/general] (YARP_DATA_DIRS)
||| did not find general
yarp: cannot read from 
[ERROR]

There seems to be a conflict with the  actElevation  configuration file 
[ERROR]Check the iCub_parts_activation.ini under $ICUB_ROOT/app/simConfig  
[ERROR]If it still does not work (update svn) and re-install the app folder

 
[ERROR]The iCub simulator will not start.

Please help to resolve this issue.

traversaro, 2019-08-12:
Hi @smwasif, 
   which operating systems are you using? If you are using Linux, are you manually defining the env variables or are you using the `setup.sh` script?

smwasif, 2019-08-12:
I am using ubuntu 18.04 and setting environment variables manually using following commands.
>> export ROBOTOLOGY_SUPERBUILD_SOURCE_DIR=/home/wasif/robotology-superbuild/

>> export ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX=$ROBOTOLOGY_SUPERBUILD_SOURCE_DIR/build/install

>> export PATH=$PATH:$ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX/bin

>> export YARP_DATA_DIRS=$YARP_DATA_DIRS:$ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX/share/yarp:$ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX/share/iCub:$ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX/share/ICUBcontrib:$ROBOTOLOGY_SUPERBUILD_SOURCE_DIR/robotology/icub-tests/suits

>> export CMAKE_PREFIX_PATH=$CMAKE_PREFIX_PATH:$ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX

>> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX/lib 

>> export BLOCKFACTORY_PLUGIN_PATH=$ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX/lib/blockfactory

traversaro, 2019-08-13:
To copy&paste code or other text that should be rendered as code, you can use multiple lines surrounded by `~~~`, see for example:
~~~
this 
~~~

That is strange, looking from your error message, it seems that the `YARP_DATA_DIRS` that you use contain the directory `/home/wasif/opt/yarp` and `/home/wasif/opt/icub/share/iCub`, and not the directory that you listed. To verify this, can you type the `env` command in the exact terminal in which you launch the `iCub_SIM` command? Note that, even if you modified your `.bashrc`, you need to open a new terminal to propagate the changes in the `.bashrc` to the terminal that you use.

smwasif, 2019-08-13:
Dear @traversaro 
                                You are right after setting environment variables in same terminal which I used to launch  `iCub_SIM`, it started normally and it is working.  Then i manually edited the `.bashrc `to set these environment values and remove previous values. Thank you very much for your kind support.

closed 2019-08-13 by smwasif