robotology/robotology-superbuild #237: Fix compilation of ROBOLOGY_ENABLE_ICUB_HEAD  profile option on macOs

traversaro, 2019-08-22:
At the moment, the ROBOLOGY_ENABLE_ICUB_HEAD option fails on macOs with a dragonfly related error, see https://github.com/robotology/robotology-superbuild/pull/234/checks?check_run_id=200516475 .

traversaro, 2019-08-22:
Probably it is sufficient to add  https://brewformulas.org/Libdc1394 as a dependency.

traversaro, 2019-08-23:
cc @kouroshD did you experience any problem like that?

kouroshD, 2019-08-25:
No, it was fine for me. I did not have any problem.

traversaro, 2019-09-27:
cc @nunoguedelha

nunoguedelha, 2019-09-27:
@traversaro , so both the `superbuild` and `icub-main` Readmes should be updated right?

traversaro, 2019-09-27:
Exactly.

nunoguedelha, 2019-10-17:
I think we can also close this issue for now, considering https://github.com/robotology/robotology-superbuild/pull/268.

traversaro, 2019-11-08:
As the option is now tested in macOS as part of the CI with https://github.com/robotology/robotology-superbuild/pull/276, I agree that we can close the issue.

closed 2019-11-08 by traversaro