robotology/robotology-superbuild #238: Fix compilation of ROBOLOTOLOGY_USES_GAZEBO option on GitHub Actions in Ubuntu 18.04

traversaro, 2019-08-23:
The compilation of ROBOTOLOGY_USE_GAZEBO on the Ubuntu 18.04 used in GitHub Actions fails with the following error: 
~~~
CMake Error at /usr/local/share/cmake-3.12/Modules/FindBoost.cmake:2048 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.69.0

  Boost include path: /usr/local/share/boost/1.69.0/include

  Could not find the following Boost libraries:

          boost_signals

  Some (but not all) of the required Boost libraries were found.  You may
  need to install these additional Boost libraries.  Alternatively, set
  BOOST_LIBRARYDIR to the directory containing Boost libraries or BOOST_ROOT
  to the location of Boost.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:159 (find_package)
  cmake/BuildGazeboYARPPlugins.cmake:31 (find_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:235 (include)
  CMakeLists.txt:122 (find_or_build_package)
~~~
the reason is probably that Gazebo binaries for Ubuntu 18.04 have been compiled with the system boost library (version 1.65, see https://packages.ubuntu.com/bionic/libboost-all-dev) but for some reason the GitHub Actions `ubuntu-18.04` contains instead Boost 1.69 (see last line in https://help.github.com/en/articles/software-in-virtual-environments-for-github-actions#ubuntu-1804-lts). Another source of problems could be that in that image there is  CMake 3.12, while the default Ubuntu 18.04 version is CMake 3.10

traversaro, 2019-08-23:
Cc @j-rivero, you may encounter this as well if you start using GitHub actions.

traversaro, 2019-10-17:
As verified in https://github.com/robotology/robotology-superbuild/pull/241 , this was probably fixed by some changes on the GitHub Actions images.

closed 2019-10-17 by traversaro

j-rivero, 2019-10-17:
> cc @j-rivero, you may encounter this as well if you start using GitHub actions.

Thanks Silvio. If I understand correctly the change in the boost version provided by default in 18.04 has been reverted. Not the best idea to provide modified Ubuntu images for CI testing IMHO.