robotology/robotology-superbuild #240: Move yarp-matlab-bindings under CORE profile

kouroshD, 2019-08-28:
Currently,`yarp-matlab-bindings` option is under `Dynamics` profile[here](https://github.com/robotology/robotology-superbuild/blob/7b4c456058277e27d82c1bd14426dd442082cc95/CMakeLists.txt#L141-L143), since it has been used mostly by people who use `Dynamics` profile. Since, some users may want to use this option but not necessarily `Dynamics` repositories, I suggest either we move it under `CORE` profile [check here](https://github.com/robotology/robotology-superbuild/blob/7b4c456058277e27d82c1bd14426dd442082cc95/cmake/Buildyarp-matlab-bindings.cmake#L14), or we make it as `cmake_dependent_option ` in superbuild.

cc @fjandrad @traversaro

diegoferigo, 2019-08-29:
:+1: Yarp is in `CORE`, if Matlab is found in the system I don't see any reason why the bindings should not be in core too.

fjandrad, 2019-09-06:
+1 for core xD

traversaro, 2019-10-25:
For anyone interested: feel free to open a PR to change the README and the CMake files.

traversaro, 2019-11-13:
Fixed by @kouroshD  in https://github.com/robotology/robotology-superbuild/pull/297 .

closed 2019-11-13 by traversaro