robotology/robotology-superbuild #242: Use standard osqp

traversaro, 2019-09-01:
Apparently the robotology-superbuild is currently using a fork of osqp for dealing with some `.gitattributes` problems (see https://github.com/robotology/robotology-superbuild/pull/87/files#diff-fcad91dd1d4b0f9e2cc5f258bfa8b79bR10). As new version of osqp are being released, and sometimes they also contain breaking changes (see https://github.com/robotology/osqp-eigen/issues/31) we should switch to use upstream osqp, optionally on a specific release tag, so that we can update to use new osqp version by just changing the tag.

traversaro, 2019-09-01:
The original issue was discussed in https://github.com/robotology/robotology-superbuild/pull/87#issuecomment-458987172 and the following comments. Dropping Ubuntu 16.04 should be sufficient to solve the problem.

traversaro, 2019-09-01:
As of  https://github.com/robotology/robotology-superbuild/pull/243, the issue seem to be solved, probably due to using the latest ExternalProject module via requiring the latest YCM (see https://github.com/robotology/robotology-superbuild/pull/224 and https://github.com/robotology/robotology-superbuild/pull/199).

traversaro, 2019-09-24:
In https://github.com/robotology/robotology-superbuild/pull/260 we used a custom tag for osqp-eigen with our fork of osqp. To fix this we need to update all the software that depends on osqp-eigen in the superbuild to deal with this new version, and then we can update. 

@GiulioRomualdi let me know if you need my help on this.

GiulioRomualdi, 2020-01-18:
@S-Dafarra check this: https://github.com/robotology/robotology-superbuild/issues/242#issuecomment-526896522 :smile:

traversaro, 2020-01-18:
Just for clarity, I think the incompatibility for which the fork was necessary was just for Ubuntu 16.04 + GitHub Actions, not Ubuntu 16.04 on its own (that in any case we stopped testing in CI).

closed 2020-01-21 by traversaro