robotology/robotology-superbuild #256: small mistake in README.m file (setup environment instructions on Linux system)

valegagge, 2019-09-18:
While I was installing the Superbuild at first time, I noticed a small mistake about setup environment instructions on Linux system.

In order to use the updated .bashrc (with `setup.sh`) it is necessary to add the definition of  `$ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX` variable before including `setup.sh`.

An other solution, is change the suggested command 
` source $ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX/share/robotology-superbuild/setup.sh `
in something like
` /directory/where/you/downloaded/robotology-superbuild/build/install `

Maybe this is only fussiness...

traversaro, 2019-09-18:
Indeed, I think it is a correct suggestion, feel free to suggest a PR.

traversaro, 2019-10-21:
Fixed  in https://github.com/robotology/robotology-superbuild/pull/278 .

valegagge, 2019-10-21:
Can we close the issue?

traversaro, 2019-10-21:
Yes, thanks!

closed 2019-10-21 by traversaro