robotology/robotology-superbuild #263: OsqpEigen compilation problem

traversaro, 2019-09-24:
@Gianlucorso originally posted in https://github.com/robotology/yarp/issues/2099#issuecomment-534607304 : 

Thanks for the quick reply.
Your suggestion helps to solve the cited problem.

However the compilation still fails on `OsqpEigen`.

```
/home/gcorsini/robotology-superbuild/robotology/OsqpEigen/src/Solver.cpp: In member function ‘bool OsqpEigen::Solver::initSolver()’:
/home/gcorsini/robotology-superbuild/robotology/OsqpEigen/src/Solver.cpp:82:44: error: cannot convert ‘OSQPWorkspace**’ to ‘const OSQPData*’ for argument ‘1’ to ‘OSQPWorkspace* osqp_setup(const OSQPData*, OSQPSettings*)’
                   m_settings->getSettings()) != 0 ){
                                            ^
CMakeFiles/OsqpEigen.dir/build.make:110: recipe for target 'CMakeFiles/OsqpEigen.dir/src/Solver.cpp.o' failed
make[5]: *** [CMakeFiles/OsqpEigen.dir/src/Solver.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/OsqpEigen.dir/all' failed
make[4]: *** [CMakeFiles/OsqpEigen.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/OsqpEigen.dir/build.make:107: recipe for target 'robotology/OsqpEigen/CMakeFiles/YCMStamp/OsqpEigen-build' failed
make[2]: *** [robotology/OsqpEigen/CMakeFiles/YCMStamp/OsqpEigen-build] Error 2
CMakeFiles/Makefile2:2278: recipe for target 'CMakeFiles/OsqpEigen.dir/all' failed
make[1]: *** [CMakeFiles/OsqpEigen.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

Should I open a new issue in `OsqpEigen` repository?

Gianlucorso, 2019-09-24:
> (OT, as this is not related to this YARP issue: did you updated to the latest superbuild by doing git pull and then make update-all in the build?

I pulled as you suggested. There was a commit related to Osqp. 
The compilation still fails as before. Shall I clean the build before running `make` command?

traversaro, 2019-09-24:
@Gianlucorso commented:

I have installed [MATMPC](https://github.com/chenyutao36/MATMPC) which already contains a version of osqp and qpoases. 

I leave in attachment the `CMakeCache.txt` of OsqpEigen repository.  
[CMakeCache.txt](https://github.com/robotology/robotology-superbuild/files/3648073/CMakeCache.txt)

traversaro, 2019-09-24:
> I leave in attachment the `CMakeCache.txt` of OsqpEigen repository.  
[CMakeCache.txt](https://github.com/robotology/robotology-superbuild/files/3648073/CMakeCache.txt)

That seems to be using correctly the osqp from the superbuild, so hopefully there are no conflicts with `MATMPC` . Just to make sure  that the include dirs are correct, can you compile the superbuild/osqp with `make VERBOSE=1` and paste the failing command? Thanks! 

P.s. please continue the discussion in the robotology-superbuild issue tracker, thanks!

Gianlucorso, 2019-09-24:
I run `make VERBOSE=1` command only for `OsqpEigen` repository. Below you can find the result.

```
/usr/bin/cmake -H/home/gcorsini/robotology-superbuild/robotology/OsqpEigen -B/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen/CMakeFiles /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
make -f CMakeFiles/OsqpEigen.dir/build.make CMakeFiles/OsqpEigen.dir/depend
make[2]: Entering directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
cd /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/gcorsini/robotology-superbuild/robotology/OsqpEigen /home/gcorsini/robotology-superbuild/robotology/OsqpEigen /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen/CMakeFiles/OsqpEigen.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
make -f CMakeFiles/OsqpEigen.dir/build.make CMakeFiles/OsqpEigen.dir/build
make[2]: Entering directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
[ 25%] Building CXX object CMakeFiles/OsqpEigen.dir/src/Solver.cpp.o
/usr/bin/c++  -DOsqpEigen_EXPORTS -isystem /usr/include/eigen3 -I/home/gcorsini/robotology-superbuild/robotology/OsqpEigen/include -isystem /home/gcorsini/robotology-superbuild/build/install/include/osqp  -O3 -DNDEBUG -fPIC   -std=c++14 -o CMakeFiles/OsqpEigen.dir/src/Solver.cpp.o -c /home/gcorsini/robotology-superbuild/robotology/OsqpEigen/src/Solver.cpp
/home/gcorsini/robotology-superbuild/robotology/OsqpEigen/src/Solver.cpp: In member function ‘bool OsqpEigen::Solver::initSolver()’:
/home/gcorsini/robotology-superbuild/robotology/OsqpEigen/src/Solver.cpp:82:44: error: cannot convert ‘OSQPWorkspace**’ to ‘const OSQPData*’ for argument ‘1’ to ‘OSQPWorkspace* osqp_setup(const OSQPData*, OSQPSettings*)’
                   m_settings->getSettings()) != 0 ){
                                            ^
CMakeFiles/OsqpEigen.dir/build.make:110: recipe for target 'CMakeFiles/OsqpEigen.dir/src/Solver.cpp.o' failed
make[2]: *** [CMakeFiles/OsqpEigen.dir/src/Solver.cpp.o] Error 1
make[2]: Leaving directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/OsqpEigen.dir/all' failed
make[1]: *** [CMakeFiles/OsqpEigen.dir/all] Error 2
make[1]: Leaving directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

traversaro, 2019-09-24:
Thanks @Gianlucorso , the include path seems fine, the issue seems definitly the wrong combination of osqp and osqp-eigen,  @GiulioRomualdi you have any insight in which one is wrong?

gabrielenava, 2019-09-26:
Hi @GiulioRomualdi, any suggestion for this issue?

GiulioRomualdi, 2019-09-26:
Hi @Gianlucorso, can you copy here the outcome of `git status` in 
1. `robotology-superbuild/external/osqp`
2. `robotology-superbuild/robotology/OsqpEigen`

Gianlucorso, 2019-09-26:
Hi, I leave here the outcome of `git status` in the 2 repositories you asked.

1. `robotology-superbuild/external/osqp`

```
gcorsini@gcorsini-XPS-15-9570:~/robotology-superbuild/external/osqp$ git status
On branch fix-lf-problems
Your branch is up to date with 'origin/fix-lf-problems'.

nothing to commit, working tree clean
gcorsini@gcorsini-XPS-15-9570:~/robotology-superbuild/external/osqp$ 
```

2. `robotology-superbuild/robotology/OsqpEigen`

```
gcorsini@gcorsini-XPS-15-9570:~/robotology-superbuild/robotology/OsqpEigen$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
gcorsini@gcorsini-XPS-15-9570:~/robotology-superbuild/robotology/OsqpEigen$
```

Gianlucorso, 2019-09-26:
At the moment me and @gabrielenava we have excluded the compilation of `osqp`, `OsqpEigen`, `UnicyclePlanner` and `WalkingControllers` from the `DYNAMICS` profile, by editing the `CMakeList.txt` of the `robotology-superbuild`. 
This solves the compilation error and this will be only a temporaly fix.

GiulioRomualdi, 2019-09-26:
@Gianlucorso
> 2. `robotology-superbuild/robotology/OsqpEigen`
>
> ```
> gcorsini@gcorsini-XPS-15-9570:~/robotology-superbuild/robotology/OsqpEigen$ git status
> On branch master
> Your branch is up to date with [...]

In order to use the `superbuild`, `osqp-eigen` branch has to be `v0.4.1`

Gianlucorso, 2019-09-26:
Thank you for the suggestion @GiulioRomualdi.
Switching to branch `v0.4.1` in `osqp-eigen` repository solves the issue. Now the compilation reaches the end without any error (before running the compilation, I restored the default options for `DYNAMICS` profile).

GiulioRomualdi, 2019-09-26:
If for you is ok, we can close the issue.

Gianlucorso, 2019-09-26:
Yes you can close. Thanks again

traversaro, 2019-09-27:
Great, probably the superbuild was cloned/last updated before https://github.com/robotology/robotology-superbuild/pull/260 and this was the problem, this should not happen on new clones of the superbuild.

closed 2019-09-27 by traversaro