robotology/robotology-superbuild #264: Make enviroment scripts relocatable

traversaro, 2019-09-24:
In some cases, saving the `install` folder of a superbuild and coping it to a new system is fast and quick way of deploying some binaries. Unfortunately, the `setup.sh`, `setup.bat` and `addPathsToUserEnvVariables.ps1` scripts  are currently hardcoding inside them their installation directory, and so they are not **relocatable**, while ideally they should be relocatable.

To make them relocatable, the scripts should automatically infer the install prefix from their location, and use that as the prefix with which configure the environment. This is typically possible by script-specific variables that provide the absolute location of the script file in the file system, similar to the [`CMAKE_CURRENT_LIST_DIR`](https://cmake.org/cmake/help/v3.10/variable/CMAKE_CURRENT_LIST_DIR.html) in CMake. These variable are listed in the following. 

| Shell  |  Variable | Docs | 
|:------:|:----------:|:-------:|
| Bash |  `dirname realpath $0` |  |
| Cmd | `%~dp0` |  https://stackoverflow.com/questions/3827567/how-to-get-the-path-of-the-batch-script-in-windows | 
| Powershell | `Split-Path -Path $PSCommandPath` | |

traversaro, 2019-10-12:
cc @valegagge @vtikha I do not know if this can be interesting for docker-based deployment.

drdanz, 2019-10-14:
FYI, walkman superbuild bash scripts used to do that, I don't remember exactly how, but we should be able to find them

traversaro, 2020-11-13:
For bash, I used also another line that you can find in https://github.com/robotology/robotology-superbuild-dependencies-vcpkg/blob/master/scripts/setup-deps.sh#L1, not sure which one is the best one.

drdanz, 2020-11-14:
```bash
scriptDirectory=$(dirname $(readlink --canonicalize --no-newline $BASH_SOURCE))
# Equivalent to 
scriptDirectory=$(dirname $(readlink -fn $BASH_SOURCE))
```
vs.
```bash
scriptDirectory="$( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null 2>&1 && pwd )" 
installerRootPath=`realpath $scriptDirectory/..`
```

The first one seems cleaner to me:
1. takes the path of the current bash script, removes symbolic links, and takes the dirname.
2. takes the dirname of the current bash script, then in a subshell changes directory and print current directory.

AFAIK The second version does not remove symbolic links, though, but it might not be that important, since `realpath` in the second command does that, and the `scriptDirectory` does not seem to be used anywhere else.

I ignore the exact difference between `$BASH_SOURCE` and `${BASH_SOURCE[0]}`, but they should be the same, see also https://stackoverflow.com/questions/35006457/

Also `realpath` vs `readlink -f`, According to https://unix.stackexchange.com/questions/136494/ `readlink -f` seems to be available everywhere, while `realpath` might not be available on some distros.

Besides that, they should both work fine, I just had some time to waste :rofl: .

traversaro, 2021-02-04:
Related issue: https://github.com/robotology/robotology-superbuild/issues/608 .

traversaro, 2021-02-10:
Interestingly, it seems that zsh is now the default shell on macOS (see https://support.apple.com/en-us/HT208050) so if we do this change we may need to ensure zsh compatibility.

traversaro, 2021-07-14:
> 
> 
> ```shell
> scriptDirectory=$(dirname $(readlink --canonicalize --no-newline $BASH_SOURCE))
> # Equivalent to 
> scriptDirectory=$(dirname $(readlink -fn $BASH_SOURCE))
> ```
> 
> vs.
> 
> ```shell
> scriptDirectory="$( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null 2>&1 && pwd )" 
> installerRootPath=`realpath $scriptDirectory/..`
> ```
> 
> The first one seems cleaner to me:
> 
>     1. takes the path of the current bash script, removes symbolic links, and takes the dirname.
> 
>     2. takes the dirname of the current bash script, then in a subshell changes directory and print current directory.
> 
> 
> AFAIK The second version does not remove symbolic links, though, but it might not be that important, since `realpath` in the second command does that, and the `scriptDirectory` does not seem to be used anywhere else.
> 
> I ignore the exact difference between `$BASH_SOURCE` and `${BASH_SOURCE[0]}`, but they should be the same, see also https://stackoverflow.com/questions/35006457/
> 
> Also `realpath` vs `readlink -f`, According to https://unix.stackexchange.com/questions/136494/ `readlink -f` seems to be available everywhere, while `realpath` might not be available on some distros.
> 
> Besides that, they should both work fine, I just had some time to waste 🤣 .

I tried the first version verbatim (see https://github.com/robotology/robotology-superbuild/runs/3066395353), and is failing with:
~~~
readlink: missing operand
Try 'readlink --help' for more information.
dirname: missing operand
Try 'dirname --help' for more information.
~~~
Probably the fix is trivial, but for the time being I just used the second version.