robotology/robotology-superbuild #265: YCM build failing with Xcode

nunoguedelha, 2019-09-25:
## Issue description
The YCM build is failing when building the superbuild on Macos using Xcode toolchain.

Steps to reproduce the issue:
- update the `robotology-superbuild` repo to the latest commit on `master` branch today,
- run cmake on an empty `build` folder (`cmake ./ -G Xcode`),
- set to ON the options:
```
ROBOTOLOGY_ENABLE_CORE
ROBOTOLOGY_ENABLE_DYNAMICS
ROBOTOLOGY_ENABLE_ICUB_HEAD
ROBOTOLOGY_USES_GAZEBO
ROBOTOLOGY_USES_MATLAB
```
(This configuration has been used for a long time on the single motor test setup, for starting a set of EMS+2FOC boards driving a single iCub motor).
- a few repos are set with `YCM_EP_DEVEL_MODE` ON.
- But the following devel mode option is OFF:
```
YCM_EP_DEVEL_MODE_YCM
```
- run `xcodebuild -target ALL_UPDATE`,
- run `xcodebuild -target ALL_BUILD`.

The last commands fails.

## Failure log
```
...
PhaseScriptExecution CMake\ Rules build/robotology-superbuild.build/Debug/YCM.build/Script-5053C3E3F7E142C7BBCF2381.sh
    cd /Users/nunoguedelha/dev/robotology-superbuild
    /bin/sh -c /Users/nunoguedelha/dev/robotology-superbuild/build/robotology-superbuild.build/Debug/YCM.build/Script-5053C3E3F7E142C7BBCF2381.sh
echo "Skipping update step for 'YCM'"
Skipping update step for 'YCM'
cd /Users/nunoguedelha/dev/robotology-superbuild/robotology/YCM && /usr/local/Cellar/cmake/3.15.3/bin/cmake -E echo_append
echo "No patch step for 'YCM'"
No patch step for 'YCM'
/usr/local/Cellar/cmake/3.15.3/bin/cmake -E echo_append
/usr/local/Cellar/cmake/3.15.3/bin/cmake -E touch /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YCM/CMakeFiles/YCMStamp/Debug/YCM-patch
echo "Performing configure step for 'YCM'"
Performing configure step for 'YCM'
cd /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YCM && /usr/local/Cellar/cmake/3.15.3/bin/cmake --no-warn-unused-cli -DCMAKE_PREFIX_PATH:PATH=/Users/nunoguedelha/dev/robotology-superbuild/build/install -C/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YCM/CMakeFiles/YCMTmp/YCM-cache-Debug.cmake -GXcode /Users/nunoguedelha/dev/robotology-superbuild/robotology/YCM
Not searching for unused variables given on the command line.
loading initial cache file /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YCM/CMakeFiles/YCMTmp/YCM-cache-Debug.cmake
CMake Error: Error: generator : Xcode
Does not match the generator used previously: Unix Makefiles
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.
make: *** [/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YCM/CMakeFiles/YCMStamp/Debug/YCM-configure] Error 1
Command /bin/sh failed with exit code 2

** BUILD FAILED **
```

nunoguedelha, 2019-09-25:
@drdanz gave a quick look and suspects the following: it seems that the YCM has been compiled a first time with the normal cmake tool, and a second time with the Xcode toolchain, which would result in the logged error.
Actually the superbuild downloads and compiles YCM at the bootstrap phase and then again when building all the other repos.

nunoguedelha, 2019-09-25:
Indeed, a `grep -Rs "CMAKE_GENERATOR:INTERNAL=" robotology/*` in the main `build` folder gives:
```
robotology/BlockFactory/CMakeCache.txt:CMAKE_GENERATOR:INTERNAL=Xcode
robotology/ICUB/CMakeCache.txt:CMAKE_GENERATOR:INTERNAL=Xcode
robotology/RobotTestingFramework/CMakeCache.txt:CMAKE_GENERATOR:INTERNAL=Xcode
robotology/YARP/CMakeCache.txt:CMAKE_GENERATOR:INTERNAL=Xcode
robotology/YCM/CMakeCache.txt:CMAKE_GENERATOR:INTERNAL=Unix Makefiles
robotology/icub-firmware-shared/CMakeCache.txt:CMAKE_GENERATOR:INTERNAL=Xcode
robotology/icub-gazebo/CMakeCache.txt:CMAKE_GENERATOR:INTERNAL=Xcode
robotology/wholeBodyInterface/CMakeCache.txt:CMAKE_GENERATOR:INTERNAL=Xcode
```
Where the line **robotology/YCM/CMakeCache.txt:CMAKE_GENERATOR:INTERNAL=Unix Makefiles** shows that the YCM Cmake files had already been generated with the `Unix Makefiles` generator (by running `cmake` without the `-G Xcode` option).

nunoguedelha, 2019-09-26:
Copying MacOSX users. Hope I didn't forget anyone...

CC @lrapetti @DanielePucci @kouroshD @claudia-lat

traversaro, 2019-09-27:
Solving https://github.com/robotology/robotology-superbuild/issues/267 should prevent this kind of issues from happening.

traversaro, 2019-10-20:
I was able to reproduce this in https://github.com/robotology/robotology-superbuild/pull/276 .

diegoferigo, 2019-10-25:
> I did not mentioned that before, but clearly a workaround is to use a normally installed YCM instead of bootstrapping it.

I use Ninja but my setup also has this configuration, probably this is the reason why this problem never occurred to me :)

traversaro, 2019-11-08:
Hi @nunoguedelha , this will be fixed upstream in https://github.com/robotology/ycm/pull/294 , and I added a temporary workaround i for the robotology-superbuild in https://github.com/robotology/robotology-superbuild/pull/276 , let me know if this solves your problems.

traversaro, 2019-11-08:
As https://github.com/robotology/ycm/issues/290 is tracking the problem on YCM side, if the problem is  solved for you @nunoguedelha feel free to close the issue, thanks!

traversaro, 2019-11-14:
I close as this was fixed in https://github.com/robotology/robotology-superbuild/pull/276 .

closed 2019-12-04 by traversaro