robotology/robotology-superbuild #273: codyco-modules configure step fails on devel branch

nunoguedelha, 2019-10-14:
## Description of the failure

Steps for building the superbuild with option `ROBOTOLOGY_ENABLE_CORE`:
- full update of the `robotology-superbuild` repo (commit `a43defa6c4af0a892001075274e189da6dc06bbb`),
- change `./CMakeLists.txt` for loading `ProjectsTagsDevel` instead of `ProjectsTagsMaster`,
- run `ccmake ../ -G Xcode` and set the default options `ROBOTOLOGY_ENABLE_CORE`, `ROBOTOLOGY_USES_GAZEBO`, configure and generate,
- update: `xcodebuild -verbose -target ALL_UPDATE`,
- build: `xcodebuild -verbose -target ALL_BUILD`. => OK

Further steps for building the superbuild with option `ROBOTOLOGY_ENABLE_DYNAMICS`:
- run `ccmake ../ -G Xcode` and set the option `ROBOTOLOGY_ENABLE_DYNAMICS`, configure and generate,
- update: `xcodebuild -verbose -target ALL_UPDATE`. => OK
- build: `xcodebuild -verbose -target ALL_BUILD`. => **FAIL**

The `codyco-modules` build fails with the error:
```
-- ending plugin library: codycomod
CMake Deprecation Warning at /Users/nunoguedelha/dev/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:242 (message):
  The YARP_INCLUDE_DIRS variable is deprecated
Call Stack (most recent call first):
  src/modules/wholeBodyDynamics3/CMakeLists.txt:9999 (_yarp_deprecated_variable_warning)
  src/modules/wholeBodyDynamics3/CMakeLists.txt:11 (include_directories)

CMake Deprecation Warning at /Users/nunoguedelha/dev/robotology-superbuild/build/install/share/YCM/modules/InstallBasicPackageFiles.cmake:307 (message):
  TARGETS_PROPERTY is deprecated.  Use EXPORT instead
Call Stack (most recent call first):
  CMakeLists.txt:88 (install_basic_package_files)

CMake Warning (dev) at /Users/nunoguedelha/dev/robotology-superbuild/build/install/share/YCM/modules/InstallBasicPackageFiles.cmake:369 (message):
  Compatibility variables are no longer generated.  Use
  ENABLE_COMPATIBILITY_VARS to re-enable them (deprecated) or define them
  using either INCLUDE_FILE or INCLUDE_CONTENT (recommended).
Call Stack (most recent call first):
  CMakeLists.txt:88 (install_basic_package_files)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
CMake Error: CMake can not determine linker language for target: wholeBodyDynamicsSettings
CMake Error: CMake can not determine linker language for target: wholeBodyDynamicsSettings
CMake Error: CMake can not determine linker language for target: wholeBodyDynamicsSettings
CMake Error: CMake can not determine linker language for target: wholeBodyDynamicsSettings
-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
make: *** [/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/codyco-modules/CMakeFiles/YCMStamp/Debug/codyco-modules-configure] Error 1
Command /bin/sh failed with exit code 2

** BUILD FAILED **

The following build commands failed:
	PhaseScriptExecution CMake\ Rules build/robotology-superbuild.build/Debug/codyco-modules-configure.build/Script-BA4D41ACCB53464583FAE660.sh
(1 failure)
```

It actually already fails with the configure step:
```
> xcodebuild -verbose -target codyco-modules-configure
```

Switching to branch `master` in the `codyco-modules` repo, commit `8b09ba072be90c743078ef89f5172578387c7f95`, solves the problem. I could build all the repos.

I've realized that the branch `devel` in `codyco-modules` is very old, and new developments are being pushed to `master`. Is this intentional?

nunoguedelha, 2019-10-15:
CC @GiulioRomualdi

traversaro, 2019-10-15:
> I've realized that the branch `devel` in `codyco-modules` is very old, and new developments are being pushed to `master`. Is this intentional?

Indeed, yes. The `master` branch of `codyco-modules` should work fine with both YARP `master` and `devel`, and that `devel` branch was a leftover. I just updated it to match the `master`, and this should fix the issue. This is another issue that we would have avoided if we had https://github.com/robotology/robotology-superbuild/issues/101 .

nunoguedelha, 2019-10-15:
Thanks. Closing then.

nunoguedelha, 2019-10-16:
Having the issue again on the purple icub-head . Same exact error, after starting the build from scratch.

nunoguedelha, 2019-10-16:
It's kind of blocking and hard to analyse as usual with cmake.

traversaro, 2019-10-20:
@nunoguedelha if I remember correctly, we found a solution on the iCubGenova02 system.

nunoguedelha, 2019-10-20:
Yes absolutely, sorry.
So, for quick recap: the issue was in the Cmake configuration of `wholeBodyDynamicsSettings` module due to a deprecated interface in `Yarp` which was actually broken and not just deprecated. Additionally, generating the files in the source target, instead of the build target was causing build issues.
Fixed by https://github.com/robotology/codyco-modules/pull/282 .
I've added the more detailled explanations.

nunoguedelha, 2019-10-20:
### It's worth mentioning the analysis steps:
#### First issue
1. Initial observed **configure** step error:
`CMake Error: CMake can not determine linker language for target: wholeBodyDynamicsSettings`
2. the "linker language" is determined by the source files to be compiled, in this case the auto-generated files.
3. the first thing to do is verify if those files are properly found/identified. The target being the `wholeBodyDynamicsSettings`, our check starts in the respective `CMakeLists.txt`.
4. the variables identifying the files are `WBDSETTINGS_SRC`, `WBDSETTINGS_HEADERS`, `WBDSETTINGS_INCLUDES`. An echo of those revealed that `WBDSETTINGS_SRC` was empty.
5. A check of the `yarp_idl_to_dir` macro definition lead to notice the deprecation.

#### Second issue
Is probably related to an old Yarp issue https://github.com/robotology/yarp/issues/368. In this case, @traversaro 's experience on this was the main "debugging tool" :D .

nunoguedelha, 2019-10-20:
@traversaro If it's ok for you I'll move the issue to `codyco-modules`.

traversaro, 2019-11-18:
> @traversaro If it's ok for you I'll move the issue to `codyco-modules`.

I think that given that it was fixed in https://github.com/robotology/codyco-modules/pull/282 , we can just close it, thanks!

closed 2019-11-18 by traversaro