robotology/robotology-superbuild #280: Use own repository in custom tag (ref. ROBOTOLOGY_PROJECT_TAGS)

valegagge, 2019-10-28:
I think the recently added feature about projects tag choice is very useful. I'm writing here to ask if it is possible to expand the "custom" tag by indicating the repository to use in addition to the tag in the custom file.

For example, my use case is the following:
I'd liker to crate a container with the superbuild as usual, but I'd like to use my icub-main fork where I developed some stuff.

I don't know if this is possible and how much effort its development requires, so please feel free to reject this proposal.

traversaro, 2019-10-28:
I did not document it because I am not really sure about all the implications, but all the arguments of the `ycm_ep_helper` can be overridden by CMake variables, see http://robotology.github.io/ycm/gh-pages/v0.10/manual/ycm-superbuild.7.html#overriding-parameters and https://github.com/robotology/ycm/blob/7ff014a958c3a7286644eab4324944de041ffcfb/modules/YCMEPHelper.cmake#L771 . For this reason, you can just add: 
~~~
set(ICUB_REPOSITORY valegagge/icub-main.git
~~~
and in theory (at least on a clean superbuild) you could specify the repo in place of the official one. However, I never tested this so I am not sure if it works or other problems arise.

traversaro, 2019-10-31:
Hi @valegagge , if the answer solves your issue, feel free to close this issue. If we want to document this feature, we can open a new issue to track this.

valegagge, 2019-10-31:
hi @traversaro , I think your answer is the solution, but I had no time to verify if it works. For this reason I don't close this issue yet. 
Feel free to close this issue and in case I'll get in trouble with your solution I'll reopen it.

traversaro, 2019-10-31:
Ok, let's wait for the verification if you plan to test it!

traversaro, 2020-02-18:
As I guess there is no plan to test this, I think we can close this issue @valegagge .

closed 2020-02-18 by traversaro