robotology/robotology-superbuild #285: Links broken in doc/change-project-tags.md

xEnVrE, 2019-11-05:
Links to `cmake/ProjectTagsStable.cmake` and `cmake/ProjectTagsUnstable.cmake` in [doc/change-project-tags.md](https://github.com/robotology/robotology-superbuild/doc/change-project-tags.md) are broken.

They point to 

`https://github.com/robotology/robotology-superbuild/blob/master/doc/cmake/ProjectTagsStable.cmake`

and 

`https://github.com/robotology/robotology-superbuild/blob/master/doc/cmake/ProjectTagsUnstable.cmake` that do not exist.

traversaro, 2019-11-05:
Thanks,  can you check if https://github.com/robotology/robotology-superbuild/pull/286 fixes the issue for you?

xEnVrE, 2019-11-05:
That works!

closed 2019-11-05 by traversaro

traversaro, 2019-11-05:
Fixed in https://github.com/robotology/robotology-superbuild/pull/286 .