robotology/robotology-superbuild #289: Support and test ROBOTOLOGY_USES_PYTHON on macOS

traversaro, 2019-11-07:
Things to do: 
* Decide which python installation to support by default 
   * probably homebrew one, that by default is python 3 now 
* Decide how to handle python3, that is not supported by rtf, see https://github.com/robotology/robot-testing-framework/issues/37 
  * For the time being, probably it make sense to just do not compile the python part of RTF when ROBOTOLOGY_USES_PYTHON is  set to ON

traversaro, 2019-11-07:
cc @nunoguedelha @diegoferigo

diegoferigo, 2019-11-08:
> Decide which python installation to support by default/recommend/test 

Given our Linux maintenance policy, I would find the closest match on OSX, that is for sure at least Python 3.6.

> Decide how to handle python3, that is not supported by rtf

Waiting for the fix, I would just do as you said. I'm not even sure if there are out there any OSX users using RTF from Python to be honest.

traversaro, 2019-11-08:
> Given our Linux maintenance policy, I would find the closest match on OSX, that is for sure at least Python 3.6.

In which sense? I mean, if we go for python distributed by homebrew, we need to get whatever version they distribute, that now is Python 3.7 (see  https://formulae.brew.sh/formula/python ).

diegoferigo, 2019-11-08:
I'm not familiar with Homebrew, I thought users could choose the version. Anyway, it's fine if 3.7 is the closest versions. What's important is not using an older version than what we support in GNU/Linux.

traversaro, 2021-09-07:
Fixed in https://github.com/robotology/robotology-superbuild/pull/663, `ROBOTOLOGY_USES_PYTHON` is fully supported on macOS when compiling with conda-forge dependencies.

closed 2021-09-07 by traversaro