robotology/robotology-superbuild #293: YCM build failing on Windows

GiulioRomualdi, 2019-11-08:
I'm trying to compile the `robotology-superbuild` on `Windows 10` with `Visual Studio 2017`
The process failed when `YCM` is built.  
We noticed: 
1.  `YCM` builds directly in the folder `build\robotology\YCM` without throwing errors.
2. When `YCM` is compiled by the superbuild (i.e. running `cmake --build . --config Release` in `build\`) the process failed with the following error:
    ```
    2>------ Build started: Project: YCM (ExternalProjectTargets\YCM\YCM), Configuration: RelWithDebInfo x64 ------
    2>Creating directories for 'YCM'
    2>Building Custom Rule C:/Users/icub/sb/CMakeLists.txt
    2>CMake does not need to re-run because C:/Users/icub/sb/build/CMakeFiles/generate.stamp is up-to-date.
    2>Performing download step (git clone) for 'YCM'
    2>-- Avoiding repeated git clone, repository already exists
    2>Skipping update step for 'YCM'
    2>No patch step for 'YCM'
    2>Performing configure step for 'YCM'
    2>Not searching for unused variables given on the command line.
    2>loading initial cache file C:/Users/icub/sb/build/robotology/YCM/CMakeFiles/YCMTmp/YCM-cache-RelWithDebInfo.cmake
    2>CMake Error : error : generator : Visual Studio 15 2017 Win64
    2>Does not match the generator used previously: Visual Studio 15 2017
    2>Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.
   2>Done building project "YCM.vcxproj" -- FAILED.
   ```

We tried to change the branch of `YCM` from `v0.10.4` (the default one of the [superbuild](https://github.com/robotology/robotology-superbuild/blob/bf8ddd6e8b44811d98fc796920c07cc31ef06922/CMakeLists.txt#L119)) to `master`  but nothing changed

@kouroshD @traversaro

GiulioRomualdi, 2019-11-08:
We noticed that the by default `YCM` compiles in `Win32` and it is not possible to switch from `Win32` to `x64`

[image]

traversaro, 2019-11-08:
I think this is related to https://github.com/robotology/robotology-superbuild/issues/265 . Can you try to change the if in https://github.com/robotology/robotology-superbuild/blob/master/CMakeLists.txt#L116 to also cover MSVC/Windows, and try again on a clean build? Thanks!

traversaro, 2019-11-08:
Otherwise build externally YCM is another workaround.

GiulioRomualdi, 2019-11-08:
> I think this is related to #265 . Can you try to change the if in https://github.com/robotology/robotology-superbuild/blob/master/CMakeLists.txt#L116 to also cover MSVC/Windows, and try again on a clean build? Thanks!

This one fixes the problem, should I open a PR?

traversaro, 2019-11-08:
Yes, thanks! 

> 
> 
> > I think this is related to #265 . Can you try to change the if in https://github.com/robotology/robotology-superbuild/blob/master/CMakeLists.txt#L116 to also cover MSVC/Windows, and try again on a clean build? Thanks!
> 
> This one fixes the problem, should I open a PR?

Yes, thanks. I avoided to just change the TAG as that is still an experimental commit, when the PR https://github.com/robotology/ycm/pull/294 , we will switch the tag for everyone.

traversaro, 2019-11-11:
@GiulioRomualdi @kouroshD can you provide the code that is working for you? Thanks!

traversaro, 2019-11-14:
Fixed by https://github.com/robotology/robotology-superbuild/pull/295 .

closed 2019-11-14 by traversaro