robotology/robotology-superbuild #313: Unstable build fails for HUMAN_DYNAMICS component

traversaro, 2019-12-04:
Related comments: 
* https://github.com/robotology/yarp/issues/2118#issuecomment-561614196
* https://github.com/robotology/wearables/issues/56#issuecomment-561614604

traversaro, 2019-12-05:
Related YARP issue: https://github.com/robotology/yarp/issues/2142 .

lrapetti, 2019-12-10:
Unfortunately I think I will not be able to have a look at this within this week.

traversaro, 2019-12-10:
> 
> 
> Unfortunately I think I will not be able to have a look at this within this week.

It would be great to have someone looking into this (https://github.com/robotology/yarp/issues/2142) @DanielePucci , as the release of YARP has been already done, and release of the rest of iCub software is rapidly approaching.

DanielePucci, 2019-12-11:
Thanks @traversaro for this. I believe we are going to address this next week

traversaro, 2019-12-11:
Not a big problem, but just to let you know, probably this friday we are going to switch the "stable" branches to use yarp 3.3, so the failure will be propagated from unstable branches to stable branches. Future "releases" of this kind will be done on a time-based basis, so for the future we will be able to deal with this problems properly.

DanielePucci, 2019-12-11:
CC @lrapetti @Yeshasvitvs

traversaro, 2019-12-13:
After the merge of https://github.com/robotology/robotology-superbuild/pull/320, the failure now is also there for stable branches (i.e. the default robotology-superbuild configuration).

lrapetti, 2019-12-16:
The problem should be fixed for `wearable` with https://github.com/robotology/wearables/pull/71

lrapetti, 2019-12-16:
Analogously https://github.com/robotology/human-dynamics-estimation/pull/168 should fix the problem for `human-dynamics-estimation` `master` (the `devel` should be already compiling)

lrapetti, 2019-12-17:
PRs have been merged. @traversaro can you trigger a test to see if `HUMAN_DYNAMICS` build is now working?

traversaro, 2019-12-17:
> 
> 
> PRs have been merged. @traversaro can you trigger a test to see if `HUMAN_DYNAMICS` build is now working?

Done at https://github.com/robotology/robotology-superbuild/pull/323 .

closed 2019-12-17 by traversaro