robotology/robotology-superbuild #321: GitHub Actions Ubuntu's CI is broken

traversaro, 2019-12-13:
See for example: https://github.com/robotology/robotology-superbuild/commit/453c1f586a41cbb191f332509f84ffaca072c16d/checks . 

The specific error is: 
~~~
2019-12-13T08:05:11.5792806Z W: Conflicting distribution: https://packages.microsoft.com/repos/microsoft-ubuntu-bionic-prod bionic InRelease (expected bionic but got stable)
2019-12-13T08:05:11.5793976Z E: Repository 'https://packages.microsoft.com/repos/microsoft-ubuntu-bionic-prod bionic InRelease' changed its 'Origin' value from 'microsoft-ubuntu-bionic-prod bionic' to 'vscode stable'
2019-12-13T08:05:11.5794868Z E: Repository 'https://packages.microsoft.com/repos/microsoft-ubuntu-bionic-prod bionic InRelease' changed its 'Label' value from 'microsoft-ubuntu-bionic-prod bionic' to 'vscode stable'
2019-12-13T08:05:11.5795732Z E: Repository 'https://packages.microsoft.com/repos/microsoft-ubuntu-bionic-prod bionic InRelease' changed its 'Suite' value from 'bionic' to 'stable'
2019-12-13T08:05:11.5796558Z E: Repository 'https://packages.microsoft.com/repos/microsoft-ubuntu-bionic-prod bionic InRelease' changed its 'Codename' value from 'bionic' to 'stable'
2019-12-13T08:05:11.5797145Z W: Conflicting distribution: https://packages.microsoft.com/ubuntu/16.04/prod xenial InRelease (expected xenial but got stable)
2019-12-13T08:05:11.5797939Z E: Repository 'https://packages.microsoft.com/ubuntu/16.04/prod xenial InRelease' changed its 'Origin' value from 'microsoft-ubuntu-xenial-prod xenial' to 'vscode stable'
2019-12-13T08:05:11.5798481Z E: Repository 'https://packages.microsoft.com/ubuntu/16.04/prod xenial InRelease' changed its 'Label' value from 'microsoft-ubuntu-xenial-prod xenial' to 'vscode stable'
2019-12-13T08:05:11.5799813Z E: Repository 'https://packages.microsoft.com/ubuntu/16.04/prod xenial InRelease' changed its 'Suite' value from 'xenial' to 'stable'
2019-12-13T08:05:11.5800292Z E: Repository 'https://packages.microsoft.com/ubuntu/16.04/prod xenial InRelease' changed its 'Codename' value from 'xenial' to 'stable'
2019-12-13T08:05:11.5833619Z ##[error]Process completed with exit code 100.
~~~

We have this issue in a lot of other repos (cc @AlbertoRemusIIT), but I will use this issue to track the resolution.

traversaro, 2019-12-13:
The issue has been reported for GitHub Actions in https://github.community/t5/GitHub-Actions/ubuntu-latest-Apt-repository-list-issues/td-p/41122 . In that discussion, a workaround has been reported in https://github.community/t5/GitHub-Actions/ubuntu-latest-Apt-repository-list-issues/m-p/41138#M4501 .

However, the core issue seems in dotnet's official apt repos that are serving malformed files, see https://github.com/dotnet/core/issues/3988  .

AlbertoRemusIIT, 2019-12-13:
Thanks for the insight

traversaro, 2019-12-18:
This has been fixed last week, see  https://github.community/t5/GitHub-Actions/ubuntu-latest-Apt-repository-list-issues/m-p/41199/highlight/true#M4529 and https://github.com/dotnet/core/issues/3995#issuecomment-565830031 .

closed 2019-12-18 by traversaro