robotology/robotology-superbuild #322: Compilation failure

bemilio, 2019-12-16:
The robotology compilation on Ubuntu 18.04 fails with 

`[ 31%] Linking CXX shared library ../../lib/libWBToolbox.so
/usr/bin/ld: /usr/local/lib/libqpOASES.a(MessageHandling.cpp.o): relocation R_X86_64_PC32 against symbol `stdout@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value`

bemilio, 2019-12-16:
I tried pointing at the folder where I compiled qpOASES from source with no success

traversaro, 2019-12-16:
The problem is that you are trying to link a static library of qpOASES (compiled without [-fPIC](https://stackoverflow.com/questions/3961446/why-does-gcc-not-implicitly-supply-the-fpic-flag-when-compiling-static-librarie)) with a dynamic library (in this case WB-Toolbox). This is not possible, so you need either to use a different qpOASES library, either a static one compiled with the -fPIC option, or with a shared one. 

How did you installed the qpOASES library available in `/usr/local/lib`? Compiled from source? Which version  of qpOASES did you used? You can  try to set to `FALSE` the `USE_SYSTEM_qpOASES` option in the suberbuild (see http://robotology.github.io/ycm/gh-pages/git-master/manual/ycm-superbuild.7.html#ycm-superbuild-manual-for-developers) to use the qpOASES compiled by the superbuild instead, but I am not sure if that works correctly to be honest.

traversaro, 2020-01-03:
Hi @bemilio , did you solve the problem? As mentioned in my original comment, I do not think the problem is related at all to the superbuild.

bemilio, 2020-01-03:
Yes, sorry if I kept this open. I had a messy installation, cleaning the system and the build solved the problem.

traversaro, 2020-01-03:
Ok, thanks for the feedback @bemilio .

closed 2020-01-03 by traversaro