robotology/robotology-superbuild #331: Collect tags for 2020.02 release

traversaro, 2020-01-20:
Hi everyone, in particular maintainers of the components of the robotology-superbuild and of the projects contained  in the robotology-superbuild. As discussed in https://github.com/robotology/QA/issues/377#issue-538458329, there is the plan of creating a "distribution" release for the software contained in the robotology-superbuild every three months. 

The next planned release is the 2020.02, tentatively planned for the mid of February. For this reason, I would like to start collecting the set of (compatibile) releases of the software contained in the robotology-superbuild that could be used for this 2020.02 release, to provide a specific `ProjectTags.cmake` file, as for example was proposed  in https://github.com/robotology/robotology-superbuild/pull/318 . If you mantain any of the software listed in the following, feel free to comment on this issue to propose an already released version of the software that could be include in the 2020.02 distribution release. Even if the release is not done yet, but you still plan to do a release before February 2020, feel free to  comment, thanks! 

Proposed tags for 2020.02  :  
~~~
# External projects
set_tag(qpOASES_TAG ???)
set_tag(osqp_TAG v0.6.0)

# Core profile 
set_tag(YARP_TAG v3.3.1)
set_tag(ICUB_TAG ???)
set_tag(ICUBcontrib_TAG ???)
set_tag(icub-models_TAG ???)
set_tag(robots-configuration_TAG ???)
set_tag(GazeboYARPPlugins_TAG ???)
set_tag(icub-gazebo_TAG ???)
set_tag(yarp-matlab-bindings_TAG v3.3.0)

# Robot Testing 
set_tag(RobotTestingFramework_TAG v2.0.0)
set_tag(icub-tests_TAG ???)

# Dynamics 
set_tag(iDynTree_TAG v1.0.0)
set_tag(codyco-modules_TAG v0.3.0)
set_tag(BlockFactory_TAG v0.8)
set_tag(WBToolbox_TAG v5)
set_tag(OsqpEigen_TAG v0.5.0)
set_tag(UnicyclePlanner_TAG ???)
set_tag(walking-controllers_TAG ???)
set_tag(icub-gazebo-wholebody_TAG ???)
set_tag(whole-body-controllers_TAG v2.0)

# Teleoperation
set_tag(walking-teleoperation_TAG ???)

# Human Dynamics 
set_tag(forcetorque-yarp-devices_TAG ???)
set_tag(wearables_TAG ???)
set_tag(human-dynamics-estimation_TAG ???)
set_tag(human-gazebo_TAG ???)

# iCub Head
set_tag(icub-firmware-shared_TAG ???)
set_tag(xsensmt-yarp-driver_TAG ???)
~~~

traversaro, 2020-01-20:
cc @lrapetti @Yeshasvitvs @GiulioRomualdi @prashanthr05 @S-Dafarra

S-Dafarra, 2020-01-20:
For ``UnicyclePlanner``: https://github.com/robotology/unicycle-footstep-planner/releases/tag/v0.1.103

traversaro, 2020-01-20:
Ack. Note that the long patch version was used in YARP and iDynTree to indicate a development version, so if you see it fit feel free to bump the minor version.

S-Dafarra, 2020-01-20:
Not sure I understood :grimacing:

traversaro, 2020-01-20:
You released version 0.1.103 of the `UnicyclePlanner`, but that use of `103` as a patch number for iDynTree and YARP is used to indicate that that version is not a stable version, see https://github.com/robotology/yarp/blob/master/.github/CONTRIBUTING.md#terminology . So, if you like you can bump the version of `UnicyclePlanner`  to 0.2.0, so it is clear that it is a stable release.

S-Dafarra, 2020-01-20:
Ok, done and updated tag: https://github.com/robotology/unicycle-footstep-planner/releases/tag/v0.2.0

traversaro, 2020-01-21:
Missing tags (I mention who I presume is the maintainer of the relative repo, correct me if I am wrong): 
* `qpOASES` @traversaro 
* `xsensmt-yarp-driver` @traversaro 
* `walking-controllers` @GiulioRomualdi 
* `icub-gazebo-wholebody` @gabrielenava 
* `walking-teleoperation` @kouroshD 
* `forcetorque-yarp-devices` @fjandrad 
* `wearables` @lrapetti 
* `human-dynamics-estimation` @Yeshasvitvs 
* `human-gazebo` @Yeshasvitvs

traversaro, 2020-01-27:
cc @vtikha

lrapetti, 2020-01-28:
For `wearables`: https://github.com/robotology/wearables/releases/tag/v1.0.0

traversaro, 2020-01-29:
Given https://github.com/robotology/robotology-superbuild/issues/124, we should also  have tags for `speech`, `icub-basic-demos` and `funny-things`. cc @pattacini

pattacini, 2020-01-29:
@traversaro we normally tag `icub-basic-demos` already.
I'll be dealing with `speech` and `funny-things` then.

Yeshasvitvs, 2020-01-29:
[human-gazebo v1.0](https://github.com/robotology/human-gazebo/releases/tag/v1.0) 

[HDE v2.0.0](https://github.com/robotology/human-dynamics-estimation/releases/tag/v2.0.0) - after the big code refactoring towards v2.0, we did not have any code on the master branch tagged for the release. This v2.0.0 tags is at the current devel branch, which is fully functional for demonstrating HDE and retargeting.  Following the code clean up https://github.com/robotology/human-dynamics-estimation/issues/101, minor releases can be done.

@traversaro @lrapetti @kouroshD

traversaro, 2020-01-29:
> [human-gazebo v1.0](https://github.com/robotology/human-gazebo/releases/tag/v1.0)
> 
> [HDE v2.0.0](https://github.com/robotology/human-dynamics-estimation/releases/tag/v2.0.0) - after the big code refactoring towards v2.0, we did not have any code on the master branch tagged for the release. This v2.0.0 tag is at the current devel branch, which is fully functional for demonstrating HDE and retargeting. Following the code clean up [robotology/human-dynamics-estimation#101](https://github.com/robotology/human-dynamics-estimation/issues/101), minor releases can be done.
> 
> @traversaro @lrapetti @kouroshD

Thanks! If even releases such as v2.0.0 is  based  on the `devel` branch,  what is the purpose of the master  branch? To keep minor release of the 1.x series?

Yeshasvitvs, 2020-01-30:
@traversaro I think we will not be using master branch anymore. I remember there was an issue related to "stable" branch, can you please point to it. If I remember well the devel branch is considered the stable branch right ?

diegoferigo, 2020-01-30:
@Yeshasvitvs If the master branch is no longer used, I would just add a new `v1.X` to the last master commit, merge devel to master, and tag the `v2.0`.

Since now icub-tech changed considerably the release schedule and new versions are tagged more often, I start seeing less and less the need of a `devel` branch on many of our projects. HDE and wearables are among those.

traversaro, 2020-01-30:
> . I remember there was an issue related to "stable" branch, can you please point to it.

Sorry, I am not really sure which issue are you referring to.

lrapetti, 2020-01-30:
> @Yeshasvitvs If the master branch is no longer used, I would just add a new `v1.X` to the last master commit, merge devel to master, and tag the `v2.0`.

I'm not sure but I think teleoperation guys are actually using `master` branch (or the branch that is being merged into master -> https://github.com/robotology/human-dynamics-estimation/pull/164) since it is actually still more stable since `devel` branch has some dependency on non-stable branches of other repos.

Probably @kouroshD can confirm it.

lrapetti, 2020-01-30:
> > @Yeshasvitvs If the master branch is no longer used, I would just add a new `v1.X` to the last master commit, merge devel to master, and tag the `v2.0`.
> 
> I'm not sure but I think teleoperation guys are actually using `master` branch (or the branch that is being merged into master -> [robotology/human-dynamics-estimation#164](https://github.com/robotology/human-dynamics-estimation/pull/164)) since it is actually still more stable since `devel` branch has some dependency on non-stable branches of other repos.
> 
> Probably @kouroshD can confirm it.

sorry, I was confused, you can ignore my previous comment. `master` is actually deprecated and can be replaced by `devel` branch.

Yeshasvitvs, 2020-01-30:
> If the master branch is no longer used, I would just add a new v1.X to the last master commit, merge devel to master, and tag the v2.0.

@diegoferigo master branch actually has the code of v2.0.

Yeshasvitvs, 2020-01-30:
> Sorry, I am not really sure which issue are you referring to.

https://github.com/robotology/QA/issues/372

Sorry, I got confused with which is considered a stable branch. So, if I understood well:

- `devel` branches are no longer used
- `master` branch becomes the devel branch that will contain new features and is under active development
- stable branches are version tagged

@traversaro please correct me if I am getting it wrong. Thanks!

traversaro, 2020-01-30:
> > Sorry, I am not really sure which issue are you referring to.
> 
> [robotology/QA#372](https://github.com/robotology/QA/issues/372)
> 
> Sorry, I got confused with which is considered a stable branch. So, if I understood well:
> 
>     * `devel` branches are no longer used
> 
>     * `master` branch becomes the devel branch that will contain new features and is under active development
> 
>     * stable branches are version tagged
> 
> 
> @traversaro please correct me if I am getting it wrong. Thanks!

That is the policy used by YCM and YARP, most of other repos remain on the older policy, i.e. : 
* master is the latest stable branches, on which new bug-fix releases are tagged 
* devel is the "unstable" branch in which new features are developed, and that is merged to master before a new release

For the repos you mantain, you are free to chose the branch policy that you prefer, at the moment master/devel is still the one most used in IIT's repos, but the one used by YARP has its own advantages.

Yeshasvitvs, 2020-01-30:
We will stick with the master/devel branch management. I rebased the `master` branch to `devel` and merged it. So, both the branches are aligned at the moment https://github.com/robotology/human-dynamics-estimation/issues/101#issuecomment-580198868. 

I created as new tag at the current master [HDE v2.0.0](https://github.com/robotology/human-dynamics-estimation/releases/tag/v2.0.0). Please ignore the tag mentioned in https://github.com/robotology/robotology-superbuild/issues/331#issuecomment-579897237

@lrapetti @kouroshD

traversaro, 2020-01-30:
Thanks for the effort @Yeshasvitvs , @lrapetti and @kouroshD !

traversaro, 2020-01-31:
> We will stick with the master/devel branch management. I rebased the `master` branch to `devel` and merged it. So, both the branches are aligned at the moment [robotology/human-dynamics-estimation#101 (comment)](https://github.com/robotology/human-dynamics-estimation/issues/101#issuecomment-580198868).
> 
> I created as new tag at the current master [HDE v2.0.0](https://github.com/robotology/human-dynamics-estimation/releases/tag/v2.0.0). Please ignore the tag mentioned in [#331 (comment)](https://github.com/robotology/robotology-superbuild/issues/331#issuecomment-579897237)
> 
> @lrapetti @kouroshD

Unfortunately the CI revealed a critical bug that prevents compilation on Windows of that version (see https://github.com/robotology/human-dynamics-estimation/issues/171), so it is not suitable to included iin 2020.02 . In general, it may be a good idea to keep a given software version in master or in devel for a short time to ensure that it works fine, before tagging a release.

GiulioRomualdi, 2020-01-31:
@traversaro 
I updated osqp-eigen tag. Now we can use this one https://github.com/robotology/osqp-eigen/releases/tag/v0.5.2 

@nunoguedelha

traversaro, 2020-02-17:
At the moments we are just missing the tags: 
* The tags for all the `icub-*` repositories
* For the walking-controllers repo, hopefully after https://github.com/robotology/walking-controllers/pull/53

Besides that, all the other repos should have the correct tags.

traversaro, 2020-02-19:
The `icub-*` repo has been tagged, and also walking-controllers has its valid release. Thanks to everyone that contributed!

traversaro, 2020-02-21:
It turns out that there are some problems in the provided tags, see https://github.com/robotology/robotology-superbuild/pull/352, specifically in `human-dynamics-estimation` and iDynTree. 

@kouroshD @traversaro @lrapetti can  you please add a new tag? Thanks!

traversaro, 2020-02-25:
The new 2020.02 is finally completed in https://github.com/robotology/robotology-superbuild/releases/tag/v2020.02 . Feel free to test it either  by checking out the `v2020.02` tag, or using the Windows binaries.

closed 2020-02-25 by traversaro