robotology/robotology-superbuild #333: Request for Comment: move all source repos in src subdirectory

traversaro, 2020-01-28:
I would like to collect feedback on the following: instead of downloading the source repo of subprojects in the `external` or `robotology` directories as it is now, download all the source repos in the `src` directory. 

Pros: 
* Consistency: all the source repos will go in the same directory, and with a directory whose meaning is clear from its name. 
* Familiarity for people used to ROS/ROS2/colcon: in ROS/ROS2 workspaces, all the repos are contained in the `src` subdirectory of the  workspace 
* We can use more easily the `vcs` tool (see https://github.com/dirk-thomas/vcstool)  to manipulate the repos if they are in devel model, for example to export the value of all the repos in the superbuild. `vcs` is the tool used in ROS2, so it would be something familiar used to ROS2 workspaces. In theory `vcs` could be used also with the current directory structure, but the saved file would have the strange format of having the `external` or `robotology` directory prefixed to the CMake package name: 
~~~
repositories:
  external/osqp:
    type: git
    url: https://github.com/oxfordcontrol/osqp
    version: master
  robotology/YARP:
    type: git
    url: https://github.com/robotology/yarp
    version: master
~~~
this would complexify its use as a reference for superbuild tags.

Cons:
* It is a change. : )

S-Dafarra, 2020-01-28:
I have no objections.

GiulioRomualdi, 2020-01-28:
It's fine!

valegagge, 2020-02-10:
For team-code is ok!
@vtikha

S-Dafarra, 2020-07-13:
In the meanwhile, I wrote a simple script to check the status of each repo:
```sh
#!/bin/bash

subdirs="$@"

if (($# == 0)); then
    subdirs="."
fi

for subdir in ${subdirs}; do \
    for i in ${subdir}/*/; do \
        echo "--------------------------------------------"
        (cd $i && echo -n "${i}: " \
        && git rev-parse --abbrev-ref HEAD \
        && git log -1 --format="%cr|%s" \
        && git status --short); \
    done
done
echo "--------------------------------------------"
```
It can be called with no arguments, or with the list of directories to check. I don't know if it is useful or if this is the right place.

traversaro, 2022-11-20:
Fixed in https://github.com/robotology/robotology-superbuild/pull/556.

closed 2022-11-20 by traversaro