robotology/robotology-superbuild #354: yarpview does not start when installed from robotology-full-installer-2020.02-win64

traversaro, 2020-02-25:
This could be something related to some old vcpkg dependency, or something else.
To investigate this it could  be useful to: 
* [ ] Try to run  yarpview from   robotology-full-installer-2020.02-win64 on  another laptop different from mine 
* [ ] Try to recompile YARP from source against the dependencies installed by robotology-dependencies-installer-2020.02-win64.exe, and see if the problem persists 
* [ ] Try to recompile YARP from source against the dependencies installed by a recent vcpkg, and see if the problem persists (I guess @pattacini you already have this configuration to test?)

pattacini, 2020-02-25:
On my system, `yarpview` is working although it's quite difficult to reconstruct the history of how the dependencies have been compiled as I've updated `vcpkg` without rebuilding the dependencies since quite long.

My suggestion is to rely on [Free Windows VM][1] and make use of VS2019 community release in case it is needed.

[1]: https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/

pattacini, 2020-02-28:
Another possibility is to use [Windows Sandbox](https://www.howtogeek.com/399290/how-to-use-windows-10s-new-sandbox-to-safely-test-apps/).

pattacini, 2020-02-28:
Using the sandbox approach I can confirm the problem exists.

drdanz, 2020-03-04:
Does yarpscope start?
This might be related to qml and/or to the opengl setup

traversaro, 2020-03-04:
> Does yarpscope start?
> This might be related to qml and/or to the opengl setup

I did not tried yarpscope, but I successfully used: 
* yarpmanager
* yarpmotorgui

drdanz, 2020-03-04:
Since the iCubGui has the same issue, I'd say it's an opengl problem...

traversaro, 2020-03-04:
> Since the iCubGui has the same issue, I'd say it's an opengl problem...

Note that the iCub_SIM instead is working correctly.

traversaro, 2020-03-04:
Possibly related issues: 
* https://github.com/robotology/robotology-superbuild/issues/134
* https://github.com/robotology/yarp/issues/2077
* https://github.com/robotology/yarp/issues/2071

traversaro, 2020-03-15:
As Qt 5.14 is now relocatable (see https://www.qt.io/blog/qt-is-relocatable) it could perhaps make sense to wait for  vcpkg to include Qt 5.14 (see https://github.com/microsoft/vcpkg/pull/10009) and then test again.

traversaro, 2020-03-22:
I tried with a vcpkg build from  vcpkg tag 2020.01, but the problem  is present also there. Probably our best hope is the new Qt 5.14 .

traversaro, 2020-04-27:
The behaviour is the same even with the latest qt 5.14 installed from vcpkg.

traversaro, 2020-04-27:
I debugged a bit this issue, and it was a bit complex due to the fact that `yarpview` and `yarpscope` do not return any kind of error message related to the error that they encounter. However, the problem was in the loading of the `qml` file in https://github.com/robotology/yarp/blob/4785668ae205f87a62c7777f77517c3b5558402c/src/yarpview/src/main.cpp#L110 . Unfortunately, there seems not to be any way of debugging the loading process of qml files (or at least I was not able to make it work).

To check, I substituted the `"qrc:/qml/QtYARPView/main.qml"` with a simple hello world qml (see https://doc.qt.io/qt-5/qml-tutorial1.html) and it was working fine. So I added back the original `.qml` line by line, and I found that the qml file was not loaded correctly if the `import QtQuick.Controls 1.1` line was present (https://github.com/robotology/yarp/blob/4785668ae205f87a62c7777f77517c3b5558402c/src/yarpview/src/qml/QtYARPView/main.qml#L20 ). It turns out that in vcpkg we were not installing the `qt5-quickcontrols` vcpkg port, as we were just installing the `qt5-quickcontrols2` port. Installing the missing port fixes the problem. So, brief recap in the following:

# Fix for existing installations
For existing installation, installing the missing port of `qt5-quickcontrols` as: 
~~~
C:/robotology/vcpkg/vcpkg.exe install --triplet x64-windows qt5-quickcontrol
~~~
solves the problem. 

# Fix for the future
Add `qt5-quickcontrol` in https://github.com/robotology/robotology-superbuild-dependencies-vcpkg .

closed 2020-04-29 by traversaro

drdanz, 2020-04-29:
@traversaro Sorry if I didn't mention this before, but for the next time the `QT_DEBUG_PLUGINS` and `QML_IMPORT_TRACE` (not sure which one of them) usually gives you some useful information about this kind of error...

traversaro, 2020-04-29:
> @traversaro Sorry if I didn't mention this before, but for the next time the `QT_DEBUG_PLUGINS` and `QML_IMPORT_TRACE` (not sure which one of them) usually gives you some useful information about this kind of error...

Actually I enabled them (both in my enviroment and uncommenting https://github.com/robotology/yarp/blob/8bc852eb80d463df96dff71a4be9dc2f7e01bae3/src/yarpview/src/main.cpp#L89), but for some reason they were not working, probably I was doing something wrong myself or the vcpkg build of Qt has some problems with those options.