robotology/robotology-superbuild #355: iCubGui does not start when installed from robotology-full-installer-2020.02-win64

traversaro, 2020-02-25:
This could be something related to some old vcpkg dependency, or something else.
To investigate this it could  be useful to: 
* [ ] Try to run  iCubGui from   robotology-full-installer-2020.02-win64 on  another laptop different from mine 
* [ ] Try to recompile icub-main from source against the dependencies installed by robotology-dependencies-installer-2020.02-win64.exe, and see if the problem persists 
* [ ] Try to recompile icub-main from source against the dependencies installed by a recent vcpkg, and see if the problem persists (I guess @pattacini you already have this configuration to test?)

pattacini, 2020-02-25:
On my system, `iCubGui` is working although it's quite difficult to reconstruct the history of how the dependencies have been compiled as I've updated `vcpkg` without rebuilding the dependencies since quite long.

My suggestion is to rely on [Free Windows VM][1] and make use of VS2019 community release in case it is needed.

[1]: https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/

pattacini, 2020-02-28:
Another possibility is to use [Windows Sandbox](https://www.howtogeek.com/399290/how-to-use-windows-10s-new-sandbox-to-safely-test-apps/).

pattacini, 2020-02-28:
Using the sandbox approach I can confirm the problem exists.

traversaro, 2020-04-27:
Even  with the fix in , the iCubGui still fails with: 
~~~
C:\src\robotology-superbuild\build\install>iCubGui
iCubGui

C:\src\robotology-superbuild\build\install>||| clearing context
||| adding context [iCubGui]
||| configuring
||| default config file specified as iCubGui.ini
||| checking [C:\src\robotology-superbuild\build\install\iCubGui.ini] (pwd)
||| checking [C:\Users\STraversaro\AppData\Roaming\yarp\config\robots\default] (robot YARP_CONFIG_HOME)
||| checking [C:\Users\STraversaro\AppData\Roaming\yarp\robots\default] (robot YARP_DATA_HOME)
||| checking [C:\ProgramData\yarp\robots\default] (robot YARP_CONFIG_DIRS)
||| checking [robots\default] (robot YARP_DATA_DIRS)
||| checking [C:/src/robotology-superbuild/build/install/share/yarp\robots\default] (robot YARP_DATA_DIRS)
||| checking [C:/src/robotology-superbuild/build/install/share/iCub\robots\default] (robot YARP_DATA_DIRS)
||| checking [C:/src/robotology-superbuild/build/install/share/ICUBcontrib\robots\default] (robot YARP_DATA_DIRS)
||| checking [config\path.d] (robot path.d YARP_DATA_DIRS)
||| checking [C:/src/robotology-superbuild/build/install/share/yarp\config\path.d] (robot path.d YARP_DATA_DIRS)
||| found C:/src/robotology-superbuild/build/install/share/yarp\config\path.d
obotology-superbuildbuildinstallshareICUBcontrib\robots\default] (robot yarp.d)
obotology-superbuildbuildinstallshareiCub\robots\default] (robot yarp.d)
||| checking [C:\Users\STraversaro\AppData\Roaming\yarp\config\contexts\iCubGui] (context YARP_CONFIG_HOME)
||| checking [C:\Users\STraversaro\AppData\Roaming\yarp\contexts\iCubGui] (context YARP_DATA_HOME)
||| checking [C:\ProgramData\yarp\contexts\iCubGui] (context YARP_CONFIG_DIRS)
||| checking [contexts\iCubGui] (context YARP_DATA_DIRS)
||| checking [C:/src/robotology-superbuild/build/install/share/yarp\contexts\iCubGui] (context YARP_DATA_DIRS)
||| checking [C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui] (context YARP_DATA_DIRS)
||| found C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui
||| checking [C:/src/robotology-superbuild/build/install/share/ICUBcontrib\contexts\iCubGui] (context YARP_DATA_DIRS)
obotology-superbuildbuildinstallshareICUBcontrib\contexts\iCubGui] (context yarp.d)
obotology-superbuildbuildinstallshareiCub\contexts\iCubGui] (context yarp.d)
||| checking [C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\iCubGui.ini] (context)
||| found C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\iCubGui.ini
freeglut (iCubimator): fgPlatformInitialize: CreateDC failed, Screen size info may be incorrect
This is quite likely caused by a bad '-display' parameter
yarp: Port /iCubGui/objects active at tcp://192.168.1.59:10005/
yarp: Port /iCubGui/textures active at tcp://192.168.1.59:10006/
yarp: Port /iCubGui/forces active at tcp://192.168.1.59:10007/
yarp: Port /iCubGui/speechText:i active at tcp://192.168.1.59:10003/
yarp: Port /iCubGui/dbgText:i active at tcp://192.168.1.59:10004/
||| finding path [geometry]
||| checking [C:\src\robotology-superbuild\build\install\skeleton.ini] (pwd)
||| checking [C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\skeleton.ini] (context)
||| found C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\skeleton.ini
yarp: Port /iCubGui/base:i active at tcp://192.168.1.59:10008/
yarp: Port /iCubGui/torso:i active at tcp://192.168.1.59:10009/
yarp: Port /iCubGui/head:i active at tcp://192.168.1.59:10010/
yarp: Port /iCubGui/left_arm:i active at tcp://192.168.1.59:10011/
yarp: Port /iCubGui/right_arm:i active at tcp://192.168.1.59:10012/
yarp: Port /iCubGui/left_leg:i active at tcp://192.168.1.59:10013/
yarp: Port /iCubGui/right_leg:i active at tcp://192.168.1.59:10014/
||| finding path [covers/hip.obj]
||| checking [C:\src\robotology-superbuild\build\install\covers/hip.obj] (pwd)
||| checking [C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/hip.obj] (context)
||| found C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/hip.obj
||| finding path [covers/torso.obj]
||| checking [C:\src\robotology-superbuild\build\install\covers/torso.obj] (pwd)
||| checking [C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/torso.obj] (context)
||| found C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/torso.obj
||| finding path [covers/head.obj]
||| checking [C:\src\robotology-superbuild\build\install\covers/head.obj] (pwd)
||| checking [C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/head.obj] (context)
||| found C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/head.obj
yarp: Port /iCubGui/inertials/rpc:i active at tcp://192.168.1.59:10015/
yarp: Port /iCubGui/inertials/measures:i active at tcp://192.168.1.59:10016/

C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/hip.obj

C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/torso.obj
[... 20 lines cut ...]
||| finding path [covers/upperleg_left.obj]
||| checking [C:\src\robotology-superbuild\build\install\covers/upperleg_left.obj] (pwd)
||| checking [C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/upperleg_left.obj] (context)
||| found C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/upperleg_left.obj
[... cut ...]
||| checking [C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/lowerleg_right.obj] (context)
||| found C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/lowerleg_right.obj
||| finding path [covers/foot_right.obj]
||| checking [C:\src\robotology-superbuild\build\install\covers/foot_right.obj] (pwd)
||| checking [C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/foot_right.obj] (context)
||| found C:/src/robotology-superbuild/build/install/share/iCub\contexts\iCubGui\covers/foot_right.obj
~~~

closed 2020-05-04 by pattacini