robotology/robotology-superbuild #366: Windows build is broken

traversaro, 2020-03-17:
This is due to https://github.com/robotology/walking-controllers/issues/58 .

traversaro, 2020-03-17:
This probably was fixed by https://github.com/robotology/walking-controllers/pull/59 , I restarted the build.

traversaro, 2020-03-17:
The build still fails, see  https://github.com/robotology/walking-controllers/issues/58#issuecomment-600107149 .

traversaro, 2020-03-20:
Hi @GiulioRomualdi, if you want I can provide a PR to walking-controllers to fix this, let me know if it is ok.

GiulioRomualdi, 2020-03-20:
Working right now on it

traversaro, 2020-03-20:
> 
> 
> Working right now on it

Great, thanks!

traversaro, 2020-03-21:
Fixed by https://github.com/robotology/walking-controllers/pull/60 .

closed 2020-03-21 by traversaro

GiulioRomualdi, 2020-03-21:
If you want to update the `TAG` in the superbuild this is the release: https://github.com/robotology/walking-controllers/releases/tag/v0.3.2

traversaro, 2020-03-21:
> If you want to update the `TAG` in the superbuild this is the release: https://github.com/robotology/walking-controllers/releases/tag/v0.3.2

Thanks, the 2020.02 release is now done and we don't plan to update. The 0.3.2 can be a good candidate for the 2020.05 release.