robotology/robotology-superbuild #372: Compilation of ROBOTOLOGY_ENABLE_ICUB_HEAD is failing on unstable branches on Ubuntu, macOS and Windows

traversaro, 2020-04-02:
The latest nightly build shown build time regression for the ROBOTOLOGY_ENABLE_ICUB_HEAD, see https://github.com/robotology/robotology-superbuild/actions/runs/68702878 . 

This  is probably due to https://github.com/robotology/icub-firmware-shared/pull/35 and https://github.com/robotology/icub-main/pull/649 fyi @Nicogene @pattacini . 

## Windows error
See https://pipelines.actions.githubusercontent.com/ZuyQm6ovz1ad02r7CsxwF1VdDAwrcBAjuVSwcIMiZZdLCwGhMg/_apis/pipelines/1/runs/567/signedlogcontent/8?urlExpires=2020-04-02T07%3A25%3A05.8395831Z&urlSigningMethod=HMACV1&urlSignature=yHlyaBTxrp6%2F0Nl0%2FsDMkb7HKEs60cgPCefvNSMhMAE%3D . 
~~~
2020-04-02T02:20:10.4038237Z        Creating library D:/a/robotology-superbuild/robotology-superbuild/build/robotology/icub-firmware-shared/eth/Debug/embobj.lib and object D:/a/robotology-superbuild/robotology-superbuild/build/robotology/icub-firmware-shared/eth/Debug/embobj.exp
2020-04-02T02:20:10.5327576Z EOYmutex.obj : error LNK2019: unresolved external symbol ace_mutex_new referenced in function eoy_mutex_New [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\icub-firmware-shared\eth\embobj.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\icub-firmware-shared.vcxproj]
2020-04-02T02:20:10.5328183Z EOYmutex.obj : error LNK2019: unresolved external symbol ace_mutex_take referenced in function s_eoy_mutex_take [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\icub-firmware-shared\eth\embobj.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\icub-firmware-shared.vcxproj]
2020-04-02T02:20:10.5328566Z EOYmutex.obj : error LNK2019: unresolved external symbol ace_mutex_release referenced in function s_eoy_mutex_release [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\icub-firmware-shared\eth\embobj.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\icub-firmware-shared.vcxproj]
2020-04-02T02:20:10.5329102Z EOYtheSystem.obj : error LNK2019: unresolved external symbol feat_yarp_time_now referenced in function eoy_sys_Initialise [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\icub-firmware-shared\eth\embobj.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\icub-firmware-shared.vcxproj]
2020-04-02T02:20:10.5728956Z D:\a\robotology-superbuild\robotology-superbuild\build\robotology\icub-firmware-shared\eth\Debug\embobj.dll : fatal error LNK1120: 4 unresolved externals [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\icub-firmware-shared\eth\embobj.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\icub-firmware-shared.vcxproj]
~~~

## macOS error
See https://pipelines.actions.githubusercontent.com/ZuyQm6ovz1ad02r7CsxwF1VdDAwrcBAjuVSwcIMiZZdLCwGhMg/_apis/pipelines/1/runs/567/signedlogcontent/4?urlExpires=2020-04-02T07%3A30%3A26.2082954Z&urlSigningMethod=HMACV1&urlSignature=aBPlgX%2Bk%2BxpuQEfcBtC4l%2FdsSaph9jJHQb6nanVNlFQ%3D . 
~~~
2020-04-02T02:37:25.6192050Z [  1%] Building CXX object embot/CMakeFiles/embot.dir/core/embot_core.cpp.o
2020-04-02T02:37:26.0036450Z In file included from /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.cpp:14:
2020-04-02T02:37:26.0037250Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:26:27: warning: alias declarations are a C++11 extension [-Wc++11-extensions]
2020-04-02T02:37:26.0037420Z     using Time          = std::uint64_t;  
2020-04-02T02:37:26.0037510Z                           ^
2020-04-02T02:37:26.0038120Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:28:27: warning: alias declarations are a C++11 extension [-Wc++11-extensions]
2020-04-02T02:37:26.0038270Z     using relTime       = std::uint32_t;    
2020-04-02T02:37:26.0038360Z                           ^
2020-04-02T02:37:26.0063700Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:31:5: error: unknown type name 'constexpr'
2020-04-02T02:37:26.0063900Z     constexpr Time time1microsec    = 1;    
2020-04-02T02:37:26.0063990Z     ^
2020-04-02T02:37:26.0067020Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:31:19: error: expected ';' after top level declarator
2020-04-02T02:37:26.0067200Z     constexpr Time time1microsec    = 1;    
2020-04-02T02:37:26.0067300Z                   ^
2020-04-02T02:37:26.0067380Z                   ;
2020-04-02T02:37:26.0079220Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:32:5: error: unknown type name 'constexpr'
2020-04-02T02:37:26.0079860Z     constexpr Time time1millisec    = 1000;
2020-04-02T02:37:26.0079960Z     ^
2020-04-02T02:37:26.0080610Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:32:19: error: expected ';' after top level declarator
2020-04-02T02:37:26.0080760Z     constexpr Time time1millisec    = 1000;
2020-04-02T02:37:26.0080850Z                   ^
2020-04-02T02:37:26.0080940Z                   ;
2020-04-02T02:37:26.0091950Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:33:5: error: unknown type name 'constexpr'
2020-04-02T02:37:26.0092130Z     constexpr Time time1second      = 1000000;
2020-04-02T02:37:26.0092230Z     ^
2020-04-02T02:37:26.0092870Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:33:19: error: expected ';' after top level declarator
2020-04-02T02:37:26.0093030Z     constexpr Time time1second      = 1000000;
2020-04-02T02:37:26.0093120Z                   ^
2020-04-02T02:37:26.0093210Z                   ;
2020-04-02T02:37:26.0102940Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:34:5: error: unknown type name 'constexpr'
2020-04-02T02:37:26.0103100Z     constexpr Time time1minute      = 60*time1second;
2020-04-02T02:37:26.0103360Z     ^
2020-04-02T02:37:26.0104020Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:34:19: error: expected ';' after top level declarator
2020-04-02T02:37:26.0104220Z     constexpr Time time1minute      = 60*time1second;
2020-04-02T02:37:26.0104330Z                   ^
2020-04-02T02:37:26.0104420Z                   ;
2020-04-02T02:37:26.0113840Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:35:5: error: unknown type name 'constexpr'
2020-04-02T02:37:26.0114010Z     constexpr Time time1hour        = 60*time1minute;
2020-04-02T02:37:26.0114100Z     ^
2020-04-02T02:37:26.0114770Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:35:19: error: expected ';' after top level declarator
2020-04-02T02:37:26.0114930Z     constexpr Time time1hour        = 60*time1minute;
2020-04-02T02:37:26.0115030Z                   ^
2020-04-02T02:37:26.0115120Z                   ;
2020-04-02T02:37:26.0125150Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:36:5: error: unknown type name 'constexpr'
2020-04-02T02:37:26.0125330Z     constexpr Time time1day         = 24*time1hour;
2020-04-02T02:37:26.0125440Z     ^
2020-04-02T02:37:26.0126320Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:36:19: error: expected ';' after top level declarator
2020-04-02T02:37:26.0126510Z     constexpr Time time1day         = 24*time1hour;
2020-04-02T02:37:26.0126620Z                   ^
2020-04-02T02:37:26.0126700Z                   ;
2020-04-02T02:37:26.0135930Z /Users/runner/runners/2.165.2/work/robotology-superbuild/robotology-superbuild/robotology/icub-firmware-shared/embot/core/embot_core.h:37:5: error: unknown type name 'constexpr'
[... cut ...]
2020-04-02T02:11:22.3130421Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/robotology/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:40:0:
2020-04-02T02:11:22.3131186Z /home/runner/work/robotology-superbuild/robotology-superbuild/robotology/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.h:27:10: fatal error: iCub/FactoryInterface.h: No such file or directory
2020-04-02T02:11:22.3131577Z  #include <iCub/FactoryInterface.h>
2020-04-02T02:11:22.3132465Z           ^~~~~~~~~~~~~~~~~~~~~~~~~
2020-04-02T02:11:22.3132776Z compilation terminated.
~~~

traversaro, 2020-04-02:
Interestingly, all the errors are actually different.  The Linux one is probably the most critical, as it means that the unstable branches are not compilable on the robot as of now (but it can be easily reproducible by enabling the compilation of the `canBusMotionControl` device in icub-main . 

The macOS is probably due to the fact that icub-firmware-shared uses feature of C++14, but does not declare in the CMake that C++14 should be used for the compilation.

The Windows failures are useful because they highlight some linking problems that are present also in macOS and Linux if those libraries are linked on an executable  (for example, the `feat_yarp_time_now` symbol does not seem to be defined anywhere). In Windows the linking process just fails earlier.

traversaro, 2020-04-02:
@Nicogene if you prefer I can open separate issue for each failure on icub-firmware-shared and icub-main .

Nicogene, 2020-04-02:
The Ubuntu build should be fixed by: https://github.com/robotology/icub-main/pull/650
The macOs build should be fixed by: https://github.com/robotology/icub-firmware-shared/commit/1c11d350353fb7db71c497f58f410ffb8b294bd1

Working on Win32

traversaro, 2020-04-02:
>  The macOs build should be fixed by: robotology/icub-firmware-shared@1c11d35

Not really, sorry. : ) 
As the C++14 features are used in the headers, you don't need to specify the C++ standard used for compiling the library itself, because that could create problems if the library is included in a bigger project via `add_subdirectory`, and because that information is not propagated to downstream projects if the library is used via `find_package(icub_firmware_shared)`. 
A more clean solution is to let know to CMake that you library **needs** C++14 features, but it can be compiled also as C++17/20, and that this requirement is propagated to downstream projects as the C++14 features are used in the library headers. For doing that, you can simply use the [`target_compile_features`](https://cmake.org/cmake/help/v3.12/command/target_compile_features.html) command: 
~~~
target_compile_features(${LIBRARY_TARGET_NAME} PUBLIC cxx_std_14)
~~~

Nicogene, 2020-04-02:
Unfortunately  https://github.com/robotology/icub-firmware-shared/commit/150815040813a6156fdf686620ec2ec31fa0609b is not enough, the linking problem in Win32 is present also in macOs:

```
Check dependencies

Ld /Users/nicologenesio/robotology/robotology-superbuild/build/robotology/icub-firmware-shared/eth/Debug/libembobj.1.15.100.dylib normal x86_64
    cd /Users/nicologenesio/robotology/robotology-superbuild/robotology/icub-firmware-shared
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -target x86_64-apple-macos10.15 -dynamiclib -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk -L/Users/nicologenesio/robotology/robotology-superbuild/build/robotology/icub-firmware-shared/eth/Debug -F/Users/nicologenesio/robotology/robotology-superbuild/build/robotology/icub-firmware-shared/eth/Debug -filelist /Users/nicologenesio/robotology/robotology-superbuild/build/robotology/icub-firmware-shared/eth/icub_firmware_shared.build/Debug/embobj.build/Objects-normal/x86_64/embobj.1.15.100.LinkFileList -install_name @rpath/libembobj.1.15.100.dylib -Xlinker -object_path_lto -Xlinker /Users/nicologenesio/robotology/robotology-superbuild/build/robotology/icub-firmware-shared/eth/icub_firmware_shared.build/Debug/embobj.build/Objects-normal/x86_64/embobj.1.15.100_lto.o -Xlinker -no_deduplicate -dynamiclib -Wl,-headerpad_max_install_names -current_version 1.15.100 -Xlinker -dependency_info -Xlinker /Users/nicologenesio/robotology/robotology-superbuild/build/robotology/icub-firmware-shared/eth/icub_firmware_shared.build/Debug/embobj.build/Objects-normal/x86_64/embobj.1.15.100_dependency_info.dat -o /Users/nicologenesio/robotology/robotology-superbuild/build/robotology/icub-firmware-shared/eth/Debug/libembobj.1.15.100.dylib
Undefined symbols for architecture x86_64:
  "_ace_mutex_new", referenced from:
      _eoy_mutex_New in EOYmutex.o
  "_ace_mutex_release", referenced from:
      _s_eoy_mutex_release in EOYmutex.o
  "_ace_mutex_take", referenced from:
      _s_eoy_mutex_take in EOYmutex.o
  "_feat_yarp_time_now", referenced from:
      _eoy_sys_Initialise in EOYtheSystem.o
      _s_eoy_sys_abstime_get in EOYtheSystem.o
      _s_eoy_sys_nanotime_get in EOYtheSystem.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

BUILD FAILED

```

traversaro, 2020-04-02:
As discussed on Teams,  for the time being the easiest solution is to temporary roll back the superbuild to use working commits of icub-main and  icub-firmware-shared before the offending PRs https://github.com/robotology/icub-firmware-shared/pull/35 and https://github.com/robotology/icub-main/pull/649 . Unfortunately the history of the repo is quite messy, so @Nicogene  could you check also you that https://github.com/robotology/icub-firmware-shared/commit/41bf5c1db3aa83808c401f608932373fcbba944d and https://github.com/robotology/icub-main/commit/4839050fde87ddbf3572b98d608f74097aacbfd6 should be the last working commits? Thanks!

Nicogene, 2020-04-02:
> Unfortunately the history of the repo is quite messy, so @Nicogene could you check also you that robotology/icub-firmware-shared@41bf5c1 and robotology/icub-main@4839050 should be the last working commits? Thanks!

Yes they should the last commits before the migration. Lets see if CI passes in https://github.com/robotology/robotology-superbuild/pull/373

Nicogene, 2020-04-14:
@traversaro the PRs have been merged, they compile fine on both Windows and macOs, I think you can re-enable the latest devel commit.

Let's close this as soon as the build on the unstable branch passes.

Nicogene, 2020-04-15:
This has been fixed, sorry for the troubles.

closed 2020-04-15 by Nicogene