robotology/robotology-superbuild #383: robotology-superbuild CMake configuration error on Debian Sid

traversaro, 2020-04-24:
By installing Gazebo via `sudo apt install libgazebo*-dev` on Debian sid (see https://github.com/robotology/robotology-superbuild/runs/615946435),  the configuration fails with the following error: 
~~~
-- Found DL: TRUE  
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/ignition-common3/ignition-common3-config.cmake:94 (find_package):
  By not providing "Findtinyobjloader.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "tinyobjloader", but CMake did not find one.

  Could not find a package configuration file provided by "tinyobjloader"
  with any of the following names:

    tinyobjloaderConfig.cmake
    tinyobjloader-config.cmake

  Add the installation prefix of "tinyobjloader" to CMAKE_PREFIX_PATH or set
  "tinyobjloader_DIR" to a directory containing one of the above files.  If
  "tinyobjloader" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:209 (find_package)
  cmake/BuildGazeboYARPPlugins.cmake:31 (find_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:235 (include)
  CMakeLists.txt:130 (find_or_build_package)
~~~

fyi @j-rivero

traversaro, 2020-04-24:
At a first glance, it seems that libignition-common-dev ( https://packages.debian.org/sid/libignition-common-dev ) is missing a dependency on libtinyobjloader-dev  ( https://packages.debian.org/sid/libtinyobjloader-dev ).

traversaro, 2020-09-15:
Thanks a lot @jspricke, we are testing with the latest docker image of debian:sid in https://github.com/robotology/robotology-superbuild/pull/448 and the problem is indeed solved!

traversaro, 2020-10-01:
Fixed by https://salsa.debian.org/science-team/ignition-common/-/commit/872e3013cbde7815445d71ac75464b441647946f .

closed 2020-10-01 by traversaro

j-rivero, 2020-10-01:
thanks Silvio, Jochen