robotology/robotology-superbuild #417: Unable to compile on Ubuntu 16.04

moongerim, 2020-05-27:
Hello! I am follow this guide. I need iDynTree, therefore I set up the ccmake environment by setting ROBOTOLOGY_ENABLE_DYNAMICS to the "ON" position. After that, I could not compile it. It returns the folowing error:

plugins/jointsensors/CMakeFiles/gazebo_yarp_jointsensors.dir/build.make:95: recipe for target 'plugins/jointsensors/CMakeFiles/gazebo_yarp_jointsensors.dir/src/JointSensorsDriver.cpp.o' failed
make[5]: *** [plugins/jointsensors/CMakeFiles/gazebo_yarp_jointsensors.dir/src/JointSensorsDriver.cpp.o] Error 1
CMakeFiles/Makefile2:1067: recipe for target 'plugins/jointsensors/CMakeFiles/gazebo_yarp_jointsensors.dir/all' failed
make[4]: *** [plugins/jointsensors/CMakeFiles/gazebo_yarp_jointsensors.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/GazeboYARPPlugins.dir/build.make:127: recipe for target 'robotology/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-build' failed
make[2]: *** [robotology/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-build] Error 2
CMakeFiles/Makefile2:1944: recipe for target 'CMakeFiles/GazeboYARPPlugins.dir/all' failed
make[1]: *** [CMakeFiles/GazeboYARPPlugins.dir/all] Error 2
Makefile:114: recipe for target 'all' failed
make: *** [all] Error 2

traversaro, 2020-05-28:
Hi @moongerim, in general we follow the distributions using the policy described in http://wiki.icub.org/wiki/YARP_Supported_Distributions, so at the moment we do not support explicitly (i.e. we do not check in CI and we do not try to work fine with system dependencies) Ubuntu 16.04 , so I suggest to update at least at Ubuntu 18.04 to get something working out of the box. 

However, if you install all the correct dependencies and in some cases if you update the compiler, the software **may** compile correctly, but that is not ensured in any way. In this case, in the error that you reported it is missing the actual compiler error, so you should provide all the relevant error lines to simplify suggesting you what could be the problem.

traversaro, 2020-05-28:
> Hello! I am follow this guide. I need iDynTree, therefore I set up the ccmake environment by setting ROBOTOLOGY_ENABLE_DYNAMICS to the "ON" position. After that, I could not compile it. It returns the folowing error:

Note that I just read that you just need iDynTree, probably in that case it should be sufficient that you install the necessary dependencies (the most basic ones are `libeigen3-dev` and `libxml2-dev`) and just install iDynTree directly without passing through the robotology-superbuild .

moongerim, 2020-05-28:
Hello, @traversaro 
Thank you, for your reply! Yes, I tried to install only iDynTree. It was compiled, but I could not find there the necessary files. Actually, what I need is this:
https://github.com/dic-iit/human-model-generator 
They asked to install the whole robotology in order to have the Matlab buildings. When I installed only iDynTree, there was no Matlab buildings. 

When I want to build robotology, it returns the following error: 
'
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found YARP: /home/robot/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.2+43-20200411.3+gita33cf54")
-- Using iCub from install
-- Found YARP: /home/robot/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.2+43-20200411.3+gita33cf54")
-- Found YARP: /home/robot/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.2+43-20200411.3+gita33cf54")
-- Found YARP: /home/robot/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.2+43-20200411.3+gita33cf54")
-- Found YARP: /home/robot/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.2+43-20200411.3+gita33cf54")
-- Created target YarpUtilities for export WalkingControllers.
-- Created target iDynTreeUtilities for export WalkingControllers.
-- Created target TimeProfiler for export WalkingControllers.
-- Created target StdUtilities for export WalkingControllers.
-- Created target SimplifiedModelControllers for export WalkingControllers.
-- Created target RobotInterface for export WalkingControllers.
-- Created target WholeBodyControllers for export WalkingControllers.
-- Created target TrajectoryPlanner for export WalkingControllers.
-- Created target KinDynWrapper for export WalkingControllers.
-- Created target RetargetingHelper for export WalkingControllers.
-- Created target LoggerClient for export WalkingControllers.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/robot/robotology-superbuild/build/robotology/walking-controllers
[ 93%] Performing build step for 'walking-controllers'
Scanning dependencies of target YarpUtilities
[  2%] Building CXX object src/YarpUtilities/CMakeFiles/YarpUtilities.dir/src/Helper.cpp.o
[  5%] Linking CXX shared library ../../lib/libWalkingControllersYarpUtilities.so
[  5%] Built target YarpUtilities
Scanning dependencies of target iDynTreeUtilities
[  7%] Building CXX object src/iDynTreeUtilities/CMakeFiles/iDynTreeUtilities.dir/src/Helper.cpp.o
[ 10%] Linking CXX shared library ../../lib/libWalkingControllersiDynTreeUtilities.so
[ 10%] Built target iDynTreeUtilities
Scanning dependencies of target TimeProfiler
[ 12%] Building CXX object src/TimeProfiler/CMakeFiles/TimeProfiler.dir/src/TimeProfiler.cpp.o
[ 15%] Linking CXX shared library ../../lib/libWalkingControllersTimeProfiler.so
[ 15%] Built target TimeProfiler
Scanning dependencies of target SimplifiedModelControllers
[ 17%] Building CXX object src/SimplifiedModelControllers/CMakeFiles/SimplifiedModelControllers.dir/src/DCMModelPredictiveController.cpp.o
[ 20%] Building CXX object src/SimplifiedModelControllers/CMakeFiles/SimplifiedModelControllers.dir/src/DCMReactiveController.cpp.o
[ 23%] Building CXX object src/SimplifiedModelControllers/CMakeFiles/SimplifiedModelControllers.dir/src/MPCSolver.cpp.o
[ 25%] Building CXX object src/SimplifiedModelControllers/CMakeFiles/SimplifiedModelControllers.dir/src/ZMPController.cpp.o
[ 28%] Linking CXX shared library ../../lib/libWalkingControllersSimplifiedModelControllers.so
[ 28%] Built target SimplifiedModelControllers
Scanning dependencies of target RobotInterface
[ 30%] Building CXX object src/RobotInterface/CMakeFiles/RobotInterface.dir/src/Helper.cpp.o
[ 33%] Building CXX object src/RobotInterface/CMakeFiles/RobotInterface.dir/src/PIDHandler.cpp.o
[ 35%] Linking CXX shared library ../../lib/libWalkingControllersRobotInterface.so
[ 35%] Built target RobotInterface
Scanning dependencies of target WholeBodyControllers
[ 38%] Building CXX object src/WholeBodyControllers/CMakeFiles/WholeBodyControllers.dir/src/InverseKinematics.cpp.o
[ 41%] Building CXX object src/WholeBodyControllers/CMakeFiles/WholeBodyControllers.dir/src/QPInverseKinematics.cpp.o
[ 43%] Building CXX object src/WholeBodyControllers/CMakeFiles/WholeBodyControllers.dir/src/QPInverseKinematics_osqp.cpp.o
[... 220 lines cut ...]
[ 17%] Built target gazebo_yarp_clock
[ 37%] Built target gazebo_yarp_controlboard
[ 48%] Built target gazebo_yarp_fakecontrolboard
[ 51%] Built target gazebo_yarp_forcetorque
[ 54%] Built target gazebo_yarp_imu
[ 56%] Built target gazebo_yarp_inertial
[ 59%] Built target gazebo_yarp_distributedinertials
[ 60%] Building CXX object plugins/jointsensors/CMakeFiles/gazebo_yarp_jointsensors.dir/src/JointSensorsDriver.cpp.o
[... cut ...]
Makefile:149: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/GazeboYARPPlugins.dir/build.make:127: recipe for target 'robotology/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-build' failed
make[2]: *** [robotology/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-build] Error 2
CMakeFiles/Makefile2:1761: recipe for target 'CMakeFiles/GazeboYARPPlugins.dir/all' failed
make[1]: *** [CMakeFiles/GazeboYARPPlugins.dir/all] Error 2
Makefile:114: recipe for target 'all' failed
make: *** [all] Error 2
`

traversaro, 2020-05-28:
Ok, the real compilation problem is: 
~~~
/home/robot/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/jointsensors/src/JointSensorsDriver.cpp:93:16: error: ‘class std::mutex’ has no member named ‘wait’
93 | data_mutex.wait();
~~~
that is not easy to fix. However, if you don't need gazebo, you can  just disable the `ROBOTOLOGY_USES_GAZEBO` . 

If you need the iDynTree MATLAB bindings, you can either compile the robotology-superbuild by enabling the `ROBOTOLOGY_USES_MATLAB` CMake option, or compile iDynTree by enabling the `IDYNYTREE_USES_MATLAB`, as documented in https://github.com/robotology/idyntree#bindings .

moongerim, 2020-05-28:
Thank you a lot! You solved my issue!!!!!

traversaro, 2020-06-04:
Based on your reply, I think we can close this issue.

closed 2020-06-04 by traversaro