robotology/robotology-superbuild #418: WBToolbox fails to compile in a clean installation

S-Dafarra, 2020-06-04:
The error is
```
[ 44%] Building CXX object toolbox/library/CMakeFiles/WBToolbox.dir/src/SimulatorSynchronizer.cpp.o
make[5]: *** No rule to make target '/usr/local/src/robot/robotology-superbuild/build/install/lib/libqpOASES.so', needed by 'lib/libWBToolbox.so'.  Stop.
CMakeFiles/Makefile2:254: recipe for target 'toolbox/library/CMakeFiles/WBToolbox.dir/all' failed
make[4]: *** [toolbox/library/CMakeFiles/WBToolbox.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/WBToolbox.dir/build.make:110: recipe for target 'robotology/WBToolbox/CMakeFiles/YCMStamp/WBToolbox-build' failed
make[2]: *** [robotology/WBToolbox/CMakeFiles/YCMStamp/WBToolbox-build] Error 2
CMakeFiles/Makefile2:2764: recipe for target 'CMakeFiles/WBToolbox.dir/all' failed
make[1]: *** [CMakeFiles/WBToolbox.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
```
It is necessary to compile ``qpOases`` first, install it , and then it works.

A simple way to reproduce this is to delete the install folder and try to recompile.

traversaro, 2020-06-04:
One thing that I don't understand is how you were able to configure WB-Toolbox if qpOASES was not installed.

S-Dafarra, 2020-06-04:
> On which OS are you experiencing this problem?

Sorry, Ubuntu 18.04. It happened when trying to update the iCub head and the laptop. In both cases I was using the ``master`` version of the robotology superbuild. The thing I thought it was fishy is that it should not try to compile ``WBToolbox`` if ``qpOases`` was not built before right?

traversaro, 2020-06-04:
The superbuild is up-to-dated with the latest commit?

I think the following happened: 
* For some reason, the qpOASES of the superbuild was found by the superbuild itself (you can check this by inspecting the `CMakeCache.txt` of the superbuild), this should be avoided in the future as  soon as the next YCM version is released thanks to https://github.com/robotology/robotology-superbuild/pull/389 .  
* On the top of that,  somebody canceled the installed file `/usr/local/src/robot/robotology-superbuild/build/install/lib/libqpOASES.so`

Did you experience this in a clean build?

S-Dafarra, 2020-06-04:
> The superbuild is up-to-dated with the latest commit?

Yes, https://github.com/robotology/robotology-superbuild/commit/1db39dfab3f89e3402c571c31ca18900c09c3322

> On the top of that, somebody canceled the installed file /usr/local/src/robot/robotology-superbuild/build/install/lib/libqpOASES.so

I did that :grin: Actually, I did it because at the beginning it did not seem to compile ``yarp`` correctly.

traversaro, 2020-06-04:
I think you should  check your `CMakeCache.txt` (of the superbuild) and remove any reference to qpOASES, that should ensure that qpOASES is configured and  installed before the WB-Toolbox.

S-Dafarra, 2020-06-04:
Ok great! Anyways, this was just to notify. By going into the ``qpOases`` build folder and install it again fixes the problem.

closed 2020-06-04 by S-Dafarra