robotology/robotology-superbuild #420: macOS builds are failing

traversaro, 2020-06-08:
The error is: 
~~~
/usr/local/Cellar/gazebo10/10.2.0_3/include/gazebo-10/gazebo/msgs/camerasensor.pb.h:54:51: error: no type named 'AuxillaryParseTableField' in namespace 'google::protobuf::internal'; did you mean 'AuxiliaryParseTableField'?
  static const ::PROTOBUF_NAMESPACE_ID::internal::AuxillaryParseTableField aux[]
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/google/protobuf/generated_message_table_driven.h:141:7: note: 'AuxiliaryParseTableField' declared here
union AuxiliaryParseTableField {
~~~
probably it is a Gazebo/Homebrew regression.

traversaro, 2020-06-09:
Upstream bug: :https://github.com/osrf/homebrew-simulation/issues/1042 .

nunoguedelha, 2020-06-15:
#### Side info on https://github.com/osrf/homebrew-simulation/issues/1042:
> Apparently a symbol was renamed in protobuf's most recent patch release protocolbuffers/protobuf#7496 (merged into homebrew in Homebrew/homebrew-core#55750), so we'll need to rebuild the ignition-msgs* bottles, but I think downstream bottles won't need to be rebuilt.

#### The problem occurs in two places:
As far as I understood, it occurs:
- in the ignition-msgs* dependencies
bottled **ignition-msgs1 release 1.0.0_8** and **ignition-msgs5 release 5.3.0_2** have already the respective fixes https://github.com/osrf/homebrew-simulation/pull/1045 and https://github.com/osrf/homebrew-simulation/pull/1043.

A way to check you have the fix in your release is doing:
```
> brew test ignition-msgs1
> brew test ignition-msgs5
```
should return no error.

- but also in the Gazebo code itself.

The solution is either to install Gazebo9 from source:
```
brew install --build-from-source gazebo9
```
Either to install Gazebo 11 from homebrew:
```
brew install gazebo11
```
Indeed Gazebo 11 release 11.0.0_2 has the fix.

I'm trying to rebuild the superbuild now...

nunoguedelha, 2020-06-15:
CC @kouroshD , @lrapetti , @claudia-lat , @DanielePucci

traversaro, 2020-06-26:
For the CI, I fixed by migrating everything to Gazebo11, as Gazebo9/10 remains broken. Should we recommend to install Gazebo 11 on macOS in the installation instructions and close this issue?

traversaro, 2020-06-29:
> 
> 
> For the CI, I fixed by migrating everything to Gazebo11, as Gazebo9/10 remains broken. Should we recommend to install Gazebo 11 on macOS in the installation instructions and close this issue?

I guess that it is ok. If anyone feel strongly otherwise, feel free to comment so we can re-open the issue.

closed 2020-06-29 by traversaro

nunoguedelha, 2020-06-29:
> For the CI, I fixed by migrating everything to Gazebo11, as Gazebo9/10 remains broken. Should we recommend to install Gazebo 11 on macOS in the installation instructions and close this issue?

Yes, sure @traversaro , sorry for the delay. Although I'm still having issues running Gazebo11 on Catalina.

nunoguedelha, 2020-06-29:
I'll open a separate issue for that if necessary.

traversaro, 2020-06-29:
> I'll open a separate issue for that if necessary.

Thanks, that  would be ideal. This issue was  mostly related to the macOS builds failure in CI, and those were fixed by migrating to Gazebo 11.