robotology/robotology-superbuild #426: devel branches are not compiling correctly on Ubuntu 18.04/20.04

traversaro, 2020-07-16:
~~~
2020-07-16T04:16:21.5109220Z [ 40%] Building CXX object plugins/fakecontrolboard/CMakeFiles/gazebo_yarp_fakecontrolboard.dir/src/FakeControlBoard.cc.o
2020-07-16T04:16:24.0896004Z In file included from /home/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/install/include/yarp/dev/ImplementControlCalibration.h:14:0,
2020-07-16T04:16:24.0898152Z                  from /home/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/install/include/yarp/dev/ControlBoardInterfacesImpl.h:26,
2020-07-16T04:16:24.0898921Z                  from /home/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/plugins/fakecontrolboard/include/yarp/dev/FakeControlBoardDriver.h:15,
2020-07-16T04:16:24.0899619Z                  from /home/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/plugins/fakecontrolboard/src/FakeControlBoard.cc:8:
2020-07-16T04:16:24.0901101Z /home/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/install/include/yarp/dev/ControlBoardHelper.h:23:16: error: ‘std::enable_if_t’ has not been declared
2020-07-16T04:16:24.0901596Z            std::enable_if_t<std::is_trivial<T>::value, int> = 0>
2020-07-16T04:16:24.0902633Z                 ^~~~~~~~~~~
2020-07-16T04:16:24.0903356Z /home/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/install/include/yarp/dev/ControlBoardHelper.h:23:27: error: expected ‘>’ before ‘<’ token
2020-07-16T04:16:24.0903849Z            std::enable_if_t<std::is_trivial<T>::value, int> = 0>
2020-07-16T04:16:24.0904176Z                            ^
2020-07-16T04:16:24.0905317Z /home/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/install/include/yarp/dev/ControlBoardHelper.h:33:16: error: ‘std::enable_if_t’ has not been declared
2020-07-16T04:16:24.0905783Z            std::enable_if_t<!std::is_trivial<T>::value, int> = 0>
2020-07-16T04:16:24.0907298Z                 ^~~~~~~~~~~
2020-07-16T04:16:24.0908399Z /home/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/install/include/yarp/dev/ControlBoardHelper.h:33:27: error: expected ‘>’ before ‘<’ token
2020-07-16T04:16:24.0908775Z            std::enable_if_t<!std::is_trivial<T>::value, int> = 0>
2020-07-16T04:16:24.0909022Z                            ^
2020-07-16T04:16:24.0910011Z /home/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/install/include/yarp/dev/ControlBoardHelper.h:34:4: error: redefinition of ‘template<class T, <typeprefixerror><anonymous> > T* allocAndCheck(int)’
2020-07-16T04:16:24.0910344Z  T* allocAndCheck(int size)
2020-07-16T04:16:24.0910583Z     ^~~~~~~~~~~~~
2020-07-16T04:16:24.0911363Z /home/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/install/include/yarp/dev/ControlBoardHelper.h:24:4: note: ‘template<class T, <typeprefixerror><anonymous> > T* allocAndCheck(int)’ previously declared here
2020-07-16T04:16:24.0911677Z  T* allocAndCheck(int size)
2020-07-16T04:16:24.0911885Z     ^~~~~~~~~~~~~
2020-07-16T04:16:26.6032764Z plugins/fakecontrolboard/CMakeFiles/gazebo_yarp_fakecontrolboard.dir/build.make:79: recipe for target 'plugins/fakecontrolboard/CMakeFiles/gazebo_yarp_fakecontrolboard.dir/src/FakeControlBoard.cc.o' failed
2020-07-16T04:16:26.6035826Z make[2]: *** [plugins/fakecontrolboard/CMakeFiles/gazebo_yarp_fakecontrolboard.dir/src/FakeControlBoard.cc.o] Error 1
2020-07-16T04:16:26.6036361Z CMakeFiles/Makefile2:977: recipe for target 'plugins/fakecontrolboard/CMakeFiles/gazebo_yarp_fakecontrolboard.dir/all' failed
2020-07-16T04:16:26.6039977Z make[1]: *** [plugins/fakecontrolboard/CMakeFiles/gazebo_yarp_fakecontrolboard.dir/all] Error 2
2020-07-16T04:16:26.6040355Z Makefile:146: recipe for target 'all' failed
2020-07-16T04:16:26.6051887Z make: *** [all] Error 2
~~~

Probably it is something due to  C++14 being enabled in YARP and Gazebo overwriting this option by specifying manually `-std=c++11`.

Nicogene, 2020-07-16:
`enable_if_t` is in c++14 see https://en.cppreference.com/w/cpp/types/enable_if, `set(CMAKE_CXX_STANDARD 14)` in `gazebo-yarp-plugins` isn't sufficient?

traversaro, 2020-07-16:
> 
> 
> `enable_if_t` is in c++14 see https://en.cppreference.com/w/cpp/types/enable_if, `set(CMAKE_CXX_STANDARD 14)` in `gazebo-yarp-plugins` isn't sufficient?

The compilaton of C++14 should be already passed by the fact that this feature is exported as public in YARP_dev: https://github.com/robotology/yarp/blob/e45ef0fb265c4f358f10f17679107870e1a332d3/src/libYARP_dev/src/CMakeLists.txt#L268  . However probably this line https://github.com/robotology/gazebo-yarp-plugins/blob/master/CMakeLists.txt#L51 is interfering with that because `GAZEBO_CXX_FLAGS` is containing explicitly `-std=c++11`. Probably removing https://github.com/robotology/gazebo-yarp-plugins/blob/master/CMakeLists.txt#L52 would help.

drdanz, 2020-07-16:
All YARP targets force c++14 in their public interface

```cmake
target_compile_features(YARP_os PUBLIC cxx_std_14)
```

traversaro, 2020-07-16:
> All YARP targets force c++14 in their public interface

Exactly, the problem is that Gazebo uses legacy compilation flag to request instead C++11 .

drdanz, 2020-07-16:
I can revert that change, if this causes issues, since we are very close to the release, and apply it back on master after the release...
Anyway YARP requires C++14, if Gazebo has problems with that, it should probably be fixed in gazebo, because it sounds a lot like a bug to me

traversaro, 2020-07-16:
> 
> 
> I can revert that change, if this causes issues, since we are very close to the release, and apply it back on master after the release...
> Anyway YARP requires C++14, if Gazebo has problems with that, it should probably be fixed in gazebo, because it sounds a lot like a bug to me

That is definitely not a problem of YARP, I would avoid to spend time in trying to find a solution at the YARP level, we will fix it here at the `gazebo-yarp-plugins` level.

traversaro, 2020-07-16:
Workaround provided in https://github.com/robotology/gazebo-yarp-plugins/pull/500 . A proper fix for Gazebo similar to https://github.com/osrf/sdformat/pull/251 would be appropriate, but in any case we need a workaround for all the Gazebo releases prior to the fix.

closed 2020-07-17 by traversaro