robotology/robotology-superbuild #427: Add docs on how to setup the GUIs on WSL2

traversaro, 2020-07-17:
Using WSL2, setting GUIs is slightly more complicated, as the WSL2 virtual machine and the Windows host now have two different IPs. 
* It is necessary to appropriately set the X11's `DISPLAY` env variable, see https://discourse.ubuntu.com/t/getting-graphical-applications-to-work-on-wsl2/11868 
* Depending on your X server, you may need to explicitly allow the IP of the WSL2 virtual machine to connect to the X server, for example for Xming https://theunixtips.com/xming-client-4-rejected-from-ip/ . 

We should document this. 

cc @Nicogene

Nicogene, 2020-07-17:
+1000 since I am encountering issues using guis with wsl2 😬 
With wsl1 I had no issues

pattacini, 2020-07-17:
A few notes from my experience while setting up WSL2:
- I've enabled the option **Allow Public Acess** from within **X410** (I don't use Xming that is free though).
- Here are a bunch of lines in my `.bashrc` that serve to address the X11 and YARP configuration:
    ```sh
    export WSL_HOST=$(grep nameserver /etc/resolv.conf | awk '{print $2}')
    export DISPLAY=${WSL_HOST}:0.0
    dbus-launch --exit-with-x11 > /dev/null 2>&1
    # ...
    yarp conf ${WSL_HOST} 10000 > /dev/null 2>&1
    ```

traversaro, 2020-07-17:
> yarp conf ${WSL_HOST} 10000 > /dev/null 2>&1

This line is because you launch the yarpserver on `WSL2` and not on Windows, right?

traversaro, 2020-07-17:
Just to understand, why the `dbus-launch` is necessary?

pattacini, 2020-07-17:
>This line is because you launch the yarpserver on WSL2 and not on Windows, right?

I do the opposite. `${WSL_HOST}` is the IP of Windows as read from WSL2, which may vary.

>Just to understand, why the dbus-launch is necessary?

I've just reported from my previous WSL1 installation. Don't know if it is still necessary, actually.

pattacini, 2020-07-17:
Removing the call to `dbus-launch` didn't bring about any problem 👍

Nicogene, 2020-07-17:
My issue using `Xming` was the following:

```
Xming.exe: client 4 rejected from IP XXX.XXX.XXX.XXX
```

But I solved it following this very useful link, that may be included in the documentation https://theunixtips.com/xming-client-4-rejected-from-ip/

Nicogene, 2020-07-17:
What about the nvidia/gpu drivers for the wsl?
Anyone has been able to enable them in wsl2 ?

pattacini, 2020-07-18:
A [video tutorial 🎥](https://youtu.be/IL7Jd9rjgrM) that may be helpful on setting up a GUI in WSL2.

traversaro, 2020-09-09:
[image]
[image]
[image]
[image]
[image]

closed 2020-09-10 by traversaro