robotology/robotology-superbuild #428: ROBOTOLOGY_USES_PYTHON: Missmatch in created PYTHONPATH entry and created directory

torstenfol, 2020-07-17:
When installing the superbuild with the ROBOTOLOGY_USES_PYTHON set to on, the binding files for Python are stored in robotology-superbuild/build/install/lib/**python3**, 
while in the setup.sh script the following entries are created:
export PYTHONPATH=${PYTHONPATH:+${PYTHONPATH}:}${ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX}/lib/**python3.8**/dist-packages
export PYTHONPATH=${PYTHONPATH}:${ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX}/lib/**python3.8**/site-packages.

Shouldn't this be consistent? Because in this way, the module couldn't be loaded and the path has to be fixed manually.

traversaro, 2020-07-18:
Thanks @torstenfol , indeed there is a problem there. The downstream projects typically compute the installation directory for python files  in a programmatic way (see for example https://github.com/robotology/idyntree/blob/master/bindings/python/CMakeLists.txt#L28), while in the superbuild we hardcode it to a given value, while we should try to have the same logic used by the downstream projects (see https://github.com/robotology/robotology-superbuild/pull/359/files for a related PR). 

Just to understand, on which system are you compiling the robotology-superbuild in which the python files get installed in `robotology-superbuild/build/install/lib/python3` ?

torstenfol, 2020-07-20:
I tested it on Ubuntu 20.04 (Python 3.8.2) and on Linux Mint 20 (Python 3.8.2).

In the search for the cause of this behavior I came across this:
The path problem depends on the different sources for the directories. For the environment variable the CMake variable "PYTHON_VERSION_PATH"  is used, while in the bindings installation the sysconfig module from the distutils is used. For Python 3 get_python_lib(1,0,prefix='') returns "lib/python3/dist-packages" while for Python 2 this returns "lib/python2.7/dist-packages".

traversaro, 2021-01-12:
We experienced the same problem with @Iaxama , so now I have a clear picture of the problem. I thought there was a difference between YARP and iDynTree, but actually they are both installed in `lib/python3/dist-packages`, so the only difference is the one that @torsten93 mentioned in the previous message.

closed 2021-01-12 by traversaro