robotology/robotology-superbuild #429: Windows CI is failing for

traversaro, 2020-07-22:
In Stable/Unstable tests the CI on Windows if failing with: 
~~~
2020-07-22T02:08:58.8885388Z   Performing download step (git clone) for 'BlockFactory'
2020-07-22T02:08:58.9223902Z   Cloning into 'BlockFactory'...
2020-07-22T02:08:59.5257779Z   Already on 'master'
2020-07-22T02:08:59.5258428Z   Your branch is up to date with 'origin/master'.
2020-07-22T02:09:00.1372948Z   Performing update step for 'BlockFactory'
2020-07-22T02:09:00.4275082Z   Current branch master is up to date.
2020-07-22T02:09:00.4500528Z   No stash entries found.
2020-07-22T02:09:00.4839345Z   No stash entries found.
2020-07-22T02:09:00.5172266Z   No stash entries found.
2020-07-22T02:09:00.5190856Z   CMake Error at D:/a/robotology-superbuild/robotology-superbuild/build/robotology/BlockFactory/CMakeFiles/YCMTmp/BlockFactory-gitupdate.cmake:136 (message):
2020-07-22T02:09:00.5191377Z     
2020-07-22T02:09:00.5191715Z   
2020-07-22T02:09:00.5193305Z     Failed to unstash changes in:
2020-07-22T02:09:00.5194197Z     'D:/a/robotology-superbuild/robotology-superbuild/robotology/BlockFactory/'.
2020-07-22T02:09:00.5194574Z   
2020-07-22T02:09:00.5194910Z   
2020-07-22T02:09:00.5195266Z     You will have to resolve the conflicts manually
2020-07-22T02:09:00.5195621Z   
2020-07-22T02:09:00.5195960Z   
~~~

While everything is working fine on the release 2020.08 . Nothing changed recently on BlockFactory, so this is probably a problem triggered by an environment change in GitHub Actions.

traversaro, 2020-07-24:
Given that: 
 * The failures occur just on branches and not on tags 
 * CMake was recently updated to CMake 3.18 on Windows GitHub actions images (see https://github.com/actions/virtual-environments/pull/1276/files)
 * CMake 3.18 did some modifications to the part of ExternalProjects that download projects.

The problem may be related to CMake 3.18. Another possible source of problem is due to the version of git, see https://gitlab.kitware.com/cmake/cmake/-/issues/21009, however that issue mentions quite hold versions of Git, so I don't think that is the problem.

traversaro, 2020-07-24:
On my system everything works fine on CMake 3.18.0 and git 2.24.0.windows.1
~~~
$ cmake --build .
Microsoft (R) Build Engine version 16.7.0-preview-20359-01+29609438b for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Creating directories for 'BlockFactory'
  Performing download step (git clone) for 'BlockFactory'
  Cloning into 'BlockFactory'...
  Your branch is up to date with 'origin/master'.
  Already on 'master'
  Skipping update step for 'BlockFactory'
  No patch step for 'BlockFactory'
  Performing configure step for 'BlockFactory'
  Not searching for unused variables given on the command line.
  loading initial cache file C:/src/robotology-superbuild/build/robotology/BlockFactory/CMakeFiles/YCMTmp/BlockFactory-cache-Debug.cmake
  -- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19041.
  -- The CXX compiler identification is MSVC 19.27.29016.0
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Preview/VC/Tools/MSVC/14.27.29016/bin/Hostx64/x64/cl.exe - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Found Matlab: C:/Program Files/MATLAB/R2020a/extern/include;C:/Program Files/MATLAB/R2020a/simulink/include (found version "9.8") found components: MX_LIBRARY ENG_LIBRARY MAIN_PROGRAM SIMULINK
  -- Configuring done
  -- Generating done
  -- Build files have been written to: C:/src/robotology-superbuild/build/robotology/BlockFactory
  Performing build step for 'BlockFactory'
  Microsoft (R) Build Engine version 16.7.0-preview-20359-01+29609438b for .NET Framework
  Copyright (C) Microsoft Corporation. All rights reserved.

    Checking Build System
    Building Custom Rule C:/src/robotology-superbuild/robotology/BlockFactory/deps/sharedlibpp/src/CMakeLists.txt
    version.cpp
    SharedLibrary.cpp
    SharedLibraryFactory.cpp
  Attempting to cancel the build...
~~~

traversaro, 2020-07-25:
An interesting aspect is that even if BlockFactory is the first repo being cloned, then also all this other repo in the superbuild are cloned correctly. As the problem seems that CMake seems some files modified that are not actually modified, and so tries to stash them failing (I had this problem in the past with repos with files names that were different in case sensitive file system but where the same on case insensitive file systems), I start to suspect that the filesystem in which the files are saved in GitHub Actions Windows changed.

traversaro, 2020-07-25:
Last working CI run (Jul 20 4 AM GMT): https://github.com/robotology/robotology-superbuild/actions/runs/175113710
First failing CI run (Jul 21 4 AM GMT): https://github.com/robotology/robotology-superbuild/actions/runs/176488552

traversaro, 2020-07-25:
The diff in the enviroment variables is: 
~~~
19c19
< ImageVersion=20200714.1
---
> ImageVersion=20200706.1
35c35
< RUNNER_TRACKING_ID=github_e3deb362-cb5c-4229-a1f0-a2c9b083f74a
---
> RUNNER_TRACKING_ID=github_4e96d6ff-88ce-472d-9081-db4c7cba6d16
52c52
< PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 79 Stepping 1, GenuineIntel
---
> PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 85 Stepping 4, GenuineIntel
74,75c74,75
< GITHUB_RUN_ID=176488552
< PROCESSOR_REVISION=4f01
---
> GITHUB_RUN_ID=175113710
> PROCESSOR_REVISION=5504
112c112
< GITHUB_RUN_NUMBER=622
---
> GITHUB_RUN_NUMBER=621
118c118
< PATH=/mingw64/bin:/usr/bin:/c/Users/runneradmin/bin:/c/Program Files/Mercurial:/c/Program Files/MongoDB/Server/4.2/bin:/c/ProgramData/kind:/c/vcpkg:/c/cf-cli:/c/Program Files (x86)/NSIS:/c/Program Files/Mercurial:/c/hostedtoolcache/windows/stack/2.3.1/x64:/c/ProgramData/chocolatey/lib/ghc.8.10.1/tools/ghc-8.10.1/bin:/c/Program Files/dotnet:/c/mysql-5.7.21-winx64/bin:/c/Program Files/Java/jdk8u252-b09/bin:/c/SeleniumWebDrivers/GeckoDriver:/c/Program Files (x86)/sbt/bin:/c/Rust/.cargo/bin:/c/Program Files (x86)/GitHub CLI:/bin:/c/hostedtoolcache/windows/go/1.14.4/x64/bin:/c/hostedtoolcache/windows/Python/3.7.8/x64/Scripts:/c/hostedtoolcache/windows/Python/3.7.8/x64:/c/hostedtoolcache/windows/Ruby/2.5.8/x64/bin:/c/npm/prefix:/c/Program Files (x86)/Microsoft SDKs/Azure/CLI2/wbin:/c/windows/system32:/c/windows:/c/windows/System32/Wbem:/c/windows/System32/WindowsPowerShell/v1.0:/c/windows/System32/OpenSSH:/c/ProgramData/Chocolatey/bin:/c/Program Files/Docker:/c/Program Files/PowerShell/7:/c/Program Files/dotnet:/c/Program Files/Microsoft SQL Server/130/Tools/Binn:/c/Program Files/Microsoft SQL Server/Client SDK/ODBC/170/Tools/Binn:/c/Program Files (x86)/Microsoft SQL Server/110/DTS/Binn:/c/Program Files (x86)/Microsoft SQL Server/120/DTS/Binn:/c/Program Files (x86)/Microsoft SQL Server/130/DTS/Binn:/c/Program Files (x86)/Microsoft SQL Server/140/DTS/Binn:/c/Program Files (x86)/Microsoft SQL Server/150/DTS/Binn:/c/Program Files (x86)/Windows Kits/10/Windows Performance Toolkit:/c/Program Files/Microsoft Service Fabric/bin/Fabric/Fabric.Code:/c/Program Files/Microsoft SDKs/Service Fabric/Tools/ServiceFabricLocalClusterManager:/c/Program Files/nodejs:/c/Program Files/OpenSSL/bin:/c/Strawberry/c/bin:/c/Strawberry/perl/site/bin:/c/Strawberry/perl/bin:/cmd:/mingw64/bin:/usr/bin:/c/tools/php:/c/Program Files (x86)/sbt/bin:/c/Program Files (x86)/Subversion/bin:/c/SeleniumWebDrivers/ChromeDriver:/c/SeleniumWebDrivers/EdgeDriver:/c/ProgramData/chocolatey/lib/maven/apache-maven-3.6.3/bin:/c/Program Files/CMake/bin:/c/Program Files/Amazon/AWSCLIV2:/c/Program Files/Amazon/SessionManagerPlugin/bin:/c/Program Files/Amazon/AWSSAMCLI/bin:/c/Program Files (x86)/Google/Cloud SDK/google-cloud-sdk/bin:/c/Users/runneradmin/.dotnet/tools:/c/Users/runneradmin/AppData/Local/Microsoft/WindowsApps
---
> PATH=/mingw64/bin:/usr/bin:/c/Users/runneradmin/bin:/c/Program Files/Mercurial:/c/Program Files/MongoDB/Server/4.2/bin:/c/ProgramData/kind:/c/vcpkg:/c/cf-cli:/c/Program Files (x86)/NSIS:/c/Program Files/Mercurial:/c/hostedtoolcache/windows/stack/2.3.1/x64:/c/ProgramData/chocolatey/lib/ghc.8.10.1/tools/ghc-8.10.1/bin:/c/Program Files/dotnet:/c/mysql-5.7.21-winx64/bin:/c/Program Files/Java/jdk8u252-b09/bin:/c/SeleniumWebDrivers/GeckoDriver:/c/Program Files (x86)/sbt/bin:/c/Rust/.cargo/bin:/c/Program Files (x86)/GitHub CLI:/bin:/c/hostedtoolcache/windows/go/1.14.4/x64/bin:/c/hostedtoolcache/windows/Python/3.7.8/x64/Scripts:/c/hostedtoolcache/windows/Python/3.7.8/x64:/c/hostedtoolcache/windows/Ruby/2.5.8/x64/bin:/c/npm/prefix:/c/Program Files (x86)/Microsoft SDKs/Azure/CLI2/wbin:/c/windows/system32:/c/windows:/c/windows/System32/Wbem:/c/windows/System32/WindowsPowerShell/v1.0:/c/windows/System32/OpenSSH:/c/ProgramData/Chocolatey/bin:/c/Program Files/Docker:/c/Program Files/PowerShell/7:/c/Program Files/dotnet:/c/Program Files/Microsoft SQL Server/130/Tools/Binn:/c/Program Files/Microsoft SQL Server/Client SDK/ODBC/170/Tools/Binn:/c/Program Files (x86)/Microsoft SQL Server/110/DTS/Binn:/c/Program Files (x86)/Microsoft SQL Server/120/DTS/Binn:/c/Program Files (x86)/Microsoft SQL Server/130/DTS/Binn:/c/Program Files (x86)/Microsoft SQL Server/140/DTS/Binn:/c/Program Files (x86)/Microsoft SQL Server/150/DTS/Binn:/c/Program Files (x86)/Windows Kits/10/Windows Performance Toolkit:/c/Program Files/Microsoft Service Fabric/bin/Fabric/Fabric.Code:/c/Program Files/Microsoft SDKs/Service Fabric/Tools/ServiceFabricLocalClusterManager:/c/Program Files/nodejs:/c/Program Files/OpenSSL/bin:/c/Strawberry/c/bin:/c/Strawberry/perl/site/bin:/c/Strawberry/perl/bin:/cmd:/mingw64/bin:/usr/bin:/c/tools/php:/c/Program Files (x86)/sbt/bin:/c/Program Files (x86)/Subversion/bin:/c/SeleniumWebDrivers/ChromeDriver:/c/SeleniumWebDrivers/EdgeDriver:/c/ProgramData/chocolatey/lib/maven/apache-maven-3.6.3/bin:/c/Program Files/CMake/bin:/c/Program Files/Amazon/AWSCLIV2:/c/Program Files/Amazon/AWSSAMCLI/bin:/c/Program Files (x86)/Google/Cloud SDK/google-cloud-sdk/bin:/c/Users/runneradmin/.dotnet/tools:/c/Users/runneradmin/AppData/Local/Microsoft/WindowsApps
~~~

traversaro, 2020-07-25:
The changes in the Windows images from 20200706.1 to 20200714.1 are: https://github.com/actions/virtual-environments/compare/win19/20200706.1...win19/20200714.1 .

traversaro, 2020-07-25:
> 
> 
> The changes in the Windows images from 20200706.1 to 20200714.1 are: [actions/virtual-environments@win19/20200706.1...win19/20200714.1](https://github.com/actions/virtual-environments/compare/win19/20200706.1...win19/20200714.1) .

I could not find anything relevant there, just WSL being enabled but I tried to execute the same command locally on my machine and nothing changed.

traversaro, 2020-07-25:
Interestingly on my machine the messages when BlockFactory is cloned are: 
~~~
  Performing download step (git clone) for 'BlockFactory'
  Cloning into 'BlockFactory'...
  Your branch is up to date with 'origin/master'.
  Already on 'master'
  Skipping update step for 'BlockFactory'
  No patch step for 'BlockFactory'
  Performing configure step for 'BlockFactory'
~~~
while on GitHub Actions, also for the builds that were working correctly, the message was:
~~~
2020-07-20T02:07:40.8124953Z   Performing download step (git clone) for 'BlockFactory'
2020-07-20T02:07:40.8456030Z   Cloning into 'BlockFactory'...
2020-07-20T02:07:41.3643492Z   Already on 'master'
2020-07-20T02:07:41.3644158Z   Your branch is up to date with 'origin/master'.
2020-07-20T02:07:41.9691181Z   Performing update step for 'BlockFactory'
2020-07-20T02:07:42.2509771Z   Current branch master is up to date.
2020-07-20T02:07:42.7511782Z   No patch step for 'BlockFactory'
2020-07-20T02:07:42.7846088Z   Performing configure step for 'BlockFactory'
~~~

So in my machine the update step is being "skipped", while in CI not.

traversaro, 2020-07-25:
Even locally, 

> Interestingly on my machine the messages when BlockFactory is cloned are:
>
> ```
>   Performing download step (git clone) for 'BlockFactory'
>   Cloning into 'BlockFactory'...
>   Your branch is up to date [...]

This difference is probably due to the use of `NON_INTERACTIVE_BUILD` CMake option, if I set it to ON also on my machine I see that the update step is not skipped also locally:
~~~
  Checking Build System
  Creating directories for 'BlockFactory'
  Performing download step (git clone) for 'BlockFactory'
  Cloning into 'BlockFactory'...
  Your branch is up to date with 'origin/master'.
  Already on 'master'
  Performing update step for 'BlockFactory'
  Current branch master is up to date.
  No patch step for 'BlockFactory'
  Performing configure step for 'BlockFactory'
  Not searching for unused variables given on the command line.
  loading initial cache file C:/src/robotology-superbuild/build/robotology/Block
~~~
However, even with that option I am unable to replicate the error.

traversaro, 2020-07-25:
In https://github.com/robotology/robotology-superbuild/pull/434 I tried to test a standalone clone of BlockFactory to understand what was going on. The output of the commands on Ubuntu (everythink is working normally) is: 
~~~
  git clone https://github.com/robotology/blockfactory
  cd blockfactory
  git status
  git diff
  shell: /bin/bash --noprofile --norc -e -o pipefail {0}
  env:
    BOOST_ROOT: 
Cloning into 'blockfactory'...
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
~~~

while on Windows is: 
~~~
  git clone https://github.com/robotology/blockfactory
  cd blockfactory
  git status
  git diff
  shell: C:\Program Files\Git\bin\bash.EXE --noprofile --norc -e -o pipefail {0}
  env:
    BOOST_ROOT: 
Cloning into 'blockfactory'...
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   example/.clang-format

no changes added to commit (use "git add" and/or "git commit -a")
~~~

So the file that is creating problems in `example/.clang-format` .

traversaro, 2020-07-25:
> So the file that is creating problems in `example/.clang-format` .

Indeed, this file is "special": it is a symbolic link, see https://github.com/robotology/blockfactory/blob/master/example/.clang-format .

traversaro, 2020-07-25:
I am not sure what changed at the GitHub Action image level, but by default, Symbolic Links are not supported in GitHub for Windows, and they need to be manually enabled during installation or in a text configuration. This for example happens if I try to see the file on Windows:
~~~
STraversaro@IITICUBLAP102 MINGW64 /c/src/blockfactory (master)
$ head example/.clang-format
../.clang-format
~~~
while with the same file on WSL I get:
~~~
traversaro@IITICUBLAP102:~/blockfactory$ head example/.clang-format
---
Language:        Cpp
# BasedOnStyle:  WebKit
AccessModifierOffset: -4
AlignAfterOpenBracket: Align
AlignConsecutiveAssignments: false
AlignConsecutiveDeclarations: false
AlignEscapedNewlines: Left
AlignOperands:   true
AlignTrailingComments: false
~~~

For this reason, I think in general, at least for repos that are meant to be multi-platform it would be better to try to avoid symbolic links, as making sure that users have the right Git setting would be quite error-prone.

traversaro, 2020-07-25:
I finally found the PR that caused this regression, by enabling Git symlinks on Windows: https://github.com/actions/virtual-environments/pull/1186 .

traversaro, 2020-07-27:
Local fix for the issue provided in https://github.com/robotology/robotology-superbuild/pull/435 . I will also open a PR to remove the symlinks from the blockfactory repo (that in any case do not work on Windows) in the next days.

diegoferigo, 2020-08-24:
Oh wow this has been a long and I think tiring debugging process for a simple symlink :/ Thanks @traversaro for taking care of it.

traversaro, 2020-08-24:
Fixed by https://github.com/robotology/blockfactory/pull/56 .

closed 2020-08-24 by traversaro