robotology/robotology-superbuild #440: GitHub Actions macOS build is failing

traversaro, 2020-07-31:
For the usual homebrew update problems:
~~~
2020-07-31T02:19:59.4766700Z ==> Installing osrf/simulation/gazebo11
2020-07-31T02:20:15.2296560Z ==> Patching
2020-07-31T02:20:15.2297100Z ==> Applying brew_python_fix.patch
2020-07-31T02:20:15.2430950Z patching file deps/opende/CMakeLists.txt
2020-07-31T02:20:15.2475190Z ==> cmake .. -DQWT_WIN_INCLUDE_DIR=/usr/local/lib/qwt.framework/Headers -DQWT_WIN_LIBRARY_DIR=/usr/local/lib/qwt.framework
2020-07-31T02:21:11.2612540Z ==> make install
2020-07-31T02:45:52.7464010Z Last 15 lines from /Users/runner/Library/Logs/Homebrew/gazebo11/02.make:
2020-07-31T02:45:52.8097010Z cd /tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0/build/gazebo && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBUILDING_DLL -DHAVE_OPENGL -DLIBBULLET_VERSION=2.89 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dlibgazebo_EXPORTS -I/tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0/test/gtest/include -I/tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0 -I/tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0/build -I/usr/local/Cellar/ffmpeg/4.3.1/include -I/usr/local/Cellar/libusb/1.0.23/include/libusb-1.0 -isystem /usr/local/Cellar/tinyxml2/8.0.0/include -isystem /usr/local/include/ignition/msgs5 -isystem /usr/local/include/ignition/math6 -isystem /usr/local/include/ignition/transport8 -isystem /usr/local/include/ignition/common3 -isystem /usr/local/include/ignition/fuel_tools4 -iframework /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Paging -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Overlay -isystem /usr/local/Cellar/sdformat9/9.2.0_3/include/sdformat-9.2 -isystem /usr/local/opt/tinyxml/include -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/include/sdformat-9.2/sdf/.. -isystem /usr/local/Cellar/tbb/2020_U2/include -isystem /usr/local/include/ignition/cmake2 -isystem /usr/local/include/eigen3 -isystem /usr/local/include/bullet -isystem /usr/local/Cellar/urdfdom/1.0.4_2/include -isystem /usr/local/lib/urdfdom_headers/cmake/../../../include -isystem /usr/local/include/simbody -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11  -s -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fPIC   -fPIC -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -std=c++17 -o CMakeFiles/libgazebo.dir/Server.cc.o -c /tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0/gazebo/Server.cc
2020-07-31T02:45:52.8098260Z [ 72%] Building CXX object gazebo/CMakeFiles/libgazebo.dir/Master.cc.o
2020-07-31T02:45:52.8101830Z cd /tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0/build/gazebo && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBUILDING_DLL -DHAVE_OPENGL -DLIBBULLET_VERSION=2.89 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dlibgazebo_EXPORTS -I/tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0/test/gtest/include -I/tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0 -I/tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0/build -I/usr/local/Cellar/ffmpeg/4.3.1/include -I/usr/local/Cellar/libusb/1.0.23/include/libusb-1.0 -isystem /usr/local/Cellar/tinyxml2/8.0.0/include -isystem /usr/local/include/ignition/msgs5 -isystem /usr/local/include/ignition/math6 -isystem /usr/local/include/ignition/transport8 -isystem /usr/local/include/ignition/common3 -isystem /usr/local/include/ignition/fuel_tools4 -iframework /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Paging -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Overlay -isystem /usr/local/Cellar/sdformat9/9.2.0_3/include/sdformat-9.2 -isystem /usr/local/opt/tinyxml/include -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/include/sdformat-9.2/sdf/.. -isystem /usr/local/Cellar/tbb/2020_U2/include -isystem /usr/local/include/ignition/cmake2 -isystem /usr/local/include/eigen3 -isystem /usr/local/include/bullet -isystem /usr/local/Cellar/urdfdom/1.0.4_2/include -isystem /usr/local/lib/urdfdom_headers/cmake/../../../include -isystem /usr/local/include/simbody -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11  -s -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fPIC   -fPIC -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -std=c++17 -o CMakeFiles/libgazebo.dir/Master.cc.o -c /tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0/gazebo/Master.cc
2020-07-31T02:45:52.8102910Z [ 72%] Building CXX object gazebo/CMakeFiles/libgazebo.dir/gazebo.cc.o
2020-07-31T02:45:52.8106030Z cd /tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0/build/gazebo && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBUILDING_DLL -DHAVE_OPENGL -DLIBBULLET_VERSION=2.89 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dlibgazebo_EXPORTS -I/tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0/test/gtest/include -I/tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0 -I/tmp/gazebo11-20200731-15435-1q91zzh/gazebo-11.0.0/build -I/usr/local/Cellar/ffmpeg/4.3.1/include -I/usr/local/Cellar/libusb/1.0.23/include/libusb-1.0 -isystem /usr/local/Cellar/tinyxml2/8.0.0/include -isystem /usr/local/include/ignition/msgs5 -isystem /usr/local/include/ignition/math6 -isystem /usr/local/include/ignition/transport8 -isystem /usr/local/include/ignition/common3 -isystem /usr/local/include/ignition/fuel_tools4 -iframework /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Paging -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Overlay -isystem /usr/local/Cellar/sdformat9/9.2.0_3/include/sdformat-9.2 -isystem /usr/local/opt/tinyxml/include -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/include/sdformat-9.2/sdf/.. -isystem /usr/local/Cellar/tbb/2020_U2/include -isystem /usr/local/include/ignition/cmake2 -isystem [... cut ...]
2020-07-31T02:45:57.0432240Z http.https://github.com/.extraheader
2020-07-31T02:45:57.0446690Z [command]/usr/local/bin/git config --local --unset-all http.https://github.com/.extraheader
2020-07-31T02:45:57.0542400Z [command]/usr/local/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2020-07-31T02:45:57.2032260Z Cleaning up orphan processes
~~~

traversaro, 2020-07-31:
Related upstream issues: 
* https://github.com/osrf/homebrew-simulation/issues/1081
* https://github.com/osrf/homebrew-simulation/issues/1076

traversaro, 2020-08-03:
Now the jobs are failing with:
~~~
2020-08-02T02:06:39.7007090Z Error: Cannot install bazelisk because conflicting formulae are installed.
2020-08-02T02:06:39.7007390Z   bazel: because Bazelisk replaces the bazel binary
2020-08-02T02:06:39.7007650Z 
2020-08-02T02:06:39.7007830Z Please `brew unlink bazel` before continuing.
~~~

pattacini, 2020-08-03:
Yep, noticed the same happening for `icub-basic-demos`.
Any suggestions to get around this?

traversaro, 2020-08-03:
I guess as suggested a `brew unlink bazel`  before all the brew related commands can be helpful, but I did not checked in detail.

pattacini, 2020-08-03:
I've checked it unsuccessfully.
But perhaps I've added it up in the wrong location.

pattacini, 2020-08-03:
@traversaro the fix in https://github.com/robotology/icub-basic-demos/pull/26 works!

Beforehand, I was unlinking `bazel` right after brew upgrade, that's why.

traversaro, 2020-08-04:
> Now the jobs are failing with:
>
> ```
> 2020-08-02T02:06:39.7007090Z Error: Cannot install bazelisk because conflicting formulae are installed.
> 2020-08-02T02:06:39.7007390Z   bazel: because Bazelisk [...]

This error now disappear on its own, and the problem is again that `gazebo11` is not installing.

nunoguedelha, 2020-08-10:
@traversaro , FYI, I downloaded Gazebo full source code and rebased the fix https://github.com/osrf/gazebo/pull/2809 (fixing the boost::bind issue) on top of gazebo11 branch. I could then build gazebo11 successfully, after installing the dependencies through homebrew:
1. `brew install gazebo11 --only-dependencies`
2. Gazebo build from source instruction here: build from source instructions here: https://github.com/osrf/gazebo_tutorials/blob/master/install_from_source/tutorial_default.md#build-and-install-gazebo
3. Add `<CMAKE_INSTALL_PREFIX>/bin` to `PATH` env variable.
4. Add `<CMAKE_INSTALL_PREFIX>/lib` to `DYLD_LIBRARY_PATH` env variable.

After this, Gazebo starts without problems. I am on Catalina.