robotology/robotology-superbuild #462: Require CMake 3.16 ?

traversaro, 2020-09-02:
We recently discovered that we accidentally require CMake 3.14 in iDynTree devel, in particular in https://github.com/robotology/idyntree/pull/726 (see https://github.com/robotology/robotology-superbuild/pull/458#issuecomment-685798127). While I could find a workaround to get it working on the CMake 3.13 that ships by default on Debian Buster (current stable), I wonder if it make sense to spend time on that, given that: 
* On Windows/macOS, we can assume that users use a recent CMake. 
* On Ubuntu 20.04, the default installed CMake is 3.16 
* On Ubuntu 18.04, we already suggest to users to install the Kitware apt repository (see https://apt.kitware.com/), so all Ubuntu 18.04 user already have the latest 3.17 
* On Debian Stretch/Old Stable (that I do not think we explicitly support anymore) we suggest user to instlal cmake from the stretch-backports, resulting in CMake 3.16 being available 
* Just on Debian Buster/Stable we are using (according to the superbuild instructions) CMake 3.13 

By just suggesting to install the CMake from `buster-backports` to Debian Buster/Stable users, as we already do for stretch user, we could be able to require CMake 3.16 for all our main supported targets. My personal opinion is that the tradeoff between the minor annoyance of installing CMake from backports on Debian Stable while avoiding all the problems in supporting old CMake is worth it, so I personally suggest that from now on we permit to **superbuild projects to require CMake 3.16**.

traversaro, 2020-09-02:
This is something we can discuss about on Friday @diegoferigo @S-Dafarra @GiulioRomualdi @drdanz @pattacini @Nicogene

traversaro, 2020-09-02:
Another benefit of requiring CMake 3.16 is that we could remove further complexity from YCM (see https://github.com/robotology/ycm/blob/0198407e557af47cc70735cf4e5761ec6c4969df/cmake-next/CMakeLists.txt#L114).

diegoferigo, 2020-09-03:
Sorry for triggering from side ways this discussion :) From my side, I think this is a reasonable requirement. From multiple directions we are also moving towards an increased Python support and usage, CMake > 3.15 has much better capabilities to select the right interpreter between system and virtualenv.

p.s. is [still this](http://wiki.icub.org/wiki/YARP_Supported_Distributions) the most updated page with the supported software?

traversaro, 2020-09-03:
> p.s. is still this the most updated page with the supported software?

Kind of, even if I guess it was not updated with the latest details on 20.04 and similar.

drdanz, 2020-09-03:
> Just on Debian Buster/Stable we are using (according to the superbuild instructions) CMake 3.13

I was going to say that also Ubuntu 19.10 Eoan is still supported and has CMake 3.13, but I realized it reached the end of life in july... 

---

> On Ubuntu 18.04, we already suggest to users to install the Kitware apt repository (see https://apt.kitware.com/), so all Ubuntu 18.04 user already have the latest 3.17

According to the policies we could drop support for ubuntu 18.04. Unfortunately we still have lots of things running that.
One problem that we have to consider in the future is that, if I remember correctly, the kitware repository does not support arm, and we do have a few arm machines (nvidia jetson for example). Anyway in the PPA we have CMake 3.16 for ubuntu 18.04 for several architectures, therefore we could eventually use recommend that.

By the way, I would like to propose to change this in our policies:
```diff
-Only the latest one is supported. When a new long term release is released, we support the previous one for either 3 or 6 extra months, in order to give some time to the users to upgrade their system.
+Only the latest one is supported. When a new long term release is released, we support the previous one for either 3 or 6 extra months after the first patch release, in order to give some time to the users to upgrade their system.
```

Since basically it has always been like, and since installing new ubuntu releases as soon as they are out, has proven in many occasions to be causing issues.

---

> Another benefit of requiring CMake 3.16 is that we could remove further complexity from YCM (see https://github.com/robotology/ycm/blob/0198407e557af47cc70735cf4e5761ec6c4969df/cmake-next/CMakeLists.txt#L114).

This part of code is not very complex, it is actually very straight forward...  it's just a matter of selecting the files and adding the right hash, calling a method to download them and install the downloaded files.
The code actually performing the download is shared with the 3rd party and cannot be removed.
Anyway it would remove some other workaround like https://github.com/robotology/ycm/blob/0198407e557af47cc70735cf4e5761ec6c4969df/CMakeLists.txt#L64-L73

---

> p.s. is still this the most updated page with the supported software?

Yes, unfortunately it must be updated manually.
I just updated Debian testing and ubuntu groovy, feel free to update it if you see that something is wrong :smile: 

---

> Kind of, even if I guess it was not updated with the latest details on 20.04 and similar.

It was updated but i messed up the table and it wasn't visible :sweat_smile: 

---

> This is something we can discuss about on Friday

I don't have a strong opinion on this, I was planning to update to require CMake 3.13 in YARP master, I feel like that requiring CMake 3.12 in advance caused a few hiccups, but nothing that could not be handled. If everyone is ok with 3.16 it's ok with me...

traversaro, 2020-09-09:
Given that there is consensus on this, I think we can proceed.

closed 2020-09-09 by traversaro