robotology/robotology-superbuild #485: Release 2020.08.patch-01

traversaro, 2020-10-11:
Version to bump:
* YARP from v3.4.0 to v3.4.1
* YCM from v0.11.3 to v0.11.4 
* GazeboYARPPlugins from v3.5.0 to v3.5.1 (still needs to be released)

traversaro, 2020-10-11:
@pattacini do you want to add some fixes for icub-main ? In that case it would be great to to a patch release of icub-main.

pattacini, 2020-10-11:
We may want to fix https://github.com/robotology/icub-main/issues/685.
Awaiting reply from @ale-git.

traversaro, 2020-10-11:
> 
> 
> We may want to fix [robotology/icub-main#685](https://github.com/robotology/icub-main/issues/685).
> Awaiting a reply from @ale-git.

Ack, even if (at least on my side) there is no urgency in fixing https://github.com/robotology/icub-main/issues/685 .

traversaro, 2020-10-12:
Yes, no problem! I just pinged you in the case you had some fix that you wanted to release, not because I wanted https://github.com/robotology/icub-main/issues/685 to be fixed anytime soon.

traversaro, 2020-10-12:
The release has been tagged at [`v2020.08.patch-01`](https://github.com/robotology/robotology-superbuild/tree/v2020.08.patch-01) and autogenerated Windows binaries are available at https://github.com/robotology/robotology-superbuild/releases/tag/v2020.08.patch-01 . Unless there is a specific problems, I would close the issue.

closed 2020-10-19 by traversaro